Bodmer/TFT_eSPI #1696: RM68120 support

Fischkopppp, 2022-03-10:
Hi Bodmer, 
do you think you could add support for the RM68120 display driver (8bit parallel)? If I would send you this display:
https://de.aliexpress.com/item/32878013749.html
With best regards!

Bodmer, 2022-03-11:
To do this I need the initialisation code sequence. Do you have a copy and have you you tested it to show it works?

Fischkopppp, 2022-03-11:
I do not have this display. I thought about buying one. I asked the seller for the datasheet and for the initialisation code sequence. I will write again when I have the datasheet.

Bodmer, 2022-03-11:
I found some init code here that might work.
https://github.com/espressif/esp-dev-kits/blob/master/esp32-s2-hmi-devkit-1/components/screen/controller_driver/rm68120/rm68120.c

An adaption of the example code there would also be a good starting point to test the display interface.

Fischkopppp, 2022-03-11:
I was wondering what display Espressif is using on the HMI Dev kit. Now I know ;-)
If I buy myself a display, should I send one to you, too? Shipping will take a month.
You can send me a pm if you want: bikedynamos@googlemail.com

Bodmer, 2022-03-11:
The HMI devkit code seems to support quite a few displays:
https://github.com/espressif/esp-dev-kits/tree/master/esp32-s2-hmi-devkit-1/components/screen/controller_driver

I am concerned that the RM68120 seems limited in vendors and it is an old chip design. So supplies may dry up. If you want touch then go for a capacitive one as the resistance screens scratch easily and are typically noisy and inaccurate. Capacitance ones often support multi-touch and swipe too.