Bodmer/TFT_eSPI #1701: PNG Transparency

ruebarb, 2022-03-11:
How do you use transparency with the png? For me that is the advantage of using png vs jpg for graphics.

Thanks

I'm working with Flash_PNG

How do I use your function, where do I get the **uint16_t transparent** value?

I thought I could just use the returned value from getAlphaMask but it is uint8_t and not the required unit16_t

**TFT_eSPI.h**
void     pushImage(int32_t x, int32_t y, int32_t w, int32_t h, const uint16_t *data, **uint16_t transparent**);

**PNGDec.cpp**
uint8_t PNG::getAlphaMask(PNGDRAW *pDraw, **uint8_t *pMask**, uint8_t ucThreshold)

In the png_transparency example

uint8_t ucMask[40];
png.getAlphaMask(pDraw, ucMask, 255)

Bodmer, 2022-03-12:
I have not used the PNGdec library much so am not particularly familiar with the functions. It appears getAlphaMask function only returns a 1bit per pixel mask with the set 8 bit alpha threshold.

Probably best ask Larry Bank, the author of the [PNGdec library](https://github.com/bitbank2/PNGdec). I don't think the library has a function to return the 8 bit alpha channel.

The TFT_eSPI library provides an alphaBlend function to support an alpha channel. The TFT must be one that is supported for reading if you wish to render with an alphaChannel direct to the TFT. Alternatively a PNG  can be rendered into a sprite.

There is a function to support alpha blending on a [per pixel basis here](https://github.com/Bodmer/TFT_eSPI/blob/master/TFT_eSPI.cpp#L3674-L3684), but a more efficient line based rendering function would be needed to improve performance.

closed 2022-03-12 by Bodmer

Bodmer, 2022-03-12:
P.S. I put together some example sketches [here](https://github.com/Bodmer/PNG_TEST_ONLY) using the [pngle ](https://github.com/kikuchan/pngle)code by kikuchan. That png decoder does provide access to the alpha channel (if present) as shown[ here](https://github.com/Bodmer/PNG_TEST_ONLY/blob/master/png_to_sprite/support_functions.h#L23). Although the alpha is not used in the examples it should be quite easy to adapt the code.

At some point I may build an Arduino PNG decode library based on pngle but that is not a priority at the moment.

Bodmer, 2022-03-12:
I see that kitchen has started to migrate the pngle code into an Arduino library. However at the moment this is not in the right format and the Arduino ino example freezes due to a bug. Worth keeping an eye on that library.