Bodmer/TFT_eSPI #1720: Problem with ST7735s

lenapepl, 2022-03-15:
Hi everyone,

After going through all threads and tutorials possible, I still can't find a solution to my problem. I try to run TFT_eSPI on ESP32 with 128x160 LCD display with ST7735s driver, but the display is blank white, no matter what changes to User_Setup I make. Also, I tried choosing the driver from User_Setup_Select, still nothing. I've tried out many sketches from examples library, the display is still blank. Here is my configuration: 

`//                            USER DEFINED SETTINGS
//   Set driver type, fonts to be loaded, pins used and SPI control method etc
//
//   See the User_Setup_Select.h file if you wish to be able to define multiple
//   setups and then easily select which setup file is used by the compiler.
//
//   If this file is edited correctly then all the library example sketches should
//   run without the need to make any more changes for a particular hardware setup!
//   Note that some sketches are designed for a particular TFT pixel width/height

// ##################################################################################
//
// Section 1. Call up the right driver file and any options for it
//
// ##################################################################################

// Define STM32 to invoke optimised processor support (only for STM32)
//#define STM32

// Defining the STM32 board allows the library to optimise the performance
// for UNO compatible "MCUfriend" style shields
//#define NUCLEO_64_TFT
//#define NUCLEO_144_TFT

// STM32 8 bit parallel only:
// If STN32 Port A or B pins 0-7 are used for 8 bit parallel data bus bits 0-7
// then this will improve rendering performance by a factor of ~8x
//#define STM_PORTA_DATA_BUS
//#define STM_PORTB_DATA_BUS

// Tell the library to use 8 bit parallel mode (otherwise SPI is assumed)
//#define TFT_PARALLEL_8_BIT

// Display type -  only define if RPi display
//#define RPI_DISPLAY_TYPE // 20MHz maximum SPI

// Only define one driver, the other ones must be commented out
//#define ILI9341_DRIVER       // Generic driver for common displays
//#define ILI9341_2_DRIVER     // Alternative ILI9341 driver, see https://github.com/Bodmer/TFT_eSPI/issues/1172
#define ST7735_DRIVER      // Define additional parameters below for this display
//#define ILI9163_DRIVER     // Define additional parameters below for this display
//#define S6D02A1_DRIVER
//#define RPI_ILI9486_DRIVER // 20MHz maximum SPI
//#define HX8357D_DRIVER
//#define ILI9481_DRIVER
//#define ILI9486_DRIVER
//#define ILI9488_DRIVER     // WARNING: Do not connect ILI9488 display SDO to MISO if other devices share the SPI bus (TFT SDO does NOT tristate when CS is high)
//#define ST7789_DRIVER      // Full configuration option, define additional parameters below for this display
//#define ST7789_2_DRIVER    // Minimal configuration option, define additional parameters below for this display
//#define R61581_DRIVER
//#define RM68140_DRIVER
//#define ST7796_DRIVER
//#define SSD1351_DRIVER
//#define SSD1963_480_DRIVER
//#define SSD1963_800_DRIVER
//#define SSD1963_800ALT_DRIVER
//#define ILI9225_DRIVER
//#define GC9A01_DRIVER

// Some displays support SPI reads via the MISO pin, other displays have a single
// bi-directional SDA pin and the library will try to read this via the MOSI line.
// To use the SDA line for reading data from the TFT uncomment the following line:

// #define TFT_SDA_READ      // This option is for ESP32 ONLY, tested with ST7789 and GC9A01 display only

// For ST7735, ST7789 and ILI9341 ONLY, define the colour order IF the blue and red are swapped on your display
[... 266 lines cut ...]
#define SPI_READ_FREQUENCY  20000000

// The XPT2046 requires a lower SPI clock rate of 2.5MHz so we define that here:
#define SPI_TOUCH_FREQUENCY  2500000

// The ESP32 has 2 free SPI ports i.e. VSPI and HSPI, the VSPI is the default.
// If the VSPI port is in use and pins are not accessible (e.g. TTGO T-Beam)
// then uncomment the following line:
//#define USE_HSPI_PORT

// Comment out the following #define if "SPI Transactions" do not need to be
// supported. When commented out the code size will be smaller and sketches will
// run slightly faster, so leave it commented out unless you need it!

// Transaction support is needed to work with SD library but not needed with TFT_SdFat
// Transaction support is required if other SPI devices are connected.

// Transactions are automatically enabled by the library for an ESP32 (to use HAL mutex)
// so changing it here has no effect

// #define SUPPORT_TRANSACTIONS`

I was wondering if maybe the ST7735 setup doesn't work for ST7735s driver, but from what I get from other threads it shouldn't be a problem. As you might've guessed, I am a super newbie in learning process, will be grateful for any advice!

Best regards

Bodmer, 2022-03-16:
Have you tried reducing the zSPI frequency to 20MHz and have you tried oth colour tab options, like #define ST7735_REDTAB or #define ST7735_BLACKTAB?

lenapepl, 2022-03-16:
Thank you for the response.
I tried the 20MHz frequency and the pong sketch worked for maybe 10 secs, then went blank again. I tried all of the colour tab options.
I also changed wires, because it turned out I mixed DIN and CLK pin up, so now it is:
VCC -> VIN
GND -> GND
DIN -> D23
CLK -> D18
CS -> D15
DC -> D2
RST -> D4
BLK -> 3V3

I've been wondering about MISO pin defined in the code, as there is no such pin in my display.

Bodmer, 2022-03-16:
It is unusual for a display to only run for 10 secs, at least this shows communication is happening.
Post a picture of your setup and also the back of the display. 
I assume VIN is 5V, if so have you checked the display has a 5V to 3.3V regulator fitted? If you are not sure then I will be able to see this from the picture of the back of the display.
The ESP32 MISO pin 19 can be left unconnected, do not use that pin for anything else.

lenapepl, 2022-03-16:
Update: the display started to work after unplugging and plugging to the computer again! 
Nevertheless, here is the photo of the wiring:

[image]
[image]
[image]
[image]

Best regards

lenapepl, 2022-03-16:
Oh, thank you so much. Is it necessary to connect Backlight to some pin? It seems to work without it after rewiring.

Bodmer, 2022-03-16:
The backlight is powered via a small transistor on the back of the board.

Since this is biased in the ON state by default in this display you do not need to connect the BL pin.

I expect if you connect the BL pin to 0v (GND) this backlight will be turned off. The pin allows the on/off state (or brightness via a PWM signal) to be controlled from the CPU if desired.

Bodmer, 2022-03-16:
I have added a warning here for others: #1721

closed 2022-03-16 by lenapepl