Bodmer/TFT_eSPI #1726: "exit status 1 Error compiling for board ESP32 Dev Module. "

farhank5023, 2022-03-18:
hi,
i'm facing an issue in esp32 board. plz help with the error message below.

Arduino: 1.8.19 (Windows 10), Board: "ESP32 Dev Module, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 115200, None"

In file included from C:\Users\farha\Documents\Arduino\libraries\DHTLib\dht.h:18:0,

                 from C:\Users\farha\Documents\Arduino\libraries\DHTLib\dht.cpp:30:

C:\Users\farha\Documents\Arduino\libraries\DHTLib\dht.cpp: In member function 'int dht::_readSensor(uint8_t, uint8_t)':

C:\Users\farha\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.3-rc3\cores\esp32/Arduino.h:106:91: error: cannot convert 'volatile uint32_t* {aka volatile unsigned int*}' to 'volatile uint8_t* {aka volatile unsigned char*}' in initialization

 #define portInputRegister(port)     ((volatile uint32_t*)((port)?GPIO_IN1_REG:GPIO_IN_REG))

                                                                                           ^

C:\Users\farha\Documents\Arduino\libraries\DHTLib\dht.cpp:116:29: note: in expansion of macro 'portInputRegister'

     volatile uint8_t *PIR = portInputRegister(port);

                             ^

Multiple libraries were found for "WiFi.h"

 Used: C:\Users\farha\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.3-rc3\libraries\WiFi

 Not used: C:\Program Files (x86)\Arduino\libraries\WiFi

exit status 1

Error compiling for board ESP32 Dev Module.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

moreauftheobald, 2022-03-18:
It seems that your problem is not relative to tft_SPI but to DHTlib. Maybe you should ask for help to the person who Made it ?

Bodmer, 2022-03-18:
The errors are associated with the DHTlib library.

If it is [this library ](https://www.arduino.cc/reference/en/libraries/dhtlib/)then note it appears to be designed for the AVR processors, i.e UNO etc. So it not surprising that it generates arrors on an ESP32.

You should be able to find a DHT library for the ESP32.

May be this will help:
https://randomnerdtutorials.com/esp32-dht11-dht22-temperature-humidity-sensor-arduino-ide/

closed 2022-03-18 by Bodmer

farhank5023, 2022-03-18:
@Bodmer @moreauftheobald thankyou for your quick response, i will get back with updates soon.