Bodmer/TFT_eSPI #1728: Issues Using Free Font With Sprite

audreylim, 2022-03-18:
I'm trying to display changing numbers without flickering. Following this example and using NotoSans font worked:

https://github.com/Bodmer/TFT_eSPI/blob/master/examples/Smooth%20Fonts/FLASH_Array/Font_Demo_4_Array/Font_Demo_4_Array.ino

```
  spr.loadFont(AA_FONT_LARGE);
  spr.setTextColor(TFT_WHITE, TFT_BLACK);
  tft.setCursor(0, 95); 
  spr.printToSprite("123");
  spr.unloadFont();
```

However when trying to use free font, the following didn't work:

```
 #define FF_LARGE &FreeSans18pt7b

  tft.setFreeFont(FF_LARGE);
  spr.setTextColor(TFT_WHITE, TFT_BLACK);
  spr.drawString("123", 0, 95, GFXFF);
```
and
```
  #define FF_LARGE &FreeSans18pt7b

  tft.setFreeFont(FF_LARGE);
  tft.setCursor(0, 95); 
  spr.printToSprite("123");
```

Is it possible to use free font with sprites? What am I missing?

Bodmer, 2022-03-18:
I suspect if you use:
```
spr.setFreeFont(FF_LARGE);
```

It will work.

Since the text is printed in the sprite first, it is the sprite that needs to have the font loaded.

I know this can get confusing...

audreylim, 2022-03-18:
Oh, yes I also tried spr.setFreeFont(FF_LARGE) and it didn't work...

Bodmer, 2022-03-18:
Ah, yes. I have just checked and printToSprite only works for smooth fonts!

You can print a free font in a sprite that has already been created.

Example[ here](https://github.com/Bodmer/TFT_eSPI/blob/master/examples/Sprite/Sprite_scroll_16bit/Sprite_scroll_16bit.ino#L121-L149).

audreylim, 2022-03-18:
That works, thank you! It'll be great if printToSprite supports free fonts too.

closed 2022-03-18 by audreylim

Bodmer, 2022-03-18:
I tend to add features to support my own projects and I got lazy when I added printToSprite. I will add to my (long) to do list!