Bodmer/TFT_eSPI #1732: ESP32 SPI Interface ST7796(320x480) LVGL Benchmark problem

LusterWong, 2022-03-19:
LCD resolution is 320x480 and SPI frequency is 6000000，I tried using 8000000 but it didn't show up.
I use esp32 Arduino 2.02 library, and TFT_eSPI library is up to date (updated today).
LVGL was then used and the performance test was performed with the following results. I thought double buffering & DMA would give me the maximum FPS, but it was a surprise. The test is divided into two parts, the first is single buffer without DMA.

```
//LVGL interface. Not DMA single buffer.
void DispFlush(lv_disp_drv_t* Disp, const lv_area_t* Area, lv_color_t* Color )
{
    uint32_t w = (Area->x2 - Area->x1 + 1 );
    uint32_t h = (Area->y2 - Area->y1 + 1 );

    TFT.startWrite();
    TFT.setAddrWindow(Area->x1, Area->y1, w, h);
    TFT.pushPixels((uint16_t*)&Color->full, w * h, true);
    TFT.endWrite();

    lv_disp_flush_ready(Disp);
}

//single buffer in sram
//Opa.speed:93% FPS:15
static lv_color_t Buf1[TFT_WIDTH * 10];
lv_disp_draw_buf_init(&DrawBuf, Buf1, NULL, TFT_WIDTH * 10);

//single buffer in sram
//Opa.speed:90% FPS:20   <----------------------The top fps
static lv_color_t Buf1[TFT_WIDTH * TFT_HEIGHT / 10];
lv_disp_draw_buf_init(&DrawBuf, Buf1, NULL, TFT_WIDTH * TFT_HEIGHT / 10);

//double buffers in sram
//Opa.speed:93% FPS:16
static lv_color_t Buf1[TFT_WIDTH * 10];
static lv_color_t Buf2[TFT_WIDTH * 10];
lv_disp_draw_buf_init(&DrawBuf, Buf1, Buf2, TFT_WIDTH * 10);

//double buffers in sram
//Opa.speed:94% FPS:19
static lv_color_t Buf1[TFT_WIDTH * TFT_HEIGHT / 20];
static lv_color_t Buf2[TFT_WIDTH * TFT_HEIGHT / 20];
lv_disp_draw_buf_init(&DrawBuf, Buf1, Buf2, TFT_WIDTH * TFT_HEIGHT / 20);

//double buffers in sram
//build error "DRAM segment data does not fit."    <----------------------Error
static lv_color_t Buf1[TFT_WIDTH * TFT_HEIGHT / 10];
static lv_color_t Buf2[TFT_WIDTH * TFT_HEIGHT / 10];
lv_disp_draw_buf_init(&DrawBuf, Buf1, Buf2, TFT_WIDTH * TFT_HEIGHT / 10);
```
The results were not as good as expected, and the fastest speed was to use a single buffer at 10% of the screen size.
Double buffering has no shape enhancement, and creating buffers 10% of the size of two screens will cause a compilation error.
The next test uses DMA.
```
//DMA and double buffers
void DispFlush(lv_disp_drv_t* Disp, const lv_area_t* Area, lv_color_t* Color )
{
    uint32_t w = (Area->x2 - Area->x1 + 1 );
    uint32_t h = (Area->y2 - Area->y1 + 1 );

    TFT.startWrite();
    TFT.setAddrWindow(Area->x1, Area->y1, w, h);
    TFT.pushPixelsDMA((uint16_t*)&Color->full, w * h, true);
    TFT.endWrite();

    lv_disp_flush_ready(Disp);
}

//double buffers in sram
//Opa.speed:94% FPS:17
static lv_color_t Buf1[TFT_WIDTH * TFT_HEIGHT / 20];
static lv_color_t Buf2[TFT_WIDTH * TFT_HEIGHT / 20];
lv_disp_draw_buf_init(&DrawBuf, Buf1, Buf2, TFT_WIDTH * TFT_HEIGHT / 20);

//double buffers in psram
//Opa.speed:92% FPS:14
static lv_color_t* Buf1 = (lv_color_t*)ps_malloc((TFT_WIDTH * TFT_HEIGHT / 10) * sizeof(lv_color_t));
static lv_color_t* Buf2 = (lv_color_t*)ps_malloc((TFT_WIDTH * TFT_HEIGHT / 10) * sizeof(lv_color_t));
lv_disp_draw_buf_init(&DrawBuf, Buf1, Buf2, TFT_WIDTH * TFT_HEIGHT / 10);

//run assert failed: void TFT_eSPI::pushPixelsDMA(uint16_t*, uint32_t) TFT_eSPI_ESP32.c:647 (ret == ESP_OK)
static lv_color_t* Buf1 = (lv_color_t*)ps_malloc((TFT_WIDTH * TFT_HEIGHT) * sizeof(lv_color_t));
static lv_color_t* Buf2 = (lv_color_t*)ps_malloc((TFT_WIDTH * TFT_HEIGHT) * sizeof(lv_color_t));
lv_disp_draw_buf_init(&DrawBuf, Buf1, Buf2, TFT_WIDTH * TFT_HEIGHT);
```
The result is poorer performance using DMA and double buffering.
During the last group of tests, the serial port printed a message indicating **ret = ESP_OK**. After checking **ret = ESP_ERR_NO_MEM**, I don't know why.
I used ILI9341 with resolution 240x320, preferably with DMA and double buffering enabled, and the buffering size is 10% of the screen, but this time using ST7796 results are different.

Bodmer, 2022-03-19:
The frame rate is primarily determined by:
1. The screen resolution and interface data rate. 
2. The complexity of the page rendering performed by LVGL

The ESP32 SPI rate can be an integer division of 80MHz, so 60MHz will be rounded down to 40MHz.

The colours are 16 bits. Data volume for a full screen update is 420 x 320 x 16 bits. So doing the maths for a 40Mbps SPI rate gives 16.28 fps maximum. This is a fundamental limit.

To improve whole frame fps a higher SPI rate is needed.

LVGL will manage double buffering for DMA if configured to do so and then only one buffer pointer needs to be passed to the DMA function. If the TFT_eSPI DMA function is passed two buffer pointer instead, it will take time to make a copy of the data, so that is less efficient than using LVGL to handle the buffer.

The .endWrite() function will wait for DMA to complete to release the SPI bus, so there will be no speed improvement since activities are then not being performed in parallel. If the touch controller is SPI then you will need to manage SPI bus access for the touch functions.

closed 2022-03-19 by Bodmer

LusterWong, 2022-03-19:
> LVGL will manage double buffering for DMA if configured to do so and then only one buffer pointer needs to be passed to the DMA function.

Thx.

LVGL will manage double buffering for DMA if configured to do so and then only one buffer pointer needs to be passed to the DMA function. ----How to do that? I'm not sure about the hardware configuration.

In addition, I want to use 80Mhz but the screen will be black, ST7796 should support 80Mhz, which is not understandable

Bodmer, 2022-03-19:
Some ST7796 controllers may support higher data rates than quoted in the data sheet. But bear in mind the minimum clock cycle length in the data sheet for SPI is 66ns i.e a SPI clock rate of 15.15 MHz !!!  See here:

[image]

Typically most semiconductor parts work at 2 x quoted speed. My ST7796 displays  work relaibly at 80MHz,  however this may be just luck and if the displays get hot then this limit may drop. Note that on the ESP32 you MUST use the default SPI pins as the IO multiplexer degrades the timing if the SPI pins are routed to non-native ones. See the ESP32 data sheet for more information.

For DMA double buffering the LVGL forum and documentation will need to be consulted.  A quick Google revealed this page that perhaps helps:
https://forum.lvgl.io/t/my-experience-with-dual-dma-buffer-rendering/5899

LusterWong, 2022-03-20:
> Some ST7796 controllers may support higher data rates than quoted in the data sheet. But bear in mind the minimum clock cycle length in the data sheet for SPI is 66ns i.e a SPI clock rate of 15.15 [...]

I looked through the LCD hardware manual, and this is what it says：

[image]
The maximum is 66mhz ?

I will learn more about LVGL double buffering, thank you.

Bodmer, 2022-03-20:
Can you post a link to the datasheet. It looks like it is more recent that the year 2014 v1.0 ones I found.

LusterWong, 2022-03-20:
> Can you post a link to the datasheet. It looks like it is more recent that the year 2014 v1.0 ones I found.
[ST7796U2_SPEC_V1.0.pdf](https://github.com/Bodmer/TFT_eSPI/files/8311474/ST7796U2_SPEC_V1.0.pdf)
I find it is not ST7796 but 7796U2, not sure what is the difference.
And this doc is 2020 version.