Bodmer/TFT_eSPI #1736: Use Arial font on Sprite

audreylim, 2022-03-22:
Hi, I know relates to an external library but would you know know to set these fonts on a sprite? https://github.com/FrankBoesing/TFT_eSPI_ext/blob/main/font_Arial.h

Seems like  [eSprite](https://github.com/Bodmer/TFT_eSPI/blob/master/Extensions/Sprite.cpp#L17) didn't inherit the extension https://github.com/FrankBoesing/TFT_eSPI_ext/blob/main/TFT_eSPI_ext.h#L30

Related: https://github.com/Bodmer/TFT_eSPI/discussions/1729

Bodmer, 2022-03-22:
I don't think it will work due to the way Frank's library links to TFT_eSPI.

You can use the U8g2 fonts in a sprite, see below. You may find a font that looks similar in that library.

I think the way U8g2_for_TFT_eSPI links is better so maybe that is the way forward for Frank's library.
```
/*
  Demonstrates how to use U8g2_for_TFT_eSPI library.

  U8g2_for_TFT_eSPI:
    - Use U8g2 fonts with TFT_eSPI
    - Supports UTF-8 in print statement
    - 90, 180 and 270 degree text direction
  
  List of all U8g2 fonts:    https://github.com/olikraus/u8g2/wiki/fntlistall

  TFT_eSPI library:          https://github.com/Bodmer/TFT_eSPI
  U8g2_for_TFT_eSPI library: https://github.com/Bodmer/U8g2_for_TFT_eSPI

*/
#include "SPI.h"
#include "TFT_eSPI.h"
#include "U8g2_for_TFT_eSPI.h"

TFT_eSPI tft = TFT_eSPI();   // tft instance
U8g2_for_TFT_eSPI u8f;       // U8g2 font instance
TFT_eSprite img = TFT_eSprite(&tft);

void setup() {
  tft.begin();
  tft.setRotation(1);
  tft.fillScreen(TFT_BLACK);
  img.createSprite(106, 86);
  u8f.begin(img); // connect u8g2 procedures to sprite
}

void loop() {
  uint32_t x = 0;

  u8f.setFontMode(0);                 // use u8g2 none transparent mode
  u8f.setFontDirection(0);            // left to right (this is default)
  u8f.setForegroundColor(TFT_WHITE);  // apply color

  u8f.setFont(u8g2_font_inb63_mn);    // select u8g2 font from here: https://github.com/olikraus/u8g2/wiki/fntlistall

  while (1) {
    img.fillSprite(random(0x10000));
    u8f.setCursor(2,67);             // start writing at this position
    u8f.print(x);                     // numerical value
    img.pushSprite(0,0);
    x++;
    if (x>99) x = 0;
    delay(250);
  }
} 
```

audreylim, 2022-03-22:
Hmm, I'll look into it. Thank you!

closed 2022-03-22 by audreylim

Bodmer, 2022-03-22:
Not sure how far you have progressed with using the  U8g2 library option.... but this may be of interest:

I have changed the TFT_eSPI_ext library to be able to use the fonts in sprites or on the TFT,  I will upload that later today when I have tested it and update the examples.

audreylim, 2022-03-22:
Wow, thank you! That would be extremely helpful! I really wanted to avoid flickering by using sprites with Arial.

I took a quick look yesterday at the available fonts and didn't find one that was suitable. Having Arial on this library would be great!

Bodmer, 2022-03-22:
I have updated my fork of the TFT_eSPI_ext library here:
https://github.com/Bodmer/TFT_eSPI_ext

A new API was needed, the examples have been updated. If you could test it in you project that would be great.

audreylim, 2022-03-24:
It works beautifully. No flickering, and you can combine different fonts onto the same sprite. Thank you! =)