Bodmer/TFT_eSPI #1737: RP Pico + XPT4096 weird RawZ values

aspyra, 2022-03-24:
I'm encountering problems getting the touch controller to work with RP Pico.
I changed the standard pinout:
```
#define TFT_MISO  16
#define TFT_MOSI  19
#define TFT_SCLK  18
#define TFT_CS   28  // Chip select control pin
#define TFT_DC   26  // Data Command control pin
#define TFT_RST  27  // Reset pin (could connect to Arduino RESET pin)
//#define TFT_BL   22  // LED back-light
#define TOUCH_CS 20     // Chip select pin (T_CS) of touch screen
```
The display worked fine, but I got weird values from the RawZ readings (~64k).
I disconnected the display and tried again, same data was received. I uploaded a simple sketch to obtain only the RawZ:
```
#include <SPI.h>
#include <TFT_eSPI.h>
TFT_eSPI tft = TFT_eSPI();

void setup(void) {
  Serial.begin(115200);
  tft.init();
}

void loop() {
  char str[100];
  sprintf(str, "%u", tft.getTouchRawZ());
  Serial.println(str);
  delay(100);
}
```
And plotted the received values in Arduino IDE (not touching the screen):
[image]

I tried dropping the frequency to 100kHz:
[image]

I don't quite understand why the values changed. I tried:
- adding pull-ups
- changing the screen
- redoing the wiring
- adding capacitance to the power lines

I still get those weird values. AFAIK, the controller itself should report 12-bit values, so this must be a communication problem.
I captured the communication @ 100kHz:
[image]

I looked into the datasheet and the MOSI output looks wrong - why is the start bit not set?
Do you have any suggestions why this may be happening?

Bodmer, 2022-03-24:
Have you tried this diagnostic example:
https://github.com/Bodmer/TFT_eSPI/tree/master/examples/Test%20and%20diagnostics/Test_Touch_Controller

Bodmer, 2022-03-24:
Looking at the trace I see that 16 bit transfers are occurring on the SPI bus. They should be 8 bit. Post your setup header files. Run Read_User_Setup test and diagnostic example and post results. You may have an earlier version of the library loaded which did not support touch on the RP2040.

rcla, 2022-03-25:
I have the same problem too,  TFT_eSPI version 2.4.42 installed.

> Have you tried this diagnostic example: https://github.com/Bodmer/TFT_eSPI/tree/master/examples/Test%20and%20diagnostics/Test_Touch_Controller

That diagnostic example does not compile using Arduino Mbed OS RP2040

It shows the following error:
`'class arduino::UART' has no member named 'printf';`

However I modified it a bit:
 ```
  uint16_t x, y, z;

  tft.getTouchRaw(&x, &y);
  z = tft.getTouchRawZ();

  Serial.print("x: ");
  Serial.println( x );
  
  Serial.print("y: ");
  Serial.println( y );

  Serial.print("z: ");
  Serial.println( z );
 ```
Without touching the touch screen, It always shows the same value of x and y
 ```
x: 0
y: 7936
 ```
The value of z: varies all the time.

By touching the touch screen at a fixed point on the screen, All x y and z values ​​vary all the time.

I discovered, that using earlephilhower's board manager, everything works!

aspyra, 2022-03-25:
> Looking at the trace I see that 16 bit transfers are occurring on the SPI bus. They should be 8 bit. Post your setup header files. Run Read_User_Setup test and diagnostic example and post results. You may have an earlier version of the library loaded which did not support touch on the RP2040.

I am using 2.4.42 version installed via PlatformIO. It seems that 2.4.43 is not available there yet...
Here's the diagnostic output:
```
TFT_eSPI ver = 2.4.42
Processor    = RP2040
Transactions = Yes   
Interface    = SPI   
Display driver = 9341
Display width  = 240 
Display height = 320 

MOSI    = GPIO 19
MISO    = GPIO 16
SCK     = GPIO 18
TFT_CS   = GPIO 28
TFT_DC   = GPIO 26
TFT_RST  = GPIO 27
TOUCH_CS = GPIO 20

Font GLCD   loaded
Font 2      loaded
Font 4      loaded
Font 6      loaded
Font 7      loaded
Font 8      loaded
Smooth font enabled

Display SPI frequency = 70.00
Touch SPI frequency   = 0.10
```

> I discovered, that using earlephilhower's board manager, everything works!

I'll switch to Arduino IDE and check that...

> Without touching the touch screen, It always shows the same value of x and y
> ...
> By touching the touch screen at a fixed point on the screen, All x y and z values ​​vary all the time.

I observed the same behaviour.

Bodmer, 2022-03-25:
This looks like a bug in the Arduino Mbed board support package which does not fully re-initialise the SPI port on the next transaction.
I will run some tests with the Mbed package. I use the Philhower board package as it is better maintained, supports a large number of boards and has fast bug fixes an updates.

Bodmer, 2022-03-25:
Glad you have it working now.

FYI Touch works for me with the both latest Arduino Mbed package and the Philhower package. Upgrade if you are using the deprecated package:
[image]

DoctorDeatch, 2022-08-17:
Colleagues, good afternoon. faced with the same problem
, tell me how to solve it?
I use PlatformIO for compilation
	TFT_eSPI 2.4.74
	GFX Library for Arduino 1.2.6

I get the following values without touching:
x: 0     y: 3402     z: 38 
x: 0     y: 3354     z: 41 
when touching:
x: 1989     y: 1907     z: 1755 
x: 1999     y: 1906     z: 1816 
x: 1996     y: 1908     z: 1831 
x: 1833     y: 1908     z: 1859 

how to fix this situation?

Bodmer, 2022-08-17:
@DoctorDeatch 
Those results indicate the touch interface is working OK.

DoctorDeatch, 2022-08-17:
> @DoctorDeatch Those results indicate the touch interface is working OK.

is it true that in the absence of touch x = 0 at the same time y = 3000?
how to bring to "0" and not max value, in the absence of touch?