Bodmer/TFT_eSPI #1738: Issue STMDUINO STM32F103(Blue pill) + TFT LCD ILI9341

Amitabh-wq, 2022-03-25:
Hi friends, can anyone help me? I'm trying to compile a sketch with the STMDUINO STM32F103(Blue pill) + TFT LCD ILI9341 through the IDE (STMduino) using the TFT_eSPI library, but it doesn't compile and shows the following error:

sketch\src\TFT_eSPI-master\Fonts\Font16.c:13:9: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'const'
PROGMEM const unsigned char widtbl_f16[96] =

Bodmer, 2022-03-25:
It looks like the library is not installed properly, path should be something like:

D:\XX\....\XX\Arduino\Sketches\libraries\TFT_eSPI/Fonts/Font16.c:13:9: error: ....

I don't get the same error with the same board, so you will need to describe your software environment fully including all board package + library version numbers.

Amitabh-wq, 2022-03-25:
Hi Bodmer, thanks for the feedback. Now I installed the TFT_eSPI V.2.4.43 library as recommended, I configured the User Setup.h and User Setup Select.h files, but other errors appear as listed below. I use the IDE V.1.8.12 on Windows 10. The board is a STM32F103C8T6 Development Module and the display is an ILI9341 2.2" SPI. This set works with the UCGLIB library, but I am wanting to migrate to your library because it is much more To test, I wrote a very simple sketch, just including the library, but it doesn't compile and shows the errors below (a part of them):
****************************************************
TEST CODE:
#include <TFT_eSPI.h>
#include <SPI.h>
TFT_eSPI tft = TFT_eSPI();
void setup() {
   tft.begin();
   tft.setRotation(1);
}
void loop() {
}
****************************************************
ERRORS:
In file included from I:\Users\Julio\Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp:24:0:

I:\Users\Julio\Documents\Arduino\libraries\TFT_eSPI\Processors/TFT_eSPI_STM32.c:16:46: error: no matching function for call to 'SPIClass::SPIClass(<anonymous enum>, <anonymous enum>, <anonymous enum>)'

     SPIClass spi(TFT_MOSI, TFT_MISO, TFT_SCLK);

                                              ^

I:\Users\Julio\Documents\Arduino\libraries\TFT_eSPI\Processors/TFT_eSPI_STM32.c:16:46: note: candidates are:

In file included from I:\Users\Julio\Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.h:32:0,

                 from I:\Users\Julio\Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp:17:

C:\Users\55159\AppData\Local\Arduino15\packages\stm32duino\hardware\STM32F1\2021.5.31\libraries\SPI\src/SPI.h:176:5: note: SPIClass::SPIClass(uint32)

     SPIClass(uint32 spiPortNumber);

     ^

C:\Users\55159\AppData\Local\Arduino15\packages\stm32duino\hardware\STM32F1\2021.5.31\libraries\SPI\src/SPI.h:176:5: note:   candidate expects 1 argument, 3 provided

C:\Users\55159\AppData\Local\Arduino15\packages\stm32duino\hardware\STM32F1\2021.5.31\libraries\SPI\src/SPI.h:170:7: note: constexpr SPIClass::SPIClass(const SPIClass&)

 class SPIClass {

       ^

C:\Users\55159\AppData\Local\Arduino15\packages\stm32duino\hardware\STM32F1\2021.5.31\libraries\SPI\src/SPI.h:170:7: note:   candidate expects 1 argument, 3 provided

C:\Users\55159\AppData\Local\Arduino15\packages\stm32duino\hardware\STM32F1\2021.5.31\libraries\SPI\src/SPI.h:170:7: note: constexpr SPIClass::SPIClass(SPIClass&&)

C:\Users\55159\AppData\Local\Arduino15\packages\stm32duino\hardware\STM32F1\2021.5.31\libraries\SPI\src/SPI.h:170:7: note:   candidate expects 1 argument, 3 provided

In file included from I:\Users\Julio\Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp:24:0:

I:\Users\Julio\Documents\Arduino\libraries\TFT_eSPI\Processors/TFT_eSPI_STM32.c:19:3: error: 'SPI_HandleTypeDef' does not name a type

   SPI_HandleTypeDef spiHal;

   ^

In file included from I:\Users\Julio\Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp:24:0:

I:\Users\Julio\Documents\Arduino\libraries\TFT_eSPI\Processors/TFT_eSPI_STM32.c: In member function 'void TFT_eSPI::pushBlock(uint16_t, uint32_t)':

I:\Users\Julio\Documents\Arduino\libraries\TFT_eSPI\Processors/TFT_eSPI_STM32.c:374:23: error: 'spiHal' was not declared in this scope

     HAL_SPI_Transmit(&spiHal, (uint8_t*)col, len<<1, HAL_MAX_DELAY);

                       ^

I:\Users\Julio\Documents\Arduino\libraries\TFT_eSPI\Processors/TFT_eSPI_STM32.c:374:54: error: 'HAL_MAX_DELAY' was not declared in this scope

     HAL_SPI_Transmit(&spiHal, (uint8_t*)col, len<<1, HAL_MAX_DELAY);

[... 54 lines cut ...]
       HAL_SPI_Transmit(&spiHal, (uint8_t*)col, BUF_SIZE<<1, HAL_MAX_DELAY);

                                                             ^

I:\Users\Julio\Documents\Arduino\libraries\TFT_eSPI\Processors/TFT_eSPI_STM32.c:399:74: error: 'HAL_SPI_Transmit' was not declared in this scope

       HAL_SPI_Transmit(&spiHal, (uint8_t*)col, BUF_SIZE<<1, HAL_MAX_DELAY);

                                                                          ^

I:\Users\Julio\Documents\Arduino\libraries\TFT_eSPI\Processors/TFT_eSPI_STM32.c:403:23: error: 'spiHal' was not declared in this scope

     HAL_SPI_Transmit(&spiHal, (uint8_t*)col, len<<1, HAL_MAX_DELAY);

                       ^

I:\Users\Julio\Documents\Arduino\libraries\TFT_eSPI\Processors/TFT_eSPI_STM32.c:403:54: error: 'HAL_MAX_DELAY' was not declared in this scope

     HAL_SPI_Transmit(&spiHal, (uint8_t*)col, len<<1, HAL_MAX_DELAY);

                                                      ^

I:\Users\Julio\Documents\Arduino\libraries\TFT_eSPI\Processors/TFT_eSPI_STM32.c:403:67: error: 'HAL_SPI_Transmit' was not declared in this scope

     HAL_SPI_Transmit(&spiHal, (uint8_t*)col, len<<1, HAL_MAX_DELAY);

Bodmer, 2022-03-25:
I can't test at the moment as I am busy, but try putting the include spi.h first in the sketch.

Amitabh-wq, 2022-03-25:
#include <SPI.h>
#include <TFT_eSPI.h>
Tested, got same error.

Bodmer, 2022-03-25:
It looks like you have an old deprecated board package loaded. Load version 2.2.0.

In the IDE Boards Manager the page should look like this:
[image]

Bodmer, 2022-03-25:
You may need to update the package path in the IDE Properites to see that package.

Instructions here:
https://github.com/stm32duino/Arduino_Core_STM32

Amitabh-wq, 2022-03-25:
Hi Bodmer, I followed the instructions and installed the new set of boards, now I can compile the code, but the IDE doesn't show me the option to upload the code via USB (STM32duino bootloader). Until then I use the set of boards from this link (http://dan.drown.org/stm32duino/package_STM32duino_index.json), which allows uploading via USB (through the STM32duino bootloader). The problem is that I don't have the STM32CubeProgrammer programmer. Any ideas on how to resolve it?
[image]

Bodmer, 2022-03-25:
Glad to hear it compiles.

I use an ST-Link V2 module to upload and a SparkFun serial to USB adapter to monitor the serial input/output pins A9 and A10. I have not tried other methods so cannot advise.

closed 2022-03-25 by Bodmer

Amitabh-wq, 2022-03-25:
Hi Bodmer, thanks. Do you have plans to update your library to have compatibility with the boards of http://dan.drown.org/stm32duino/package_STM32duino_index.json ? This set is very popular for the Blue Pill.
For me and others, who prefer to use the STM32F Blue Pill like an ordinary Arduino, uploading the code via USB (without any other programmer), it would be very useful. Please let me know. Regards.

Bodmer, 2022-03-26:
I am not sure what you are asking for based on that link.

Maybe you wish to use this bootloader?
https://github.com/rogerclarkmelbourne/STM32duino-bootloader

In that case it would be better to update the boot loader to operate with the latest STM32 Ardiino board package.

You may be able use an STM32 with a USB bootloader and the slow Arduino SPI calls (rather than the faster HAL calls which generate the errors in the new board package) by NOT using #define STM32 in the setup file. However that is not something I have tried.

The ST-Link and USB serial adapters are cheap and reliable and avoid the complication of uploading a bootloader.

Amitabh-wq, 2022-03-27:
Hi Bodmer, thank you for your attention and technical support. Regards.

Amitabh-wq, 2022-03-28:
Hi Bodmer, just to let you know, this is my video, showing your fantastic library working very well in my project. Thanks and Regards! https://youtu.be/JoIihCyKVwY

Bodmer, 2022-03-28:
Thank you for sharing, it is great to see.