Bodmer/TFT_eSPI #1746: [Proposal]: make function height() and width() const

hasenradball, 2022-04-03:
I have a proposal to make only reading functions `const`.

See:
```

/***************************************************************************************
** Function name:           width
** Description:             Return the pixel width of display (per current rotation)
***************************************************************************************/
// Return the size of the display (per current rotation)
int16_t TFT_eSPI::width(void)
{
  if (_vpDatum) return _xWidth;
  return _width;
}

/***************************************************************************************
** Function name:           height
** Description:             Return the pixel height of display (per current rotation)
***************************************************************************************/
int16_t TFT_eSPI::height(void)
{
  if (_vpDatum) return _yHeight;
  return _height;
}
```
So please make it:

```
int16_t TFT_eSPI::width(void) const 
int16_t TFT_eSPI::height(void) const
```

Bodmer, 2022-04-03:
The value returned is not constant, the returned values are protected and the fuction is virtual so I am not sure why this is desirable. Please elaborate on the reason for the suggestion.

hasenradball, 2022-04-03:
Hi,

thanks for your feedback.
Lets assume the following (here I have made a const member function)

```
int16_t TFT_eSPI::height() const {
  if (_vpDatum) return _yHeight;
  return _height;
}
```

This means or says that calling this member function does not change the Object itself.
Means this member function is intented for returning only an internal mamber value.

This does not mean that the member (here height) is a const value.
Ok my PR is a little misleading described.

I mean make a const member function, not make the variables itself const.