Bodmer/TFT_eSPI #1751: [ILI9341]: issue drawin with FreeFonts

hasenradball, 2022-04-03:
With the latest version (PIO lib bodmer/TFT_eSPI @ ^2.4.42) I see the following Issues.

When TR_Datum: --> Issue at right side of number
When TL_Datum: --> Cuttet right side of number

[image]

Is there some bug due to new functions?
I havent seen with older versions there.

Code look like:
```
 /* ### Data ### */
        void showTemperature(float temperatur, const cursor_pos &pos, uint16_t color_txt, uint16_t color_bg) {
            tft.setTextDatum(TR_DATUM);
            tft.setFreeFont(&FreeSansOblique18pt7b);
            int padding = tft.textWidth("-125.2", GFXFF);
            tft.setTextColor(color_txt, color_bg);
            tft.setTextPadding(padding);
            tft.drawFloat(temperatur, 1, pos.x + padding, pos.y, GFXFF);
        }

        void showHumidity(float humidity, const cursor_pos &pos, uint16_t color_txt, uint16_t color_bg) {
            tft.setTextDatum(TL_DATUM);
            tft.setFreeFont(&FreeSansOblique18pt7b);
            int padding = tft.textWidth("100.0", GFXFF);
            tft.setTextColor(color_txt, color_bg);
            tft.setTextPadding(padding);
            if ((humidity < 40) || (humidity > 60)) {
                tft.setTextColor(ILI9341_RED, color_bg);
            }
            else {
                tft.setTextColor(color_txt, color_bg);
            }
            tft.drawFloat(humidity, 1, pos.x, pos.y, GFXFF);
        }

        void showAbsHumidity(float temperature, float humidity, const cursor_pos &pos, uint16_t color_txt, uint16_t color_bg) {
            tft.setTextDatum(TL_DATUM);
            tft.setFreeFont(&FreeSansOblique12pt7b);
            float absHum {6.112F * 2.1674F * humidity * exp((17.67F * temperature) / (temperature + 243.5F)) / (273.15F + temperature)};
            int padding = tft.textWidth("100.0", GFXFF);
            tft.setTextColor(color_txt, color_bg);
            tft.setTextPadding(padding);
            tft.drawFloat(absHum, 1, pos.x, pos.y, GFXFF);
        }
```

hasenradball, 2022-04-04:
Hi,

with the font `FreeSans24pt7b` everything is fine.
Maybee an issue with the Oblique fonts...

Bodmer, 2022-04-07:
The problem occurs because some glyphs in oblique fonts (e.g. _"7"_ ) have pixels positions that extend beyond the next cursor position. Proportional fonts have different character widths so over-printing does not always obliterate the previous characters.
The text padding feature was supposed to solve this but does not function correctly for some oblique characters when the text datum is set to the right side.
The library also has a special rendering feature  for digits to avoid distracting position jiggle of digits as the value over-printed changes, this uses the cursor spacing to position numeric glyphs.

The workaround is to avoid oblique fonts for numeric over-printing.

closed 2022-04-07 by Bodmer