Bodmer/TFT_eSPI #1753: 'class TFT_eSPI' has no member named 'getTouch'

zaydsaoudi, 2022-04-04:
Hi, i need help here , a bought esp32 and tft uno shield 2.8 , first thing I can't display anything with isp display avter that i edited user_setup.h and add parallel , after that everything goes well until i cant make screen touch with tft_esp .
I need help here please

elgerg, 2022-04-04:
See:
https://github.com/Bodmer/TFT_eSPI/issues/1408

Unless it's a parallel screen in which case:
https://github.com/Bodmer/TFT_eSPI/issues/1494

You might want to search for your issue next time.

closed 2022-04-04 by Bodmer

elgerg, 2022-04-06:
This library doesnt support touch on a parallel display. As mentioned here:
https://github.com/Bodmer/TFT_eSPI#8-bit-parallel-supportset
There is a link to a library that you can use instead.