Bodmer/TFT_eSPI #1757: espressif boards manager  version 2.0.3-RC1

xinkiknix, 2022-04-06:
please check: 

https://github.com/espressif/arduino-esp32/issues/6539#issuecomment-1089226724

Bodmer, 2022-04-06:
Work around at the moment is to use 2.0.2

When the HAL API calls stabilise I will update!

Bodmer, 2022-04-07:
At the moment neither the C3 or S3 is supported. The C2 is supported but not with v2.0.3 board package. I do not have a C3 or S3 chip at the moment to work on adding support. A pull request with fully tested updates would be appreciated.

Bodmer, 2022-04-07:
I think this is worth keeping open at the moment to make it more visible to library users.

Bodmer, 2022-04-28:
Library has been updated to support S2, C3 and S3. Note that due to ongoing low level/IDF bug fixes with the ESP32 board package it is possible support will break and another update will be required!

closed 2022-04-28 by Bodmer