Bodmer/TFT_eSPI #1764: PlatformIO Registry: Broken manifest

ivankravets, 2022-04-09:
Hi,

Could we ask you to fix https://github.com/Bodmer/TFT_eSPI/blob/master/library.json ?

https://registry.platformio.org/libraries/bodmer/TFT_eSPI

# Log

ManifestValidationError(
platformio.package.exception.ManifestValidationError: Invalid manifest fields: {'keywords': {5: ['Only [a-z0-9-+. ] chars are allowed'], 8: ['Only [a-z0-9-+. ] chars are allowed']}}. 
Please check specification -> https://docs.platformio.org/page/librarymanager/config.html

closed 2022-04-10 by Bodmer

Bodmer, 2022-04-10:
Sorted, thanks for pointing this out.

ivankravets, 2022-04-10:
Sorry that disturb you, you can validate manifest with [pio pkg pack](https://docs.platformio.org/en/latest/core/userguide/pkg/cmd_pack.html) command. See

```
pio pkg pack -o /tmp/test.tar.gz ~/Downloads/TFT_eSPI-master.zip
Error: Invalid manifest fields: {'keywords': {5: ['Only [a-z0-9-+. ] chars are allowed']}}.
Please check specification -> https://docs.platformio.org/page/librarymanager/config.html
```

The `WT32_SC01` keyword uses not allowed underscore.

commit e52a0161 (2022-04-10): Fix #1760, fix #1763, fix #1764
commit 8f0f8e51 (2022-04-10): Fix #1764