Bodmer/TFT_eSPI #1767: [   707][E][esp32-hal-gpio.c:95] __pinMode(): Invalid pin selected

homonto, 2022-04-10:
hi Bodmer,

My pins are defined in custom config this way:

```
#define ST7789_DRIVER      // Full configuration option, define additional parameters below for this display
#define TFT_WIDTH  240 // ST7789 240 x 240 and 240 x 320
#define TFT_HEIGHT 240 // ST7789 240 x 240
#define TFT_CS   -1  // Chip select control pin, use -1 if not in use
#define TFT_RST   2 
#define TFT_DC    3  
#define TFT_MOSI 4
#define TFT_SCLK 5
#define TFT_MISO -1  // use -1 if not in use
```

Whenever I start the program it shows me this error:
12:46:39.679 -> [   707][E][esp32-hal-gpio.c:95] __pinMode(): Invalid pin selected
12:46:39.679 -> E (271) gpio: gpio_set_level(226): GPIO output gpio_num error
12:46:39.679 -> [   707][E][esp32-hal-spi.c:208] spiAttachMISO(): HSPI Does not have default pins on ESP32S2!
12:46:39.716 -> [   715][E][esp32-hal-gpio.c:95] __pinMode(): Invalid pin selected
12:46:39.716 -> E (285) gpio: gpio_set_level(226): GPIO output gpio_num error

Funny thing: all works.

I narrowed down that this message comes from:  esp32-hal-spi.c - it is repeated there many times - see here: https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-spi.c

the moment I change "-1" in unused pins (TFT_CS and TFT_MISO) and put i.e. "40" (that does not exist btw on ESP32-S2), there is no more error

Is that a bug or so? 
Apparently "-1" is not the best option if one wants to have clean code without warnings

thx

Bodmer, 2022-04-10:
In the Arduino world an I/O pin number of -1 is OK to signify to a function that a pin allocation is not wanted, but for low level calls it can throw a warning.
It is odd that an apparently illegal pin number does not generate an error!
Expressif seem to have put a new software team in place and this has lead to some negative impact on existing ESP32 libraries. Software engineers love to change things their predecessors did and make it "better"! I expect this will sort itself out eventually! In the meantime your workaround will be needed.

homonto, 2022-04-10:
I tried to remove the number completely but then, many functions get crazy (like: "give me the number or so!") 
I started with number 100 but then - not working
40 seems ok ;-)
I would give any legal number but I have one project with... all pins taken so I tried 40 and it ... looks ok ;-)

Bodmer, 2022-04-10:
The SPI port can operate in half duplex mode so MOSI and MISO are on same pin. So maybe try setting MISO and MOSI to pin 4.

Bodmer, 2022-04-20:
@homonto, I am thinking about assigning TFT_MISO = TFT_MOSI  if TFT_MISO is not defined. Can you test that works OK in your setup.

homonto, 2022-04-20:
@Bodmer  just tell me what to do please: simply do it this way:? Compile and report back? ;-) I can do it for both ILI9341 320x240 and ST7789 240x240
```
#define TFT_MOSI 4 //used
#define TFT_MISO 4 //fake
```

Bodmer, 2022-04-20:
Yes, that is exactly right,  thanks.

Bodmer, 2022-04-20:
This link is interesting:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/api-reference/peripherals/gpio.html#:~:text=The%20ESP32%2DS2%20chip%20features,GPIO21%20and%20GPIO26%20~%20GPIO46).

Looks like valid GPIO for the ESP32 S2 are 0-21 and 26-46.

homonto, 2022-04-20:
library: 2.4.45, ILI9341:  before changes (MOSI=4, MISO=40):

[image]

after changes (both are 4):
[image]

the only difference I see is my hand is shaking more "after" ;-)
should I check ST7789 as well?

homonto, 2022-04-20:
> This link is interesting: https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/api-reference/peripherals/gpio.html#:~:text=The%20ESP32%2DS2%20chip%20features,GPIO21%20and%20GPIO26%20~%20GPIO46).
> 
> Looks like valid GPIO for the ESP32 S2 are 0-21 and 26-46.

`GPIO40 ->JTAG`

that is why, when I used MISO=40 it worked as I am not using JTAG:

Bodmer, 2022-04-20:
All looks good, no more testing needed. Thank you.
I like your display arrangement! Looks good.

homonto, 2022-04-20:
topic closed I think as solution provided ;-)
I am sure in next version you will say what exactly is to be done: keep "-1" and you redo your part or something else ;-)

closed 2022-04-20 by homonto

Bodmer, 2022-04-20:
Great, thanks again.

Bodmer, 2022-04-20:
Master repository updated.

dejavu1987, 2023-03-26:
I have tried several different pins on my touch display and all of them have worked, I finally decided to solder everything and now I am getting that error, it seem to be running fine as when I touch on the screen It is logging the coordinates just the display is black, don't know if its just the LED / BL pin that is coincidentally connected to a READ ONLY PIN. FML
```
#define TFT_MISO 33 
#define TFT_MOSI 27
#define TFT_SCLK 12
#define TFT_CS 4 
#define TFT_RST 32
#define TFT_DC 25
#define TFT_BL 35 

#define TOUCH_CS 26 
```

dejavu1987, 2023-03-26:
Indeed.
[image]
[image]

😮‍💨

CForst1965, 2024-11-25:
Yes, use a spare unused pin -1 doesn't work for the SPI

commit e4778d77 (2022-04-20): Fix #1767
commit d6544c9c (2022-04-21): Fix #1767 and update ILI9481 init code option 8