Bodmer/TFT_eSPI #1772: ESP32 S3 support?

richcj10, 2022-04-12:
Is S3 support coming?

Bodmer, 2022-04-12:
I do intend to add S3 support, but prices of boards are rather high and I can't justify buying more boards at the moment. The S2 is supported.

I see that v2.0.3 is on Github but it has not been released yet and appears to be going through some detailed changes and bug fixes.

closed 2022-04-12 by Bodmer

Miraculix200, 2022-04-12:
It's a release candidate. I doubt there will be much changes (came here because I just tested your library with ESP32-S3 and failed to compile it with both #define USE_HSPI_PORT and without). 

If you have a soldering iron you could try my ESP32-S3 PCB design and solder a 4$ WROOM module from aliexpress on it. Would be less expensive than the dev boards (maybe 15-20$ with all the SMD parts/connector, at least if JLCPCB shipping to your country is cheap): https://oshwlab.com/Miraculix200/esp32-s3-mini_copy

Though I fully understand you don't want to buy every possible MCU.

DustinWatts, 2022-04-18:
Can I donate an ESP32 S3 board for you to test? I have an ESP32-S3-DevKitC-1-N8R2 and a ESP32-S3-DevKitC-1-N8.

Bodmer, 2022-04-18:
Hi Dustin, in a moment of weakness I ordered one of these boards earlier today:
https://shop.pimoroni.com/products/pros3-esp32-s3?variant=39762607210579

The aim is to use it with a Node-RED system as a battery/solar powered sensor node. But I can use it to try changes to TFT_eSPI. Hopefully it will not take too much effort as Summer is approaching and software activities tend to get shelved then!

I have some of your ESP32 and Pico TFT support boards from PCBWay, they are handy to build test setups and for projects. Incidentally, I find the backlight LED can be dim on 3.3V so I did some surgery on the boards to run the display from 5V. Not all displays have a 5V regulator though. Maybe add a link option to the next PCB? Connecting up MISO would be useful too (as a link option to support boards that do not tristate SDO).

elgerg, 2022-04-19:
@Bodmer can you share pictures of your DIY efforts? I also found that the displays were a little dim when using the boards. I did want to add a transistor to allow dimming via one of the pins.. Thanks!

Bodmer, 2022-04-19:
@elgerg. I assume that you are interested in the Pico board mods since Dustin's ESP32 board does have a GPIO pin connected to the backlight LED control pin. Many display boardss already have a transistor connected to this pin, typically labelled Q1 on the PCB.

The 5V supply hack for the Pico board involved cutting the 3.3V track and linking in the 5V pin to that PCB track instead:
[image]
Make sure the track is cut well and totally disconnected from 3.3V otherwise the hack will put 5V on the 3.3V rail and destroy things!

I did not mod the Pico board to drive the LED pin from a GPIO, but here is what you need to do. The Pico board links power to the LED control pin so you need to cut the track indicated below, scrape off the solder resist from the track stub and wire link the display LED pin to a free RP2040 Pico GPIO pin:

[image]

Bodmer, 2022-04-19:
For Dustin's ESP32 board, this is the hack to drive the display with 5V instead of 3.3V. Again the 3.3V power feed track is cut and the display track stub is linked to 5V. Make sure the display is fitted with a 5V to 3.3V regulator (tyically labelled U1) and that it has not been bypassed (typically the bypass link is labelled J1)!
[image]

Typical display circuit.  Q1 is LED backlight control transistor, cnnected via R6 to display connector LED pin.. U1 is 5V to 3.3V regulator. J1 is regulator bypass link position. U2 is touch controller.
[image]

DustinWatts, 2022-04-20:
> Hi Dustin, in a moment of weakness I ordered one of these boards earlier today: https://shop.pimoroni.com/products/pros3-esp32-s3?variant=39762607210579
>
> The aim is to use it with a Node-RED system [...]

Links are always a good idea to make boards more flexible. Thanks for your input. Will be in the next release :)

Bodmer, 2022-04-22:
There is a branch here to try with the ESP32 S3:
https://github.com/Bodmer/TFT_eSPI/tree/ESP32_S3

This is the setup I used to test  with board package 2.0.3-RC1 and ILI9341:
https://github.com/Bodmer/TFT_eSPI/blob/ESP32_S3/User_Setups/Setup70b_ESP32_S3_ILI9341.h

DMA is not working on the ESP32 S3 and it may be some time before I look at that.

DustinWatts, 2022-04-22:
Nice! I'll give it a try this weekend!

Bodmer, 2022-04-22:
I have tested with ILI9488 (not touch though) and found a problem. This has just been corrected in the branch.

richcj10, 2022-04-23:
What example would be best to test this?

Bodmer, 2022-04-23:
Most examples should run excet those using DMA and those that read from the TFT.
It depends what you wish to test:
1 To test performance of colour block rendering and demo basic graphics fns: TFT_graphicstest_one_lib
2. Single pixel performance: TFT_Starfield
3. Correctness of colours: Colour_Test
4. To test image rendering and correctness of colour gamma: TFT_Rainbow_one_lib and alphaBlend_Test
5. To test touch: Test_Touch_Controller

richcj10, 2022-04-23:
Thanks, I have it running.

richcj10, 2022-04-23:
I did have a problem with user defined  file you recommended. I used instead is 
#define TFT_CS   10 //     10 or 34

#define TFT_MOSI 11 //     11 or 35
#define TFT_SCLK 12 //     12 or 36
#define TFT_MISO 13 //     13 or 37

Miraculix200, 2022-04-23:
11, 12 and 13 are the default pins for SPI in arduino-esp32 (though when they started the esp32-s3 repository, it was 35,36,37). I can also confirm it works for me

Bodmer, 2022-04-23:
Thanks for letting me know, it's good to hear other users have it working.

Yes, pins will have to be adapted depending on the board at the moment. I am using the[ UM ProS3 ](https://images.squarespace-cdn.com/content/v1/5a8cc639a803bbfaa029615a/b130b68e-03c8-427f-84e3-bfecd8d28102/pins_pros3.jpg?format=1500w)which has GPIO 10 allocated to measure battery voltage and pin 11 is not available.

The default SPI pins are defined when the correct board is selected in the Arduino IDE "Board:" menu, so check the boards documentation.

DustinWatts, 2022-04-24:
> I have tested with ILI9488 (not touch though) and found a problem. This has just been corrected in the branch.

I can confirm that the on the EP32-S3 DevKitC the ILI9488 working! Using this branch: https://github.com/Bodmer/TFT_eSPI/tree/ESP32_S3 and the Arduino ESP32 core 2.0.3-RC1.

Bodmer, 2022-04-24:
Hi Dustin, great! Raise a new issue if you find any problems that need to be solved.

Bodmer, 2022-04-25:
The S3 support has been merged into the master.

DustinWatts, 2022-04-25:
> The S3 support has been merged into the master.

Would be great if there were parallel support!

HabobB6, 2025-09-13:
@Bodmer we need support for c6 also