Bodmer/TFT_eSPI #1773: Issue with parallel ST7789 LCD with Pi Pico

DottiestDuke47, 2022-04-13:
Hi, I know that parallel is untested as of yet on the pi pico, but I still tried to setup my LCD with parallel. It compiles and flashes fine but the LCD output is garbled.

My config is as follows:
`
#define ST7789_DRIVER     // Configure all registers

#define TFT_WIDTH  240
#define TFT_HEIGHT 280

#define CGRAM_OFFSET      // Library will add offsets required

#define TFT_PARALLEL_8_BIT

// The ESP32 and TFT the pins used for testing are:
#define TFT_CS  12  // Chip select control pin (library pulls permanently low
#define TFT_DC  11  // Data Command control pin - must use a pin in the range 0-31
#define TFT_RST 13  // Reset pin, toggles on startup

#define TFT_WR  10  // Write strobe control pin - must use a pin in the range 0-31
#define TFT_RD  14  // Read strobe control pin

#define TFT_D0   2  // Must use pins in the range 0-31 for the data bus
#define TFT_D1   3  // so a single register write sets/clears all bits.
#define TFT_D2   4  // Pins can be randomly assigned, this does not affect
#define TFT_D3   5  // TFT screen update performance.
#define TFT_D4   6
#define TFT_D5   7
#define TFT_D6   8
#define TFT_D7   9

#define LOAD_GLCD   // Font 1. Original Adafruit 8 pixel font needs ~1820 bytes in FLASH
#define LOAD_FONT2  // Font 2. Small 16 pixel high font, needs ~3534 bytes in FLASH, 96 characters
#define LOAD_FONT4  // Font 4. Medium 26 pixel high font, needs ~5848 bytes in FLASH, 96 characters
#define LOAD_FONT6  // Font 6. Large 48 pixel font, needs ~2666 bytes in FLASH, only characters 1234567890:-.apm
#define LOAD_FONT7  // Font 7. 7 segment 48 pixel font, needs ~2438 bytes in FLASH, only characters 1234567890:-.
#define LOAD_FONT8  // Font 8. Large 75 pixel font needs ~3256 bytes in FLASH, only characters 1234567890:-.
//#define LOAD_FONT8N // Font 8. Alternative to Font 8 above, slightly narrower, so 3 digits fit a 160 pixel TFT
#define LOAD_GFXFF  // FreeFonts. Include access to the 48 Adafruit_GFX free fonts FF1 to FF48 and custom fonts

// Comment out the #define below to stop the SPIFFS filing system and smooth font code being loaded
// this will save ~20kbytes of FLASH
#define SMOOTH_FONT
`

DottiestDuke47, 2022-04-13:
[image]

DottiestDuke47, 2022-04-13:
Most of the screen is gibberish but a few lines can be seen moving

DottiestDuke47, 2022-04-13:
Using an incorrect driver yields some different results but still unusable[image]

DottiestDuke47, 2022-04-13:
[image]
the datasheet picture for this LCD

Bodmer, 2022-04-13:
The 8 bit parallel mode on the RP2040 has been tested with a few display types and works fine. However I have not tested it with the ST7789 chip. 

I suspect the interface may be running too fast for that chip but this can be slowed down. To do this you need to edit this line:
https://github.com/Bodmer/TFT_eSPI/blob/master/Processors/TFT_eSPI_RP2040.h#L89

Change the 1 to a 4. If that works you can try 2.  Save the file and compile+upload again.

DottiestDuke47, 2022-04-13:
[image]

That seems to have fixed the issue, Thank you very much for your help, I began spending a crazy amount of time triple checking my wiring 😂

DottiestDuke47, 2022-04-13:
```
  #if defined (ILI9481_DRIVER)
    #define DIV_UNITS 1
    #define DIV_FRACT 160
  #elif defined (ST7789_DRIVER) //ST7789 needs to run at a slower speed
    #define DIV_UNITS 4
    #define DIV_FRACT 0
  #else
    #define DIV_UNITS 1
    #define DIV_FRACT 0
  #endif
```

Is it ok if i make a pull request for this, bear in mind it would be my first one.

Bodmer, 2022-04-13:
What setting worked? I think a setting of 2 should be OK according to the datasheet timing values. The lower the setting, the faster the interface runs.

Bodmer, 2022-04-13:
It is a simple change so no pull request is needed. I will make it user configurable too.

DottiestDuke47, 2022-04-13:
stepped to 3 then 2 and can confirm they work fine

Bodmer, 2022-04-13:
OK, thanks. If you overclock the processor then the TFT interface speed increases pro rata, so bear that in mind.

DottiestDuke47, 2022-04-13:
I've just overclocked the pico to 250MHz, and this is amazing! any way I can suport/buy you a coffee?

Bodmer, 2022-04-13:
Happy to help. Renumeration not required!
250MHz may be unreliable but 200MHz seems very stable on my setups.

I am surprised the display copes, 250MHz with a divider of 2 is the same rate as 125MHz and divider of 1. Timing must be right on the edge of whats possible with the ST7789.

Bodmer, 2022-04-13:
Ah, makes sense now. I can add another display to the tested list now.

closed 2022-04-15 by DottiestDuke47