Bodmer/TFT_eSPI #1774: STM32F103C8T6 and TFT ILI9481 IPS 3.5in 320x480 not working (black screen)

Amitabh-wq, 2022-04-13:
Hello Bodmer.
I'm trying to run the TFT_eSPI examples on a TFT ILI9481 IPS 3.5in 320x480 display SPI interface with the STM32F103C8T6 MCU, but I only get a black screen. I did all the configuration files (User_Setup_Select and User_Setup and pin and driver selection) and wirings accordingly. The compilation and upload went without errors but the screen stays black (only the backlight lit). If you can give me any tips I would be very grateful!

Bodmer, 2022-04-13:
The ILI9481 is slow so try 10MHz SPI clock rate to start with. Run the test and diagnostics example and post the serial output here.

Amitabh-wq, 2022-04-13:
Actually the clock freq measured on pin SCK is 9MHz (not 27MHz), even so it didn't work.
[image]

Bodmer, 2022-04-13:
Try the ILI9486 driver. The vendor does not always know what they are selling. Post a picture of the back of the display board.

Amitabh-wq, 2022-04-13:
Here it is
[image]
:

Amitabh-wq, 2022-04-14:
The display worked using the ILI9486 driver (instead of the ILI9481 as specified by seler on Aliexpress). However it has quality problems, see the videos below, artifacts appear (lines and vertical ripples) and the colors are out of standard. And it only works with 9MHz SPI clock, I tried to increase this frequency and it doesn't work with higher frequencies. Could you give me a hint on how to solve these problems? Thank you!

https://user-images.githubusercontent.com/89787514/163292739-dae5193f-d996-4908-b9a1-e9c8413181f6.mp4

https://user-images.githubusercontent.com/89787514/163292932-0278392f-51a0-4138-b9c8-868864300e2b.mp4

!

Bodmer, 2022-04-14:
There are a number of ILI9481 init sequences that can be invoked, [as detailed here](https://github.com/Bodmer/TFT_eSPI/blob/master/TFT_Drivers/ILI9481_Init.h#L7-L14). You could try each init sequence and see if there is one that works better. You must only enable 1 init sequence at a time.

Amitabh-wq, 2022-04-14:
> There are a number of ILI9481 init sequences that can be invoked, [as detailed here](https://github.com/Bodmer/TFT_eSPI/blob/master/TFT_Drivers/ILI9481_Init.h#L7-L14). You could try each init sequence and see if there is one that works better. You must only enable 1 init sequence at a time.

Hi Bodmer, I tested all 6 options with the ILI9481 driver, unfortunately none work. It works only with the ILI9486 and with some graphic issues. Any idea?

Bodmer, 2022-04-14:
OK, it looks like it is not an ILI9481 or ILI9486 display.

Some displays that physically look like yours are ILI9488, so try that driver.

Amitabh-wq, 2022-04-15:
> OK, it looks like it is not an ILI9481 or ILI9486 display.
> 
> Some displays that physically look like yours are ILI9488, so try that driver.

I redid all tests with the ILI9488 driver, but unfortunately it doesn't work (black screen). The only compatible driver is ILI9486, but with problems. I don't know what else to do :(

Bodmer, 2022-04-15:
It looks like some of your connection from the processor to display are very long. That will affect signal integrity. Inn particular the 0V (GND)  connection must be short to prevent ground bounce.

Amitabh-wq, 2022-04-15:
> It looks like some of your connection from the processor to display are very long. That will affect signal integrity. Inn particular the 0V (GND) connection must be short to prevent ground bounce.

I use the same setup for a 3.2" ILI9341, which works very well. I will try to see if I can get some improvemet. Thanks.

Amitabh-wq, 2022-04-15:
It didn't solve the problem. I'm contacting the seller of the ILI9481 display to see if he can help me resolve this issue.

Bodmer, 2022-04-15:
The only other I can think might be the problem is that the time gap between the DC line changes and the SPI data transfer is too small. The display works OK on an ESP32 but that processor is quite slow at toggling GPIO (50ns between changes). I can create a slower STM32.h file to try if you are interested.

Bodmer, 2022-04-15:
Here is a replacement file for the one [in the library here](https://github.com/Bodmer/TFT_eSPI/blob/master/Processors/TFT_eSPI_STM32.h).
[TFT_eSPI_STM32.zip](https://github.com/Bodmer/TFT_eSPI/files/8496800/TFT_eSPI_STM32.zip)

I have slowed down both the TFT_CS and the TFT_DC line at lines 231 and 249 by using the slower Arduino digitalWrite().

Let me know if that does the trick with the ILI9486 driver.

Amitabh-wq, 2022-04-15:
> Here is a replacement file for the one [in the library here](https://github.com/Bodmer/TFT_eSPI/blob/master/Processors/TFT_eSPI_STM32.h). [...]

Hi Bodmer, thanks for your attention. i'll try it later today.

Amitabh-wq, 2022-04-15:
I did new tests using ILI9486, replaces the TFT_eSPI_STM32.h file in the library, as indicated, but unfortunately it didn't solve the problem, the lines and vertical ripples still continue. I also tested the ILI9481 but it didn't work.

Amitabh-wq, 2022-04-16:
Hi Bodmer, The seller of the TFT display ILI9481 (9486?) sent me the attached demo files. Please could you take a look at these files, maybe you can check some configuration that works in my case. I don't know how to use these files. Thank you!
[wetransfer_3-5-ips-ili9481-demo_stm32f103rct6_software_spi-zip_2022-04-16_0451.zip](https://github.com/Bodmer/TFT_eSPI/files/8499971/wetransfer_3-5-ips-ili9481-demo_stm32f103rct6_software_spi-zip_2022-04-16_0451.zip)

Amitabh-wq, 2022-04-16:
Hi Bodmer, The seller of the TFT display ILI9481 (ILI9486?) sent me the attached demo files. Please could you take a look at these files, maybe you can check some configuration that works in my case. I don't know how to use these files. Thank you!
[3-5-ips-ili9481-demo_stm32f103rct6_software_spi-zip_2022-04-16_0451.zip](https://github.com/Bodmer/TFT_eSPI/files/8499997/3-5-ips-ili9481-demo_stm32f103rct6_software_spi-zip_2022-04-16_0451.zip)

Amitabh-wq, 2022-04-16:
Hi Bodmer, The seller of the TFT display ILI9481 (ILI9486?) sent me the attached demo files. Please could you take a look at these files, maybe you can check some configuration that works in my case. I don't know how to use these files. Thank you!
[3-5-ips-ili9481-demo_stm32f103rct6_software_spi-zip_2022-04-16_0451.zip](https://github.com/Bodmer/TFT_eSPI/files/8500055/3-5-ips-ili9481-demo_stm32f103rct6_software_spi-zip_2022-04-16_0451.zip)

Bodmer, 2022-04-16:
The vendor has sent a lot of code! The PDF in the zip file is for an ILI9488 based diplay.

The lcd software I found buried inside the archive includes 2 init sequences for an ILI9481 display. For some reason both the init sequences are run! all rather peculiar.

I have put the 2 init sequences provided by the vendor in the ILI9481 init file below and enabled both of them to run (init 7 and init 8). No idea if this is correct for your display but maybe worth a try.
[ILI9481_Init.zip](https://github.com/Bodmer/TFT_eSPI/files/8500677/ILI9481_Init.zip)

It's a replacement for this file in the library:
https://github.com/Bodmer/TFT_eSPI/blob/master/TFT_Drivers/ILI9481_Init.h

Other folk seem to have trouble with these ILI9481 diplays, see [here ](https://github.com/Bodmer/TFT_eSPI/issues/1056)and [here](https://www.youtube.com/watch?v=sIhnILBBUdI).
I am running out of ideas.  If you have an ESP32 then you could try it on that processors as others seem to have got an identical looking display running on that processors

Amitabh-wq, 2022-04-17:
Hi Bodmer, I ran new tests and got promising results:

1-I tested the same display with ESP32 and it worked perfectly with the ILI9481 driver (and in the ILI9486, but with graphical problems and inversions), I used the init "#define ILI9481_INIT_2 // CPT29 - perfect with ESP32" which got perfect colors and images.  So the display really is an ILI9481.

2-I tested the same display with STMP32F using the new "ILI9481_Init" file that you generated, and I selected the init "#define ILI9481_INIT_8 // From #1774" and it worked with the ILI9481 driver, but with slightly distorted colors ( hue) and a slight vertical ripple.

My conclusion is that the problem with ILI9481 is in the STM32F processor configuration, that needs to be improved. See the photos below.

-ESP32 USING ILI9486 DRIVER:
[image]

-ESP32 USING ILI9481 DRIVER:
[image]

-STM32F USING ILI9481 DRIVER:
[image]

Bodmer, 2022-04-17:
There is a Colour Test example in the Test and Diagnostics folder that does colour inversion tests. I expect INIT_8 supplied by the vendor of your display has the wrong invert setting.

I am surprised that you cannot get INIT_2 running on the STM32, but I see you have already tried that earlier and you got no display... that is a surprise because all that is different between INIT_8 and INIT_2 is the register values, the timing is the same. So if INIT_8 loads then so should INIT_2.

The tests suggest the timing is still marginal on the STM32. Maybe it needs to be even slower. I attach below another slowed down test file for the one [here](https://github.com/Bodmer/TFT_eSPI/blob/master/Processors/TFT_eSPI_STM32.h), this has DC and CS slowed even more.
[TFT_eSPI_STM32.zip](https://github.com/Bodmer/TFT_eSPI/files/8502283/TFT_eSPI_STM32.zip)

The STM32 board you have has quite a weak regulator and the display is designed to be powered from 5V (a regulator is fitted to drop the 5V to 3.3V). So that may be the cause of the ripple effect. However using the wrong init code can also cause this effect as various internal display voltages are set by the init code.

Amitabh-wq, 2022-04-17:
-Including the "tft.invertDisplay( false );" the display shows the correct colors, this proves that it is inverted (negative).
-Unsing the new slowed down test file: the same, it works only with the ILI9481_INIT_8 (none of others worked - no display) as previous tests.
-I am powering the display with 5 volts (the display can be powered with 5/3.3v).

Bodmer, 2022-04-19:
I will close as you seem to have it working now. It appears that no modification to the TFT_eSPI library is needed and that the potential speed problem was not the issue.

closed 2022-04-19 by Bodmer

Amitabh-wq, 2022-04-20:
Hi Bodmer, is it possible you do a change in the "ILI9481_Init.h" file in order to get non inverted colors (positive instead negative) as default, not being necessary to use the  "tft.invertDisplay( false );" in the sketch? Thanks

Bodmer, 2022-04-20:
Yes, which init sequence option?

Amitabh-wq, 2022-04-20:
> Yes, which init sequence option?

Hi, it is the INIT_8, because it is the only one tha is working with the ILI9481. Thanks.

Bodmer, 2022-04-20:
I have updated in the Github master.

Amitabh-wq, 2022-04-21:
> I have updated in the Github master.

Hi Bodmer, thanks for the support!