Bodmer/TFT_eSPI #1776: setBitmapColor() does not work anymore for 1 bit sprite

alf45tar, 2022-04-14:
The following code always display sprite in black and white with the latest version of the library.

        TFT_eSprite sprite = TFT_eSprite(&display);

        sprite.setColorDepth(1);
        sprite.createSprite(display.width(), display.height() - 30);
        sprite.setFreeFont(&DSEG14_Classic_Bold_100);
        switch (currentProfile) {
          case 0:
            p = "A." + ((currentBank > 9  ? "" : "0") + String(currentBank));
            sprite.setBitmapColor(TFT_RED, TFT_BLACK);
            break;
          case 1:
            p = "B." + ((currentBank > 9  ? "" : "0") + String(currentBank));
            sprite.setBitmapColor(TFT_GREEN, TFT_BLACK);
            break;
          case 2:
            p = "C." + ((currentBank > 9  ? "" : "0") + String(currentBank));
            sprite.setBitmapColor(TFT_BLUE, TFT_BLACK);
            break;
        }
        sprite.setTextDatum(MC_DATUM);
        sprite.drawString(p, sprite.width() / 2, sprite.height() / 2);
        sprite.pushSprite(0, 30);
        sprite.deleteSprite();

Bodmer, 2022-04-14:
Can you try the "Sprite_scroll_1bit" example an tell me if it is in just black and white. Thanks.

Bodmer, 2022-04-14:
Just tested the above exmaple and it plots the 1 bit sprite in different colours.

However the "One_bit_Sprite_Demo" does only plot back and white.

I will investigate.

Bodmer, 2022-04-14:
I will fix this. In the meantime the work-around is to always use the tft instance, e.g.  tft.setBitmapColor(TFT_GREEN, TFT_BLACK);

Bodmer, 2022-04-14:
Fixed in Github master.

commit f790c868 (2022-04-14): Fix #1776
commit d0494af0 (2022-04-14): Fix #1776