Bodmer/TFT_eSPI #1784: exit status 1 'class TFT_eSPI' has no member named 'initDMA'; did you mean 'init'?

iwordly, 2022-04-19:
`#pragma GCC optimize("O3")
 
 
#include <Arduino.h>
 
#include <TFT_eSPI.h>
 
#include <Wire.h>
 
 
TFT_eSPI TFT = TFT_eSPI();
 
TFT_eSprite eSprite = TFT_eSprite(&TFT);
 
void setup() {
 
    TFT.init();
 
    TFT.setRotation(1);
 
    TFT.fillScreen(TFT_BLUE);
 
    TFT.initDMA();
 
    eSprite.setTextFont(1);
 
    eSprite.setTextColor(TFT_WHITE);
 
    eSprite.createSprite(240, 240);
 
}
 
 
void loop() {
 
    eSprite.setTextFont(1);
 
    eSprite.setCursor(20, 20);
 
    eSprite.println("Hello World\n");
 
   
 
    eSprite.setCursor(40, 40);
 
    eSprite.println("Hello World\n");
 
    eSprite.pushSprite(0, 0);
 
 
    delay(1000);
 
}`

output:
`exit status 1
'class TFT_eSPI' has no member named 'initDMA'; did you mean 'init'?`

Bodmer, 2022-04-19:
Which processor and display interface are you using?
Write only DMA is supported on ESP32, STM32 and RP2040 with SPI interface. Parallel interface DMA is only supported on RP2040.

iwordly, 2022-04-19:
ok think you

Bodmer, 2022-04-19:
I have updated the ReadMe to make this clear.

closed 2022-04-19 by Bodmer