Bodmer/TFT_eSPI #1787: Can't drive ST7789 Washare 1.3inch LCD Display

Technerder, 2022-04-21:
I've hooked up my Waveshare display like shown below but nothing seems to be appearing on the display. The module does have the backlight on though. What am I doing wrong?

[image]

Wiring (Display -> ESP32-C3):

> VCC -> 5V
> GND -> GND
> DIN -> 10
> CLK ->18
> CS -> 19
> DC -> 1
> RST -> 2
> BL -> 3

Display: [Waveshare 1.3inch LCD display Module](https://www.waveshare.com/product/displays/lcd-oled/lcd-oled-3/1.3inch-lcd-module.htm)
Board: [ESP32-C3-DevKitC-02](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/hw-reference/esp32c3/user-guide-devkitc-02.html)

Code:
```c
#include <TFT_eSPI.h>
#include <SPI.h>

TFT_eSPI tft = TFT_eSPI();

void setup(void) {
  tft.init();
  tft.setRotation(1);
  tft.fillScreen(TFT_BLACK);
}

void loop() {
  tft.setTextColor(TFT_WHITE);
  tft.setCursor(12, 5);
  tft.print("Helo World!");
}
```

User_Setup.h (all contents):
```c
#define ST7789_DRIVER
#define TFT_WIDTH  240
#define TFT_HEIGHT 240

#define TFT_CS 19 // CS on display module
#define TFT_SCLK 18 // CLK on display module
#define TFT_MOSI 10 // DIN on display module 

#define TFT_DC 1 // DC on display module
#define TFT_RST 2 // RST on display module

#define TFT_BL 3 // BL on display module
#define TFT_BACKLIGHT_ON HIGH

#define LOAD_GLCD
#define LOAD_FONT2
#define LOAD_FONT4
#define LOAD_FONT6
#define LOAD_FONT7
#define LOAD_FONT8
#define LOAD_GFXFF

#define SMOOTH_FONT
#define SPI_FREQUENCY   27000000
#define SPI_TOUCH_FREQUENCY  2500000
```

stef-ladefense, 2022-04-21:
change RST pin and not use gpio 2

GPIO2, GPIO8, and GPIO9 are strapping pins of the ESP32-C3 chip. These pins are used to control several chip functions depending on binary voltage values applied to the pins during chip power-up or system reset. For description and application of the strapping pins, please refer to Section Strapping Pins in [ESP32-C3 Datasheet](https://www.espressif.com/sites/default/files/documentation/esp32-c3_datasheet_en.pdf).

Technerder, 2022-04-21:
> change RST pin and not use gpio 2
>
> GPIO2, GPIO8, and GPIO9 are strapping pins of the ESP32-C3 chip. These pins are used to control several chip functions depending on binary voltage values applied [...]

That was a dumb mistake on my part, although after instead using GPIO 1 for the RST pin the screen still doesn't show any text (backlight is still on though). Not quire sure if its a software or hardware issue at this point

Bodmer, 2022-04-24:
I started adding code for the C3 but it is not working and so that processor can be considered as unsupported at the moment. See [discussion here](https://github.com/Bodmer/TFT_eSPI/discussions/1712).

I do not have a C3 anyway t test, but recently adapted and tested a[ branch here](https://github.com/Bodmer/TFT_eSPI/tree/ESP32_S3) for the S3 and I suspect the C3 will need similar code changes.

The Arduino ESP32 board package appears to have a new software development team and so breaking changes are being made which means libraries need adaption for each processor variant! As an example no DS18B20 library seems to work with the S3 because low level calls have changed. Looking at the number of outstanding issues on the ESP32 board package indicates it will be some time before things settle down on these ESP32 variants!

Bodmer, 2022-04-24:
I patched the [branch here](https://github.com/Bodmer/TFT_eSPI/tree/ESP32_S3) so that it might now work with the ESP32 C3. Note I do not have an ESP32 C3 to test it with but it compiled OK under v 2.0.3 RC1 of the ESP32 Arduino board package!

Technerder, 2022-04-24:
I downloaded the branch and replaced all files in my libraries folder with the extracted output (excluding the `User_Setup.h` file) and it didn't seem to work. I'll try an ESP32-S2 and an RP2040 to make sure my display/driver isn't borked. What pins should I use on the ESP32-S2, are there any recommended pins?

Bodmer, 2022-04-24:
This one can be used:
https://github.com/Bodmer/TFT_eSPI/blob/ESP32_S3/User_Setups/Setup71_ESP32_S2_ST7789.h

Technerder, 2022-04-24:
The `boing ball` example sketch seems to at least be rendering on the display but something weird happens when the ball goes toward one of the edges on the screen.

https://user-images.githubusercontent.com/15988124/164991759-9d7069e9-7cd7-4e3c-ad64-8fe85981dee1.mp4

Bodmer, 2022-04-25:
Great, you have it working. The C3 is not supported at the moment.

The oddity with that example is to be expected as no graphics clipping is used at the edge of the screen. It is working fine. That example is for larger displays of 240 x 320 minimum.

closed 2022-04-25 by Bodmer

Bodmer, 2022-04-27:
ESP32 C3 is now supported in v2.4.61

Technerder, 2022-04-28:
> ESP32 C3 is now supported in v2.4.61

Was this tested on a physical board and if so would you be able to let me know what board, pins and sketch you used?

Bodmer, 2022-04-28:
Yes, M5 Stamp C3.
Setup here:
https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setups/Setup70c_ESP32_C3_ILI9341.h
Most sketches run except DMA.

Technerder, 2022-04-28:
Trying to run the `Gradient_Fill` sketch in the `Generic` folder with the following `User_Setup.h` file leads to the screen staying black, not sure if the sketch has been tested but it seems to be very basic and not use DMA.
```h
#define ST7789_DRIVER
#define TFT_WIDTH  240
#define TFT_HEIGHT 240

#define TFT_SCLK 4 // CLK on display module
#define TFT_MOSI 6 // DIN on display module 
#define TFT_CS 7 // CS on display module

#define TFT_DC 5 // DC on display module
#define TFT_BL 19 // BL on display module
#define TFT_RST 18 // RST on display module

#define TFT_BACKLIGHT_ON HIGH

#define LOAD_GLCD
#define LOAD_FONT2
#define LOAD_FONT4
#define LOAD_FONT6
#define LOAD_FONT7
#define LOAD_FONT8
#define LOAD_GFXFF

#define SMOOTH_FONT
#define SPI_FREQUENCY   27000000
#define SPI_TOUCH_FREQUENCY  2500000
```

Bodmer, 2022-04-28:
I tested with ESP32 board package 2.0.3 RC1.

Technerder, 2022-04-28:
Updated the ESP32 package to 2.0.3-RC1 but getting the same black screen (backlight is on)

```c
#include <TFT_eSPI.h>

TFT_eSPI tft = TFT_eSPI();

void setup() {
  tft.init();
  tft.setRotation(1);
  tft.fillScreen(TFT_DARKGREY);
  tft.setTextFont(2);
  Serial.begin(115200);
}

void loop() {
  Serial.println("Render");
  tft.fillRectHGradient(0, 0, 160, 50, TFT_MAGENTA, TFT_BLUE);
  tft.setCursor(10,10);
  tft.print("Horizontal gradient");
  tft.fillRectVGradient(0, 60, 160, 50, TFT_ORANGE, TFT_RED);
  tft.setCursor(10,70);
  tft.print("Vertical gradient");
}
```

Uploading Output:
```output
In file included from C:\Users\Tech\Documents\Arduino\libraries\TFT_eSPI/TFT_eSPI.h:71,
                 from C:\Users\Tech\AppData\Local\Temp\arduino_modified_sketch_945871\Gradient_Fill.ino:1:
C:\Users\Tech\Documents\Arduino\libraries\TFT_eSPI/Processors/TFT_eSPI_ESP32_C3.h:11:3: warning: #warning >>>>------>> DMA is not supported on the ESP32 C3 (possible future update) [-Wcpp]
  #warning >>>>------>> DMA is not supported on the ESP32 C3 (possible future update)
   ^~~~~~~
In file included from C:\Users\Tech\AppData\Local\Temp\arduino_modified_sketch_945871\Gradient_Fill.ino:1:
C:\Users\Tech\Documents\Arduino\libraries\TFT_eSPI/TFT_eSPI.h:883:8: warning: #warning >>>>------>> TOUCH_CS pin not defined, TFT_eSPI touch functions will not be available! [-Wcpp]
       #warning >>>>------>> TOUCH_CS pin not defined, TFT_eSPI touch functions will not be available!
        ^~~~~~~
In file included from C:\Users\Tech\Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.h:71,
                 from C:\Users\Tech\Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp:17:
C:\Users\Tech\Documents\Arduino\libraries\TFT_eSPI\Processors/TFT_eSPI_ESP32_C3.h:11:3: warning: #warning >>>>------>> DMA is not supported on the ESP32 C3 (possible future update) [-Wcpp]
  #warning >>>>------>> DMA is not supported on the ESP32 C3 (possible future update)
   ^~~~~~~
In file included from C:\Users\Tech\Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp:17:
C:\Users\Tech\Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.h:883:8: warning: #warning >>>>------>> TOUCH_CS pin not defined, TFT_eSPI touch functions will not be available! [-Wcpp]
       #warning >>>>------>> TOUCH_CS pin not defined, TFT_eSPI touch functions will not be available!
        ^~~~~~~
Sketch uses 342588 bytes (26%) of program storage space. Maximum is 1310720 bytes.
Global variables use 10836 bytes (3%) of dynamic memory, leaving 316844 bytes for local variables. Maximum is 327680 bytes.
esptool.py v3.3
Serial port COM11
Connecting....
Chip is ESP32-C3 (revision 3)
Features: Wi-Fi
Crystal is 40MHz
MAC: 7c:df:a1:da:65:84
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Flash will be erased from 0x0000e000 to 0x0000ffff...
Flash will be erased from 0x00000000 to 0x00003fff...
Flash will be erased from 0x00010000 to 0x0006afff...
Flash will be erased from 0x00008000 to 0x00008fff...
Compressed 8192 bytes to 47...
Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.2 seconds (effective 353.0 kbit/s)...
Hash of data verified.
Flash params set to 0x022f
Compressed 12528 bytes to 9002...
Writing at 0x00000000... (100 %)
Wrote 12528 bytes (9002 compressed) at 0x00000000 in 0.5 seconds (effective 211.0 kbit/s)...
Hash of data verified.
Compressed 369728 bytes to 205309...
Writing at 0x00010000... (7 %)
Writing at 0x0001a229... (15 %)
Writing at 0x0002504f... (23 %)
Writing at 0x0002b41e... (30 %)
Writing at 0x000315db... (38 %)
Writing at 0x00038e2e... (46 %)
Writing at 0x0003ed72... (53 %)
Writing at 0x00046029... (61 %)
Writing at 0x0004c394... (69 %)
Writing at 0x0005256a... (76 %)
Writing at 0x00058609... (84 %)
Writing at 0x000602d1... (92 %)
Writing at 0x000671a8... (100 %)
Wrote 369728 bytes (205309 compressed) at 0x00010000 in 5.7 seconds (effective 517.0 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 128...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (128 compressed) at 0x00008000 in 0.1 seconds (effective 186.5 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

Serial monitor output:
```output
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x438
load:0x403ce000,len:0x90c
load:0x403d0000,len:0x2358
SHA-256 comparison failed:
Calculated: a9753a4fc647c6545c1b919ef08db429130a48592727edca270f1e5a3da0d0a9
Expected: 3bf6ef2cf3b9eefcd4b3c70cc5d1ce5138292d101a5cb1d5db6fbebf081b0a19
Attempting to boot anyway...
entry 0x403ce000
Render
Render
Render
Render
Render
```

Bodmer, 2022-04-28:
If the SHA comparison failed this suggests the new sketch did not load correctly, so probably the previous sketch ran. It may be that there is an issue using certain pins. If you can I would use the pins exactly as in the tested setup provided.

Bodmer, 2022-04-28:
I have tried the pins you used and the ILI9341 display works fine.

I had a look at the display schematic and it does NOT have a 5V to 3.3V regulator so you MUST NOT power from 5V as that will very likely damage the display.

If you do indeed have the [display here](If you do indeed have the display here. ) as per your link above, then it is essential you power the display from 3.3V only.

If it has failed then ask Waveshare/vendor for a replacement as the information here indicating that you can power from 5V is wrong:

[image]

The schematic from the link is here:
https://www.waveshare.com/w/upload/0/0c/1.3inch_LCD_Module_Schematic.pdf

This indicates there is NO regulator fitted.

Bodmer, 2022-04-28:
Note also that parasitic paths via GPIO from a display powered from 5V could also damage the ECP32 C3, but it is likely to have survived as you have only wired C3 outputs to the display.