Bodmer/TFT_eSPI #1788: warning >>>>------>> TOUCH_CS pin not defined in TFT_eSPI-2.4.50

stef-ladefense, 2022-04-21:
Hello,
I like warnings, it often avoids problems.
on the other hand, this is only valid if the screen actually has a touch .
otherwise, the warning appears for any screen without touch.

it would be wise to include in the xxxx_defines.h file screens without touch for example
#define NOTOUCH

and in the TFT_eSPI.h on line 875

#elif!defined(NO_TOUCH)
     #warning >>>>------>> TOUCH_CS pin not defined, TFT_eSPI touch functions will not be available!
#endif

moreover, nothing prevents including a #define NOTOUCH manually in the User_Setup.h file in the case of non-use of the touch function of a screen which would be equipped with it.

What do you think ?

Regards,
Stéphane

pentatonico, 2022-04-22:
Yes please!

homonto, 2022-04-24:
I came here just to say: I DON'T LIKE THIS WARNING ;-)
I like my compilation to be CLEAN and without the warnings where I don't expect them 
just saying ;-)

Bodmer, 2022-04-27:
Added capability to stop alerts.

// Define in setup to disable all #warnings in library (can be put in User_Setup_Select.h)
#define DISABLE_ALL_LIBRARY_WARNINGS

closed 2022-04-27 by Bodmer

chriskuku, 2024-03-28:
Allow me to chime in here: my CYD device (cyd2usb) has capacitive touch and I'm wondering why I get this warning and the message that touch functions will not be available.
``
In file included from .pio/libdeps/cyd/TFT_eSPI/TFT_eSPI.cpp:16:
.pio/libdeps/cyd/TFT_eSPI/TFT_eSPI.h:973:8: warning: #warning >>>>------>> TOUCH_CS pin not defined, TFT_eSPI touch functions will not be available! [-Wcpp]
       #warning >>>>------>> TOUCH_CS pin not defined, TFT_eSPI touch functions will not be available!
        ^~~~~~~
``
platform.ini:

```
[platformio]
src_dir = .
default_envs = cyd2usb

[env]
platform = espressif32
board = esp32dev
framework = arduino
lib_deps = 
	bodmer/TFT_eSPI@^2.5.33
monitor_speed = 115200
monitor_filters = esp32_exception_decoder
# upload_speed = 921600
upload_speed = 460800
board_build.partitions=min_spiffs.csv
build_flags =
	-DUSER_SETUP_LOADED
	-DILI9341_2_DRIVER
	-DUSE_HSPI_PORT
	-DTFT_WIDTH=240
	-DTFT_HEIGHT=320
	-DTFT_MISO=12
	-DTFT_MOSI=13
	-DTFT_SCLK=14
	-DTFT_CS=15
	-DTFT_DC=2
	-DTFT_RST=-1
	-DTFT_BL=21
	-DTFT_BACKLIGHT_ON=HIGH
	-DTFT_BACKLIGHT_OFF=LOW
	-DLOAD_GLCD
	-DSPI_FREQUENCY=55000000
	-DSPI_READ_FREQUENCY=20000000
	-DSPI_TOUCH_FREQUENCY=2500000
	-DLOAD_FONT2
	-DLOAD_FONT4
	-DLOAD_FONT6
	-DLOAD_FONT7
	-DLOAD_FONT8
	-DLOAD_GFXFF

[env:cyd]
build_flags =
	${env.build_flags}
	-DTFT_INVERSION_OFF

[env:cyd2usb]
build_flags =
	${env.build_flags}
	-DTFT_INVERSION_ON

```

zRCrackiiN, 2024-03-28:
> Allow me to chime in here: my CYD device (cyd2usb) has capacitive touch and I'm wondering why I get this warning and the message that touch functions will not be available.
>
> `In file included from [...]

samesies! its weird! ive defined the touch_cs pin in user_setup.h so im confused...

BasPaap, 2024-06-12:
I would also prefer a more fine-grained way to disable warnings. Just disabling all library warnings seems excessive, I just want to be able to tell the library that I don't need a touch pin and that it shouldn't warn me about it, while still telling me about other warnings that may pop up. Not defining a touch pin is a valid choice, not a potential error.

fungreenfox, 2025-09-09:
Esp32 Wroom, ili89488-and touch. Been using same script for a year, and building on the script frequently.
When the esp32 is reaching void setup, the TOUCH_CS is undef'ed and defined as pin 1. Touch no longer working.
How to prevent this, when its on the gpio 5, that has build in resistor, so it should not be floating.