Bodmer/TFT_eSPI #1798: ILI9488 w/ Parallel not working? (possible regression)

codewitch-honey-crisis, 2022-04-25:
I get no sign of life other than the backlight when hooking up the ILI9488 via 8-bit parallel to an ESP32 (WROVER, not anything fancy like an S3)

I'm using PlatformIO.

I have a graphics library https://honeythecodewitch.com/gfx I wrote which I am using in the alternative.

Here's the thing. I kind of lifted your code for the ILI93488 initialization, and pasted into my GFX library's associated ILI9341 driver initialization routine, replacing the code (with some slight modification in terms of where transaction boundaries are but it makes no diff with parallel) 

When I did that my display fired right up with my library. The thing is initially my parallel driver code was copied from yours (since then heavily modified), and so was this driver code.

And I know it's not my wiring because I use the same pin settings in my code as you have defined in UserSetup.h when you choose parallel.

It might be worth connecting one and seeing if your latest revision of your code still works with it.

Bodmer, 2022-04-25:
I've run the latest revision of TFT_eSPI (2.4.60) and it works fine with ESP32 and ILI9488 in 8 bit parallel mode.

Make sure the right setup file is being loaded, if you have updated your library the setup you have had previously will be over-written!

I have tried the following unmodified setups:
```
Setup13_ILI9481_Parallel.h
Setup14_ILI9341_Parallel.h
Setup16_ILI9488_Parallel.h
```
All worked as expected.

Bodmer, 2022-04-25:
I used Version 2.0.3-RC1 as the voard package but just tried 2.0.2 and that works fine too. It helps if you quote what you use.

codewitch-honey-crisis, 2022-04-26:
It came from the platformIO repo the other day. library.json says "2.4.51". TFT_eSPI.h says you worked on it last on 04/02/22

I didn't try the user setups. I just mucked with User_Setup.h. Weird because I've used it before with no issues. Anyway, I must have messed something up in there. Sorry for the churn.

closed 2022-04-26 by codewitch-honey-crisis