Bodmer/TFT_eSPI #1799: Can the library add the HX8347D driver? How do I add the HX8347D driver

kaihangchen, 2022-04-25:
Can the library add the HX8347D driver? How do I add the HX8347D driver. Thank you

Bodmer, 2022-04-26:
You need to create 3 files [here](https://github.com/Bodmer/TFT_eSPI/tree/master/TFT_Drivers) for the controller:
```
HX8347D_Defines.h
HX8347D_Init.h
HX8347D_Rotation.h
```
You will have to work out the required conent of those files based on the existing ones.

Add the files to the lists [here](https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setup_Select.h#L222-L226), [here ](https://github.com/Bodmer/TFT_eSPI/blob/master/TFT_eSPI.cpp#L755-L758)and [here](https://github.com/Bodmer/TFT_eSPI/blob/master/TFT_eSPI.cpp#L852-L855).

closed 2022-04-26 by Bodmer

Bodmer, 2022-04-26:
The code here indicates support for that controller would need soe surgery on other parts of the library so that will make the task of adding the controller much more complex and error prone. I will be unable to help as I do not have one of those displays.