Bodmer/TFT_eSPI #1800: ST7789V2 Issues

PlastiBots, 2022-04-27:
Hi.  First, thanks for this great library! I'm having an issue where it seems only part of the screen is being used.  It's as if it's rendering content but shifted up by about 30px.  This is demonstrated by the attached images.  I've got the correct image WxH set per BuyDisplay.  The first shows the fillscreen function with yellow only filling the top area with missing fill to the right and bottom. The second shows images rendering correctly, but about 30px higher than normal.  There are graphics at the top that are not showing (pushed off the top).  

My setup: ESP32 DEVKIT V1. TFT_eSPI v2.4.50, Arduino IDE 1.8.13.  ESP32 Boards 2.0.2  TFT driver is ST7789V2. The display is [this one](https://www.buydisplay.com/1-9-inch-ips-tft-lcd-display-module-170x320-for-arduino-raspberry-pi) but note the page says ST7789 but the print on the board says ST7789V2.    Interface is SPI 

The TFT_eSPI Setup file I am using is a clone of 25:

```
/ Setup for the TTGO T Display
#define USER_SETUP_ID 25

// See SetupX_Template.h for all options available

#define ST7789_DRIVER
#define TFT_SDA_READ   // Display has a bidirectional SDA pin

#define TFT_RGB_ORDER TFT_BGR  // Colour order Red-Green-Blue  ??DJA

#define TFT_WIDTH  170 
#define TFT_HEIGHT 320

#define CGRAM_OFFSET      // Library will add offsets required

//#define TFT_MISO -1

//#define TFT_MOSI            19
//#define TFT_SCLK            18
//#define TFT_CS              5
//#define TFT_DC              16
//#define TFT_RST             23

#define TFT_SCLK 18    //SDL
#define TFT_MOSI 23   //SDA

#define TFT_CS   15  // Chip select control pin
#define TFT_DC   2 // Data Command control pin
#define TFT_RST  4  // Reset pin (could connect to RST pin)

#define TOUCH_CS            -1
#define TFT_BL          -1  // Display backlight control pin

#define TFT_BACKLIGHT_ON HIGH  // HIGH or LOW are options

#define LOAD_GLCD
#define LOAD_FONT2
#define LOAD_FONT4
#define LOAD_FONT6
#define LOAD_FONT7
#define LOAD_FONT8
#define LOAD_GFXFF

#define SMOOTH_FONT

//#define SPI_FREQUENCY  27000000
#define SPI_FREQUENCY  40000000

#define SPI_READ_FREQUENCY  6000000 // 6 MHz is the maximum SPI read speed for the ST7789V
```
[image]
[image]

PlastiBots, 2022-04-27:
Thanks.  This is what I get:

[image]

Bodmer, 2022-04-27:
OK, that makes sense as the library is limiting the y coord to 170. It tells me the offset is correct.

I will update the library within the next 24 hours.

Bodmer, 2022-04-28:
If you set the screen width to 240 instead of 170 then the above sketch will work and you can draw graphics providing you include the tft.setViewport(0, 35, 320, 170); in you sketch.

I will update the library but I have another task that is urgent.

PlastiBots, 2022-04-28:
That did the trick! Thanks so much for the quick attention to this!

commit f7e9a91e (2022-04-28): Fix #1800 Fix #1793

Added offsets for 170x320 ST7789 display
Added offsets for 240x280 display to ST7789_2 dirver
closed 2022-04-28 by PlastiBots