Bodmer/TFT_eSPI #1802: TFT_eSPI/examples/Sprite/Animated_dial - Still Working ?

bionicbone, 2022-04-28:
First, thanks for this awesome library, I have worked through most of the examples and the capabilities with just a little code are amazing. 

I wonder if you can confirm the example examples/Sprite/Animated_dial/ still works with the current version. When I initially tried this with a older versions of libraries installed it worked well, since then I updated to the latest version of several drivers and now the font in the middle is corrupted.

If I do not load the font, then a standard small font is used and then it is fine.

[image]

Arduino IDE v1.8.19
TFT_eSPI v1.2.1
ESP32 Boards Manager v2.0.2
ESP32 Wroover Module
Display ILI9341 v1.5.10 on SPI interface

To minimise effort to resolve issues the following should be provided as a minimum:
1. A description of the problem and the conditions that cause it to occur
2. IDE (e.g. Arduino or PlatformIO)
3. TFT_eSPI library version (try the latest, the problem may have been resolved!) from the Manage Libraries... menu
4. Board package version (e.g. 2.0.3) available from the Boards Manager... menu
5. Procesor, e.g RP2040, ESP32 S3 etc
6. TFT driver (e.g. ILI9341), a link to the vendors product web page is useful too.
7. Interface type (SPI or parallel)

Plus further information as appropriate to the problem:
8. TFT to processor connections used
9. A zip file containing your setup file (just drag and drop in message window - do not paste in long files!)
10. A zip file  containing a simple and complete example sketch that demonstrates the problem but needs no special hardware sensors or libraries.
11. Screen shot pictures showing the problem  (just drag and drop in message window)

The idea is to provide sufficient information so I can setup the exact same (or sufficiently similar) scenario to investigate and resolve the issue.

Put three ticks ( ` character on key next to "1" key) at the start and end of short pasted code segments to avoid format/markup anomolies and to use a monspaced font. Example:

```
  Serial.begin(115200);
  tft.init();
```

Bodmer, 2022-04-28:
Ah, a new rendering approach is used for smooth fonts that requires an additional parameter. The new approach reduces flicker dramatically.

```
// Adding a parameter "true" to the setTextColor() function fills character background
    // This extra parameter is only for smooth fonts!
    tft.setTextColor(TFT_GREEN, TFT_BLACK, true);
```
I will update the example.

Bodmer, 2022-04-28:
I have updated the library github master example.

closed 2022-04-28 by Bodmer

bionicbone, 2022-04-29:
Unfortunately this still does not work for me.

I reinstalled the library, just in case and did an overwrite of the example to make sure I got the latest updated github version.

Looking at the updated example " examples/Sprite/Animated_dial/" there is a slight bug, bg_color needed to initiated as global since it is now used in both setup() & plotNeedle() 

However, for me the font is still corrupted in the middle of the dial even with the "spr.setTextColor(TFT_WHITE, bg_color, true);" changes.

I note that in the example "examples\Smooth Fonts\FLASH_Array\Font_Demo_1_Array" you have recently updated this code to remove the overlay of the rectangle and use the ",true" statement instead and this works successfully for me perfectly. 

However, this example is not using a sprite to display the text.

Question: What is the advantage of using a sprit for text ?

Sorry this is probably a discussion now, but I did not want to loose the thread above...

bionicbone, 2022-04-29:
Just drilling in a bit using visual micro we can see as the sprite updates using breakpoints there is some corruption of the font.

[image]

[image]

[image]

[image]

[image]

Bodmer, 2022-04-29:
I have updated the sprite.cpp file to fix this.

Thanks for reporting the problem.

bionicbone, 2022-04-30:
Perfect, thank you for your trouble in resolving this.

Just so you know, because I think the example " examples/Sprite/Animated_dial/" is perfect to demonstrate flicker free needles on top of gauges.

There is a slight compile error, bg_color needs to be initialised in the header with the others because it is used in the setup() & plotNeedle() and for me I increase the spr_width using "777" the remove some artifects / white dots

```
uint16_t* tft_buffer;
bool      buffer_loaded = false;
uint16_t  spr_width = 0;
uint16_t**  bg_color = 0;
```
```
  bg_color = tft.readPixel(120, 120); // Get colour from dial centre
```
```
 spr_width = spr.textWidth("777");
```

Thank you again

Bodmer, 2022-04-30:
Thanks. Have updated example.

commit 22901f5d (2022-04-28): Fix #1802

setTextColor has a new parameter for smooth fonts to force background fill.
commit 7d3f4f21 (2022-04-29): Fix #1802
commit fe5609e3 (2022-04-30): Fix #1802