Bodmer/TFT_eSPI #1807: Animated_dial Compliation error - ESP32 / ILI9341

Amitabh-wq, 2022-05-02:
Hi, I am trying to test the example called 'Animated_dial' but geting the IDE error:

I:\Users\UserJ\Documents\Arduino\libraries\TJpg_Decoder\src/TJpg_Decoder.h:23:26: fatal error: LittleFS.h: No such file or directory

I have installed all the libraries required. Any toughts? Thanks

Bodmer, 2022-05-02:
Not all processor board packages provide LittleFS. Those that do include:
ESP32
ESP8266
RP2040 (Earle Philhowers package only)

Consult the board package documentation.

closed 2022-05-02 by Bodmer

Amitabh-wq, 2022-05-02:
I am using the ESP32:

[code]
TFT_eSPI ver = 2.4.51
Processor    = ESP32
Frequency    = 240MHz
Transactions = Yes
Interface    = SPI
Display driver = 9341
Display width  = 240
Display height = 320

MOSI    = GPIO 23
MISO    = GPIO 19
SCK     = GPIO 18
TFT_CS   = GPIO 15
TFT_DC   = GPIO 2
TFT_RST  = GPIO 4
TOUCH_CS = GPIO 5

Font GLCD   loaded
Font 2      loaded
Font 4      loaded
Font 6      loaded
Font 7      loaded
Font 8      loaded
Smooth font enabled

Display SPI frequency = 40.00
Touch SPI frequency   = 2.50
[/code]