Bodmer/TFT_eSPI #1808: how to shift the screen (i.e. to the left)?

homonto, 2022-05-02:
Hi @Bodmer 
I am not sure where to ask so feel free to delete the post please ;-)
I am working on the project with ILI9341, where big part of the screen is used to display life data (around 200 x 300px). The rest is not moving - just updating the values.
I have around 300px to work with (horizontal orientation) and I am printing the lines and pixels every second.
Of course, after 300 seconds I am ... lost - the graph is either pointing out of the screen or... 

**I see 2 options:**
1- clear the plot area and start from x=0 but that is not nice - history is lost
2- start shifting the plot area to the left: clean left side i.e by 50px, shift everything by 50px to the left having now extra 50px to accommodate next 50 seconds. And continue if needed - of course screen can accommodate only 300s with granularity 1s/1px.
This  option would be also great with shifting also up/down - not only left - as sometimes data incoming is going very down or very up and then the graph is either too close to the bottom or to the top.

I like the second option but I have NO B...Y idea how to touch it. I am looking for the examples but the only ones I found are text scrolling. But how about already printed graphs?

3- the last option (but probably even more difficult) would be: after 300 seconds passed, start compressing the graph and instead of plotting every 1s do it every 2s or so - but that would require much more as data is nowhere stored - just plotted - so maybe not the best option this one is.

Hopefully you or someone else would point me into the proper direction.

Thank you.

Bodmer, 2022-05-03:
Have a look at the UTFT_Demo sketch, at around line 100 it draws a moving sine wave.

The old pixel y coordinate values are stored in an array so the position of each pixel is known, thus the old pixel can be erased and a new one draw. In your case the new one will be the next array value. i.e. erase array[n] pixel, draw array[n+1] pixel at that same x coord. When the last one is done, copy array [n+1] to array[n] and repeat for the next sample to be drawn.

homonto, 2022-05-03:
thank you @Bodmer 
So in short: no library exists that does the magic (for lazy programmer like me) - it has to be written and then displayed again just with different values, right?
No problem. I will try ;-)

homonto, 2022-05-03:
OK, I analysed it (by adding few delays here and there - otherwise it was too fast) - it looks neat: it starts from x=0 when end of the screen is reached, and draws again new screen but keeps deleting around 30px ahead, so that it looks like ... a snake game from old fashioned Nokia, right?
So that is option 4 that I never thought about ;-)
Thank you again @Bodmer

closed 2022-05-03 by homonto

arnolde, 2023-03-31:
Hi there,
I also would like to display a realtime moving graph on a 480x320 TFT (ili9488 spi).
I tried modifying the UTFT_DEMO sketch, here is what my loops look like:
```
  for (int i=0; i<10000; i++)
  {
    y=159+(sin(((i*0.7)*3.14)/180)*(90-(i / 100)));
    buf[480]=y;
    for (int x=0; x<480; x++) {
      tft.drawPixel(x,buf[x],TFT_BLACK);
      buf[x] = buf[x+1];
      tft.drawPixel(x,buf[x],TFT_WHITE);
    }
  }
```
But even with no delay() the moving sine wave is very slow (using ESP32 at 240Mhz and 40Mhz SPI speed), one cycle takes over 1 second to move across the screen. Are the 960 drawPixel actions THAT slow that each pixeldraw takes over a millisecond?
(I tried replacing the black pixels by a black rectangle before the inner loop but that flickers a lot, as expected, no good)

Is there any way to accelerate this?

homonto, 2023-03-31:
sprite?

arnolde, 2023-03-31:
To shift the whole screen?

homonto, 2023-03-31:
why not?

arnolde, 2023-03-31:
I'll try what this person demonstrates: https://www.youtube.com/watch?v=sRGXQg7028A
I'll still skeptical that a full-screen transfer operation is faster than 960 single pixels, but ok. And I'm still surprised that the ili9488 does not have a simple internal shift operation that would reduce the SPI communication to 2 small commands (shift and draw one pixel).