Bodmer/TFT_eSPI #1809: ILI9486 touch

AurumVR, 2022-05-02:
Hi!

I have an **ILI9486** 4" touch LCD, [this type](http://www.lcdwiki.com/4.0inch_SPI_Module_ILI9486)

[image]

Wired to **ESP32 WROOM**, I could get the graphics to work with TFT_eSPI just fine, except the touch controller.

[image]

I have tested it with the [XPT2046_Touchscreen library](https://github.com/PaulStoffregen/XPT2046_Touchscreen) to make sure the hardware is okay - and it works fine with or without IRQ. (T_DO -> MISO, T_DI -> MOSI, T_CLK -> SCLK, T_CS -> GPIO4)

But not with TFT_eSPI. I have edited TOUCH_CS definition in User_Setup.h, tried different pins for CS.

Wired SCK to T_CLK, SDI(MOSI) to T_DI and SDO(MISO) to T_DO, CS is GPIO4. 

I've tested it with Keypad_480x320 and Touch_Controller_Demo from Examples. Graphics works, touch doesn't,

I'm running out of ideas, please advise :)

homonto, 2022-05-02:
I am "one click" from buying these displays - let us know when you sort it out.
Today I am working with ILI9341 - a bit different one (2.8") but also working with touch.
I simply soldered 3 pins between touch and display directly on the display board - less cables to ESP32 then (MISO, MOSI and CLK) - of course each one needs its CS pin.
What is your beginning of the custom_setup?
Mine is like this:

```
#define ILI9341_DRIVER   
// modified by ZH 
#define ILI9341_DRIVER_HOT_PLATE  // ZH: distinguish between different projects
#define TFT_CS   	15  
#define TFT_DC    	5  
#define TFT_RST   	17  
#define TFT_MOSI 	13
#define TFT_SCLK 	14
// LED: PWM or fixed 	18
#define TFT_MISO 	4 		
#define TOUCH_CS 	16  
```

AurumVR, 2022-05-02:
#define ILI9486_DRIVER

#define TFT_SDA_READ      // This option is for ESP32 ONLY, tested with ST7789 and GC9A01 display only
// this one above is irrelevant to the issue I believe

#define TFT_MISO 19
#define TFT_MOSI 23
#define TFT_SCLK 18
#define TFT_CS   5   // Chip select control pin
// standard VSPI (SPI3) pins for ESP32

#define TFT_DC   17  // Data Command control pin
#define TFT_RST  16  // Reset pin (could connect to RST pin)

#define TOUCH_CS 4     // Chip select pin (T_CS) of touch screen

#define SPI_FREQUENCY  40000000

// Optional reduced SPI frequency for reading TFT
#define SPI_READ_FREQUENCY  20000000

// The XPT2046 requires a lower SPI clock rate of 2.5MHz so we define that here:
#define SPI_TOUCH_FREQUENCY  2500000

Tried different SPI frequencies for TFT, didn't matter. Haven't changed it for the XPT2046, I believed it's good. :)

Bodmer, 2022-05-03:
Based on your picture and web page, this display has a diode fitted in the chip spected line to the display. So a modification to the board is needed, see ST7796 based board here that has the same problem:
https://github.com/Bodmer/TFT_eSPI/discussions/898

AurumVR, 2022-05-03:
Hi!

I've shorted the diode, but this did not solve the issue. Also, disconnected the display SDO (MISO) line, it did not help either. I will try to re-add the diode and add a pull-up resistor on the anode side as suggested by you in the other thread.

Bodmer, 2022-05-03:
The pullup method is not recommended. Run the touch test and diagnostic sketch to verify the touch interface.

Bodmer, 2022-05-03:
In cidentally, of the other touch screen library works then it can be used with TFT_eSPI by adapting the sketches to use that libraries API.

copyCATandpasteDOG, 2022-05-03:
Hello,
i think this has been discussed a long time ago,you have to let the MISO pin of the TFT on air (nothing)

this is a setup i use with this TFT 4.0 and esp32 Wroom:

//works with  AVEC ILI9488 & ST7796 too
//don't wire TFT_MISO ,but TOUCH_MISO YES to 19 !!!

#define ILI9488_DRIVER	

#define SPI_MISO    19
#define SPI_MOSI    23
#define SPI_SCLK    18

#define TFT_MISO    SPI_MISO //-->NOTHING
#define TFT_MOSI    SPI_MOSI
#define TFT_SCLK    SPI_SCLK
#define TFT_CS      14      // Chip select control pin
#define TFT_DC      26//27      // Data Command control pin
#define TFT_RST     33
#define TFT_BL      32       // TFT backlight  pin
#define TFT_BACKLIGHT_ON HIGH

#define TOUCH_MISO     SPI_MISO
#define TOUCH_MOSI     SPI_MOSI
#define TOUCH_SCLK     SPI_SCLK
#define TOUCH_CS       12      // TouchScreen Chip Select

#define SPI_FREQUENCY  40000000
#define SPI_READ_FREQUENCY  20000000
#define SPI_TOUCH_FREQUENCY  2500000

#define LOAD_GLCD   // Font 1. Original Adafruit 8 pixel font needs ~1820 bytes in FLASH
#define LOAD_FONT2  // Font 2. Small 16 pixel high font, needs ~3534 bytes in FLASH, 96 characters
#define LOAD_FONT4  // Font 4. Medium 26 pixel high font, needs ~5848 bytes in FLASH, 96 characters
#define LOAD_FONT6  // Font 6. Large 48 pixel font, needs ~2666 bytes in FLASH, only characters 1234567890:-.apm
#define LOAD_FONT7  // Font 7. 7 segment 48 pixel font, needs ~2438 bytes in FLASH, only characters 1234567890:-.
#define LOAD_FONT8  // Font 8. Large 75 pixel font needs ~3256 bytes in FLASH, only characters 1234567890:-.
//#define LOAD_FONT8N // Font 8. Alternative to Font 8 above, slightly narrower, so 3 digits fit a 160 pixel TFT
#define LOAD_GFXFF  // FreeFonts. Include access to the 48 Adafruit_GFX free fonts FF1 to FF48 and custom fonts

#define SMOOTH_FONT

closed 2022-05-03 by AurumVR

AurumVR, 2022-05-04:
Well, actually in my case it seems it works regardless of shorting the diode or not, as long as SDO is disconnected. Conclusion:
The problem was, when the program wanted to calibrate at the first run it received false SPI data from the display (due to the hardware problem with the diode and SDO) and stored wrong calibration data. In practice the calibration screen flashed for a moment and disappeared. After that, as calibration data existed in SPIFFS it didn't calibrate again, but supplied touch coordinates based on the wrong calibration. As such the touch "didn't work" (it did actually work just the function returned incorrect values).

ESP32 + TFT_eSPI lib:
----------------------
LCD pin - ESP32 pin
----------------------
CS - GPIO 5
RESET - GPIO 16
DC/RS - GPIO 17
SDI (MOSI) - GPIO 23
SCK - GPIO 18
LED - VCC
SDO (MISO) - leave unconnected
T_CLK - wire to SCK
T_CS - GPIO 4 (configure in User_Setup.h)
T_DIN - wire to SDI
T_DO - GPIO 19
T_IRQ - may or may not needed, rather not

Bodmer, 2022-05-04:
The correct solution is:

1. Bypass the diode
2. Do not connect MISO (Note: this is not always the case, it depends on the display controller, for example on the ST7796 it can be connected. The ILI9488 must not be connected.)

If you do not bypass the diode there is some "luck" involved whether the display will get corrupted in certain circumstances. If the diode in not bypassed the display can receive commands destined for the touch (because the display CS line is slow to rise) which will corrupt display pixels.