Bodmer/TFT_eSPI #1812: glyph background causing vertical line (`fillheight`) with custom font

MomePP, 2022-05-03:
Hi,
I got an issue while trying to padding text as a picture below. It creates black verical line after `setTextPadding()` is being used (`_fillbg` is set). I've try to debugging with `PADDING_DEBUG` flag.

[image]

Currently, I found out that a variable `fillheight` was declared as unsigned int16. In my case, this value goes negative and fill rect with height 65535, so it appears as vertical line.

[image]

[image]

After changed the variable to `int16_t`, it's fixed my issue. I'm not sure, it supposed to changed this variable type or it's fault from a custom font that being used. 
https://github.com/Bodmer/TFT_eSPI/blob/24b0e88852399690a42cbec18e16e6ae7aaa5a49/Extensions/Smooth_font.cpp#L434

BTW, a custom font that i used is Baloo2 regular with font size 12, I've created by using your [pde script](https://github.com/Bodmer/TFT_eSPI/tree/master/Tools/Create_Smooth_Font/Create_font).
Here is my converted font [baloo2_regular_12.h.zip](https://github.com/Bodmer/TFT_eSPI/files/8609850/baloo2_regular_12.h.zip)
I notice that there was only some characters such as `f` or `i` that cause the issue. (`gdY` more than `maxAscent` 😵‍💫)

Dev. environment
- PlatformIO
- ESP32 with `arduino-esp32` master branch
- `TFT_eSPI` version 2.4.61
- ST7735 1.8" with SPI interface, no touch controller

Thank you in advance.

Bodmer, 2022-05-03:
Thanks for reporting this and finding the bug, yes the result can be negative so your solution is correct.

I have updated the master library.

closed 2022-05-03 by Bodmer

MikeyMoMo, 2022-05-16:
2.4.61 fixed my issue.  Thanks for the quick work!

commit 1f7ce88d (2022-05-03): Fix #1812

Result of "fillheight = gFont.maxAscent - gdY[gNum];" can be negative.