Bodmer/TFT_eSPI #1818: My TFT screen does only show a white screen - I have tried to fix this problem myself. No luck.

Kekzeee1, 2022-05-07:
To minimise effort to resolve issues the following should be provided as a minimum:
1. A description of the problem and the conditions that cause it to occur
I have connected my ESP32 board according a guide online, but the tft screen shows only white not the screen that it is programmed to show.
2. IDE (e.g. Arduino or PlatformIO)
ESP32
3. TFT_eSPI library version (try the latest, the problem may have been resolved!) from the Manage Libraries... menu
2.3.51
4. Board package version (e.g. 2.0.3) available from the Boards Manager... menu
2.02
5. Procesor, e.g RP2040, ESP32 S3 etc
ESP32
6. TFT driver (e.g. ILI9341), a link to the vendors product web page is useful too.
ILI9341
7. Interface type (SPI or parallel)
SPI
[image]
[image]
[image]

```
// TFT_eSPI - Version: 2.4.51
#include <TFT_eSPI.h>

//xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
//xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
//
//  This is the firmware for an ESP32 module combined with a 2.8" TFT Touchdisplay (SPI)
//  The purpose of the device is to function as a door sign of your home office which
//  indicates if people may enter the room.
//  If you do not want to be disturbed the display shows a red sign with a user definable text.
//  If you may allow to enter the display shows an orange sign with a different text.
//  if you allow people to enter the display shows a green sign with another text.
//  In red or orange state people may ask to get permission to enter by pressing the touch display.
//  A pending request is indicated by a blue background color of the display.
//
//  The device is controlled via a web interface using its WLAN capabilities. It must be connected to
//  your local WLAN. You can use any browser (PC, Tablet, Smartphone) for the UI.
//  The ESP32 runs a web server with a html page showing three buttons to set green, red or orange state.
//  There is another box shown which indicates that someone has requested to enter.
//  In red or orange state you may start a countdown timer. Its start value is entered by pressing
//  one of three buttons on the web page: "5" and "10" are adding 5 or 10 minutes to the current
//  time left. "00" resets the timer.
//  the time left (in minutes) is also displayed on the TFT display (door sign). When the timer
//  elapses the sign is automatically switched in green state.
//  You can end the timer by clicking on one of the three state buttons.
//
//  You can edit the texts which are displayed on the web surface as well as those shown on the
//  TFT Display in the #define section. But the space is limited!
//  DO NOT FORGET TO EDIT THE WLAN CREDENTIALS!!!
//
//  Know problems:
//  The web server is very limited. It can only connect to a single client. If a client browser
//  is holding the connection instead of breaking it after receiving, no other client can connect
//  The device is not responsive before it can connect to a NTS server.
//
//  This sketch needs the following libraries to be installed (use the library manager):
//      ESP32 core
//      SPI, Wifi, WiFiUdp and a specifci ESP32 version of time.h (all part of the ESP32 package)
//      TFT_eSPI
//  The time.h of ESp32 core does have support for NTS over WLAN (time synchronisation) see google
//  for more information. I used this functionality to display time and date on the web GUI
//
//  The TFT library functions do need a local font file "MyFont.h" to be copied from the lib package
//  directory to the sketch diractory
//  Make sure all the display driver and pin comnenctions are correct by
//  editting the User_Setup.h file in the TFT_eSPI library folder:
//
//  pin on TFT    pin on ESP32 Dev Kit
//      VCC         V5 (+5 Volt)
//      GND         GND (DO NOT USE THE ONE NEXT TO V5!)
[... 386 lines cut ...]
            // Break out of the while loop
            break;
          } else { // if you got a newline, then clear currentLine
            currentLine = "";
          }
        } else if (c != '\r') {  // if you got anything else but a carriage return character,
          currentLine += c;      // add it to the end of the currentLine
        }
      } else {
        // do these two calls, just in case the clients holds the connection and
        // would block the responsiveness of the server
        // if the functions would only be called outside the while loop
        checktimer();
        checktouch();
      }
    }
    // Clear the header variable
    header = "";
    // Close the connection
    client.stop();
    Serial.println("Client disconnected.");
    Serial.println("");
  }
}
```

Bodmer, 2022-05-07:
Run the Read_User_Setup sketch, check the serial monitor output. Post a copy of the output here.

Kekzeee1, 2022-05-08:
`[code]
TFT_eSPI ver = 2.4.51
Processor    = ESP32
Frequency    = 240MHz
Transactions = Yes
Interface    = SPI
Display driver = 9341
Display width  = 240
Display height = 320

MOSI    = GPIO 23
SCK     = GPIO 18
TFT_CS   = GPIO 15
TFT_DC   = GPIO 0
TFT_RST  = GPIO 2
TOUCH_CS = GPIO 21

Font GLCD   loaded
Font 2      loaded
Font 4      loaded
Font 6      loaded
Font 7      loaded
Font 8      loaded
Smooth font enabled

Display SPI frequency = 27.00
Touch SPI frequency   = 2.50
[/code]`

I think that I've done it right, I couldn't upload it on the Arduino IDE it said "Error compiling for board ESP32 Dev Module, so I did it on the web arduino editor and I got this result.

Bodmer, 2022-05-08:
I assumed you had it wired according to the sketch comments, viz:
```
//
//  pin on TFT    pin on ESP32 Dev Kit
//      VCC         V5 (+5 Volt)
//      GND         GND (DO NOT USE THE ONE NEXT TO V5!)
//      CS          G15 (chip select)
//      RESET       G4
//      D/C         G2  (device instruction)
//      SDI         G23 (SPI MOSI)
//      SCK         G18 (SPI clock)
//      LED         3V3 (3.3>V for backlight)
//      SDO         G19 (SPI MISO)
//      T_CLK       G18 (Touch SPI clock)
//      T_CS        G21 (Chipselect Touch)
//      T_DIN       G23 (SPI MOSI)
//      T_DO        G19 (SPI MISO)
//      T_IRQ       not used, leave open!
//
```
The setup however allocates DC and RST to use different pins, is that what you intended?

How have you wired, it (I can't see clearly from the photos)?

Compare your wiring to the Read_User_Setup output.

Have you tried any other library examples?

Kekzeee1, 2022-05-11:
Oh, hahah now it works. Thank you so much, I only have one problem left. And that is that the touch is not working. Could you take a look?

Thanks

Bodmer, 2022-05-11:
It looks like you have not connected all these pins:
//      T_CLK       G18 (Touch SPI clock)
//      T_CS        G21 (Chipselect Touch)
//      T_DIN       G23 (SPI MOSI)
//      T_DO        G19 (SPI MISO)

Kekzeee1, 2022-05-11:
T_DO 
T_DIN
Both pins are already connected but to different pins on the Screen. 

For example it says
T_DO G19, but SDO is also connected to G19
T_DIN G23 but SDI is also connected to G23

Kekzeee1, 2022-05-11:
The other pins are connected.

Kekzeee1, 2022-05-13:
Do you have any idea about my problem?

closed 2022-05-14 by Kekzeee1