Bodmer/TFT_eSPI #1822: ILI9341 320x240 Screen suddenly turned inverted colors after few mins of running - similar to #1486

homonto, 2022-05-09:
I am using ILI9341 for quite some time but recently the same issue as described in #1486 happens: inverted colors.
Only restart helps.
The screen is with touch. MCU is ESP32.

It is not routine - it happens sometimes (very rarely but still few times per day) without any message on the Serial although I enabled verbose debug.
I am using Arduino 1.8.20 and the latest TFT_eSPI: 2.4.51

I am not sure where to start looking for the issue.
My user setup is as following for this display:

```
#define DISABLE_ALL_LIBRARY_WARNINGS
#define ILI9341_DRIVER   // Full configuration option, define additional parameters below for this display
#define ILI9341_DRIVER_HOT_PLATE 
#define TFT_CS   	15  // Chip select control pin
#define TFT_DC    	5  // Data Command control pin
#define TFT_RST   	17  // Reset pin (could connect to RST pin)
#define TFT_MOSI 	13
#define TFT_SCLK 	14
#define TFT_MISO 	4 		// MISO for touch screen
#define TOUCH_CS 	16     // Chip select pin (T_CS) of touch screen
#define LOAD_GLCD   // Font 1. Original Adafruit 8 pixel font needs ~1820 bytes in FLASH
#define LOAD_FONT2  // Font 2. Small 16 pixel high font, needs ~3534 bytes in FLASH, 96 characters
#define LOAD_FONT4  // Font 4. Medium 26 pixel high font, needs ~5848 bytes in FLASH, 96 characters
#define LOAD_FONT6  // Font 6. Large 48 pixel font, needs ~2666 bytes in FLASH, only characters 1234567890:-.apm
#define LOAD_FONT7  // Font 7. 7 segment 48 pixel font, needs ~2438 bytes in FLASH, only characters 1234567890:-.
#define LOAD_FONT8  // Font 8. Large 75 pixel font needs ~3256 bytes in FLASH, only characters 1234567890:-.
#define LOAD_GFXFF  // FreeFonts. Include access to the 48 Adafruit_GFX free fonts FF1 to FF48 and custom fonts
#define SMOOTH_FONT
#define SPI_FREQUENCY  80000000
#define SPI_READ_FREQUENCY  20000000
#define SPI_TOUCH_FREQUENCY  2500000
#define USE_HSPI_PORT
```

homonto, 2022-05-10:
@Bodmer I will change the frequency to 40MHz and see (that is why I included these numbers here as well). 
The other points, although valid are not the case - 10cm dupoint cables from PCB to display, not close to the antenna or ESP32, no relay in the circuit at all either.

homonto, 2022-05-16:
Thank you @Bodmer I was not answering yet as I changed the frequency to 40MHz but still I see sometimes "grey" or "white" screen however, I think, this is somewhere within my code, when I mistakenly call something to refresh the screen too often (inside the loop instead of on event). 
I think we could close the topic. However I have to be careful as the hotplate that suddenly stops responding would be not only disaster but also dangerous ;-)

[image]
[image]

Bodmer, 2022-05-16:
Interesting project. I have been looking at reflow approaches recently (hotplate, toaster oven, hot air) as I have 12 SMD boards to build that have tiny TSOP packages, unfortuantley that was the only package format available.

I assume you are driving a SSR via an opto-isolated interface. You may find the high power switching does get coupled into signal lines via parasitic paths.

homonto, 2022-05-16:
I was actually thinking the same: high current there (450W) so around 2-3A, the box is quite compact, so I tried to isolate all the cables etc. 
The SSR goes without opto as it is SSR-DA25 - low voltage control.
The plate is 10x10cm - more than enough for me, based on this one:
[image]

The "dirty" video about this project is here (not finished yet, not YT-like so not public etc.):
[https://www.youtube.com/watch?v=vuXMJCXx7WE](https://www.youtube.com/watch?v=vuXMJCXx7WE)

homonto, 2022-05-16:
@Bodmer  check at this time here - someone is talking about you there ;-)
[https://youtu.be/vuXMJCXx7WE?t=751](https://youtu.be/vuXMJCXx7WE?t=751)

Bodmer, 2022-05-18:
A great project. I suspect the power switching is causing voltage spikes that upset the processor and/or display. This is likely to be a common mode transient. Make sure the 0V logic line is connected to the hot plate metal box and that all the box panels are properly earthed. The SSR is internally isolated and appears to be of a type that only switches on the zero crossings of the mains input which is good, but there will still be a significant common mode voltage surge that can couple through power supplies and induce unwanted voltage transients in logic lines.

homonto, 2022-05-18:
ha, thanks, MAX31855 works that way, that thermocouple CANNOT be grounded and because I am reusing the thermocouple that is already installed there (it is in between the heater elements, inside the plate), the 0V of the ESP32 (so also MAX31855) cannot be connected to the chassis. However chassis is earthed. This has to suffice as I am not able to buy MAX31856 nowadays ;-)

Btw "tea time" mode is working already and since I can set up heating time and target temperature from Home Assistant, it is multipurpose mode: either real tea (i.e. 80C) or simply baking/desoldering (i.e. 200C) for the arbitrary time, when I can always press the STOP button if finished.
But the reflow mode is still the main target. 

The way you showed me in another post how to reuse the screen when time passes (x=max) - when I reach end of the display (X axis) I am simply starting from x=0, erasing ahead 30px (within the plot area - not full screen) and drawing again - kind of dirty hack but this way I don't need to remember the old values. So thanks for the hint.

Bodmer, 2022-05-18:
I later watched the whole video and could see that the reflow controller appeared to be connected to a PC with USB leads and long wires, these all add to the conducted EMC problem when switching power due to the loop area.

True, thermocouple tip and wires should not be grounded. You will end up with a ground from the PC via the USB shield anyway so providing the thermocouple leads are only connected to the MAX31855 pins it is OK for the system 0V line to be referenced to the local Earth. You will probably find that when used as a standalone unit without multiple signal paths to the test equipment (PC etc) that the unit behaves reliably. You could add debug print lines to see when the display upset occurs. This is often when the power is switched off rather than on due to inductive surges.

homonto, 2022-05-18:
@Bodmer  spot on - it never happened when USB was disconnected ;-)

when I finish the project I will do proper video and share on github the code - I am just a bit shy as my code is so ugly comparing to others... especially to your library ;)
but maybe someone will find it useful as there is a lot in it with discovery on Home Assistant (MQTT) etc.

Bodmer, 2022-07-22:
@homonto 
This new Widget library may be of interest:

https://github.com/Bodmer/TFT_eWidget

I have built a reflow hotplate and used this example graph during test:

https://github.com/Bodmer/TFT_eWidget/tree/main/examples/Graphs/Reflow_graph_demo

homonto, 2022-07-22:
both links the same:
[image]

Bodmer, 2022-07-22:
@homonto 
Sorry it was a private repository.

Have now made it public so you should be able access it now.