Bodmer/TFT_eSPI #1825: Compile issues with ESP32S3 Dev Module

code-is-art, 2022-05-12:
To minimise effort to resolve issues the following should be provided as a minimum:
I am getting compile errors trying to compile the TFT_Meters example using ESP32S3 Dev Module board setting with the following user settings...
```
// Setup for the ESP32 S3 with ILI9488 display
// Note SPI DMA with ESP32 S3 is not currently supported
#define USER_SETUP_ID 70
// See SetupX_Template.h for all options available
#define ILI9488_DRIVER 

                    // Typical board default pins - change to match your board
#define TFT_CS   10 //     10 or 34 (FSPI CS0) 
#define TFT_MOSI 11 //     11 or 35 (FSPI D)
#define TFT_SCLK 12 //     12 or 36 (FSPI CLK)
#define TFT_MISO 13 //     13 or 37 (FSPI Q)

// Use pins in range 0-31
#define TFT_DC    14
#define TFT_RST   15

//#define TOUCH_CS 16 // Optional for touch screen

#define LOAD_GLCD
#define LOAD_FONT2
#define LOAD_FONT4
#define LOAD_FONT6
#define LOAD_FONT7
#define LOAD_FONT8
#define LOAD_GFXFF

#define SMOOTH_FONT

// FSPI port (SPI2) used unless following defined. HSPI port (SPI3) NOT TESTED YET
//#define USE_HSPI_PORT

#define SPI_FREQUENCY  27000000
//#define SPI_FREQUENCY  40000000   // Maximum for ILI9341

// #define SPI_READ_FREQUENCY  6000000 // 6 MHz is the maximum SPI read speed for the ST7789V
// Optional reduced SPI frequency for reading TFT
#define SPI_READ_FREQUENCY  16000000

#define SPI_TOUCH_FREQUENCY 2500000
```
I am using the Arduino IDE 1.8.19 or 2.0.0 rc6 and TFT_eSPI library version 2.4.51. I have ESP32 Arduino boards package version 2.0.3. This is using an ESP32-S3-WROOM-1-N8R2 with [one of these displays](https://www.amazon.com/gp/product/B09BZZXYJD/ref=ppx_yo_dt_b_asin_title_o06_s01?ie=UTF8&th=1) using SPI. Here is a text file with the errors.
[errors.txt](https://github.com/Bodmer/TFT_eSPI/files/8674144/errors.txt)

code-is-art, 2022-05-12:
using
```
https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json
```
gives the same errors

code-is-art, 2022-05-12:
OK I got it to compile downloading the master branch but it looks like there are parts missing from the meters demo. The numbers and letters and meter dial are showing but none of the other lines or colors.
[image]

code-is-art, 2022-05-12:
Re-opening until the updated lib is pushed to the Arduino environment.

Bodmer, 2022-05-12:
Master branch is the only version that supports the S3.

code-is-art, 2022-05-12:
ok, you are welcome to close it. It was not outright obvious to me until I started looking at the code and comparing which took me a bit to figure out. Thank you.

closed 2022-05-16 by Bodmer