Bodmer/TFT_eSPI #1831: TFT screen - touch problem.

Kekzeee1, 2022-05-14:
To minimise effort to resolve issues the following should be provided as a minimum:
1. A description of the problem and the conditions that cause it to occur
Everything is working, my HTML page and my screen are showing what it is supposed to show. But the touch functionality does not work, I'm wondering if it's the code that is the problem or something with the cables.
2. IDE (e.g. Arduino or PlatformIO)
Arduino
4. TFT_eSPI library version (try the latest, the problem may have been resolved!) from the Manage Libraries... menu
The Latest
6. Procesor, e.g RP2040, ESP32 S3 etc
ESP32
8. TFT driver (e.g. ILI9341), a link to the vendors product web page is useful too.*
ILI9341
9. Interface type (SPI or parallel)
SPI
```
// TFT_eSPI - Version: Latest 
#include <TFT_eSPI.h>

//xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
//xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
//
//  This is the firmware for an ESP32 module combined with a 2.8" TFT Touchdisplay (SPI)
//  The purpose of the device is to function as a door sign of your home office which
//  indicates if people may enter the room.
//  If you do not want to be disturbed the display shows a red sign with a user definable text.
//  If you may allow to enter the display shows an orange sign with a different text.
//  if you allow people to enter the display shows a green sign with another text.
//  In red or orange state people may ask to get permission to enter by pressing the touch display.
//  A pending request is indicated by a blue background color of the display.
//
//  The device is controlled via a web interface using its WLAN capabilities. It must be connected to
//  your local WLAN. You can use any browser (PC, Tablet, Smartphone) for the UI.
//  The ESP32 runs a web server with a html page showing three buttons to set green, red or orange state.
//  There is another box shown which indicates that someone has requested to enter.
//  In red or orange state you may start a countdown timer. Its start value is entered by pressing
//  one of three buttons on the web page: "5" and "10" are adding 5 or 10 minutes to the current
//  time left. "00" resets the timer.
//  the time left (in minutes) is also displayed on the TFT display (door sign). When the timer
//  elapses the sign is automatically switched in green state.
//  You can end the timer by clicking on one of the three state buttons.
//
//  You can edit the texts which are displayed on the web surface as well as those shown on the
//  TFT Display in the #define section. But the space is limited!
//  DO NOT FORGET TO EDIT THE WLAN CREDENTIALS!!!
//
//  Know problems:
//  The web server is very limited. It can only connect to a single client. If a client browser
//  is holding the connection instead of breaking it after receiving, no other client can connect
//  The device is not responsive before it can connect to a NTS server.
//
//  This sketch needs the following libraries to be installed (use the library manager):
//      ESP32 core
//      SPI, Wifi, WiFiUdp and a specifci ESP32 version of time.h (all part of the ESP32 package)
//      TFT_eSPI
//  The time.h of ESp32 core does have support for NTS over WLAN (time synchronisation) see google
//  for more information. I used this functionality to display time and date on the web GUI
//
//  The TFT library functions do need a local font file "MyFont.h" to be copied from the lib package
//  directory to the sketch diractory
//  Make sure all the display driver and pin comnenctions are correct by
//  editting the User_Setup.h file in the TFT_eSPI library folder:
//
//  pin on TFT    pin on ESP32 Dev Kit
//      VCC         V5 (+5 Volt)
//      GND         GND (DO NOT USE THE ONE NEXT TO V5!)
//      CS          G15 (chip select)
//      RESET       G4
//      D/C         G2  (device instruction)
//      SDI         G23 (SPI MOSI)
//      SCK         G18 (SPI clock)
//      LED         3V3 (3.3>V for backlight)
[... 417 lines cut ...]
Display height = 320

MOSI = GPIO 23
SCK = GPIO 18
TFT_CS = GPIO 15
TFT_DC = GPIO 0
TFT_RST = GPIO 2
TOUCH_CS = GPIO 21

Font GLCD loaded
Font 2 loaded
Font 4 loaded
Font 6 loaded
Font 7 loaded
Font 8 loaded
Smooth font enabled

Display SPI frequency = 27.00
Touch SPI frequency = 2.50
[/code]`
```

**All pins are connected like above except these.** 
_T_DO G19, but SDO is also connected to G19
T_DIN G23 but SDI is also connected to G23_

Bodmer, 2022-05-14:
T_DO and T_DIN must be connected, the touch feature will not work without them.

Kekzeee1, 2022-05-14:
Okay, is there a way to connect multiple pins to one using male to male wires or should I do this when I solder the wires?

Bodmer, 2022-05-14:
Use more holes in the breadboard!
[image]

Kekzeee1, 2022-05-14:
Thank you for all the help, everything is working now!

closed 2022-05-14 by Kekzeee1