Bodmer/TFT_eSPI #1833: Problem with a font on the latest TFT_eSPI (2.4.51)

MikeyMoMo, 2022-05-16:
This problem does not happen on 2.4.32 version of the library with the same code running.  The only difference is compiling with 2.4.32 or 2.4.51 installed.  I will try to installed releases between these two and see exactly when the problem occurred.  There are no error messages, just a problem with font rendering.

The problem only occurs with the number 1 from the attached font file.  It causes the lower half of the screen to be overwritten with the background color all the way to the bottom.  See the good and bad versions, below.  The difference is obvious and not good.  I don't want to overwrite the bottom half of the screen to keep flickering down to once per minute.  The colors are goofy to show how it is happening.  The other numbers do not cause a problem.  The font file has not changed in a long time.  This code has worked well for a very long time.  Now, every 1 that comes up bleeds.  The other numbers do not.

If I left anything out, please let me know.
Mike

The code:
```

  tft.loadFont(TickingTime190);  // Load a different font

  myBuildString = "";
  tempHour = homeTime.hour();
  if (tempHour > 9) {
    myBuildString = "1 ";
    tempHour -= 10;
  }
  myBuildString += String(tempHour) + " : ";
  myBuildString += String(homeTime.minute() / 10) + " ";
  myBuildString += String(homeTime.minute() % 10);

  tft.setTextPadding(dispWidth);
  tft.setTextColor(TFT_WHITE, TFT_RED);  // Foreground, Background
  tft.drawString(myBuildString, xCenter - 3, 8);
```

This pic is from the compile with the latest TFT_eSPI:
[image]

This pic is from the earlier library version and has run through the 1's being displayed already.  I should have shot that but missed it:
[image]

The font file converted for use in the program:
[Ticking Time Bomb.zip](https://github.com/Bodmer/TFT_eSPI/files/8699947/Ticking.Time.Bomb.zip)

MikeyMoMo, 2022-05-16:
Update: 
Here are the results of all available (to me) versions of TFT_eSPI.  The short version is that this is a new bug with 2.4.51.

// 2.4.51 bad
// 2.4.50 bad in other ways (overlaying text, colors wrong, images moving nervously) but does not do the crying colors like shown above in the top picture.  That seems to have just started with 2.4.51
// 2.4.45 good
// 2.4.43 good
// 2.4.42 good
// 2.4.32 good

So, I am going back to 2.4.45, the last good version and hope that a fix for this will appear soon.  I wish I could give more help but without any error message(s), I am stuck for anything else to say.

Thanks for the library.  I use it every day on one of more ESP boards and displays and others use it with my code, too.  I could not do without it!
Mike

Bodmer, 2022-05-16:
I have released v2.4.61 which should be picked up and available via the IDE library manager within 24 hrs.

closed 2022-05-16 by Bodmer