Bodmer/TFT_eSPI #1837: ili9486 arduino tft_espi esp8266  Shows the screen

glenlau228, 2022-05-21:
Using tft_espi to drive 3.5-foot tft (480x320) in esp8266, using ili9488, there is no reflection, using ili9486 can display the blurry screen, I have the same wiring, but it can display normally using another ILI9486_SPI(https://github.com/ZinggJM/ILI9486_SPI), please tell me, where is the problem, I how to solve.

 using  tft_espi  ili9486
[image]

using another ILI9486_SPI
[image]

tft lcd
[image]

lcd  connect   esp8266
T_IRQ		--
T_OUT		--
T_ON		--
T_CS		D1  -  GPIO 05 - 20
T_CLK		--
SDO 		--
LED		3v
SCK		D5  -  GPIO 14 - 5
SDI		D7  -  GPIO 13 - 7
D/C		D3  -  GPIO 00 - 18
RESET		D4  -  GPIO 02 - 17
CS		D8  -  GPIO 15 - 16
GND		gnd
VCC		3v

thank you very much,

glenlau228, 2022-05-21:
Read Write Test
--------------------------------------

 Pixel value written = 1
 Pixel value read    = FFFF
 ERROR                 ^^^^
 Pixel value written = 2
 Pixel value read    = FFFF
 ERROR                 ^^^^
 Pixel value written = 4
 Pixel value read    = FFFF
 ERROR                 ^^^^
 Pixel value written = 8
 Pixel value read    = FFFF
 ERROR                 ^^^^
 Pixel value written = 10
 Pixel value read    = FFFF
 ERROR                 ^^^^
 Pixel value written = 20
 Pixel value read    = FFFF
 ERROR                 ^^^^
 Pixel value written = 40
 Pixel value read    = FFFF
 ERROR                 ^^^^
 Pixel value written = 80
 Pixel value read    = FFFF
 ERROR                 ^^^^
 Pixel value written = 100
 Pixel value read    = FFFF
 ERROR                 ^^^^
 Pixel value written = 200
 Pixel value read    = FFFF
 ERROR                 ^^^^
 Pixel value written = 400
 Pixel value read    = FFFF
 ERROR                 ^^^^
 Pixel value written = 800
 Pixel value read    = FFFF
 ERROR                 ^^^^
 Pixel value written = 1000
 Pixel value read    = FFFF
 ERROR                 ^^^^
 Pixel value written = 2000
 Pixel value read    = FFFF
 ERROR                 ^^^^
 Pixel value written = 4000
 Pixel value read    = FFFF
 ERROR                 ^^^^
 Pixel value written = 8000
 Pixel value read    = FFFF
 ERROR                 ^^^^

glenlau228, 2022-05-21:
The display problem has been solved, it can now display normally, but the touch is not reflected

glenlau228, 2022-05-21:
Touch I only got one line T_CS,   
T_IRQ		--
T_OUT		--
T_ON		--
T_CS		D2  -  GPIO 04 - 19
T_CLK

Do you need to pick up anything else?

Bodmer, 2022-05-21:
Yes, they must be connected (except T_IRQ)P
T_IRQ  not connected
T_OUT connect to display SDO and processor TFT_MISO pin
T_DIN  connect to display SDI and processsor TFT_MOSI pin
T_CS connect to rocessor TOUCH_CS pin as defined insetup (D2 ?)
T_CLK connect to display SCK and processor  TFT_SCLK pin

closed 2022-05-21 by Bodmer

glenlau228, 2022-05-22:
Thank you very much for your help, now touch is ready to use. 

I just don't understand why the interfaces such as SDI, SDO, TFT_MISO, TFT_MOSI, T_CLK only need to be connected without configuring in the configuration file.