Bodmer/TFT_eSPI #1839: How to convert image into c code

rjrajbir, 2022-05-21:
Hello there,
I am using this lib with ESP32 wrover and 3.5 inch TFT display which has ILI9488 driver.
I am able to run **TFT_eSPI>examples>480 x 320>TFT_flash_jpg** default example with my display but when I am converting my image according to the example its not working.
[image]

This is the Image 

[car.zip](https://github.com/Bodmer/TFT_eSPI/files/8747868/car.zip)
and I have included this lib and used as below

`// Sketch to display images on a 480 x 320 ILI9486 Raspberry Pi 3.5" TFT (Waveshare design)
// which has a 16 bit serial interface based on 74HC04, 74HC4040 and 2 x 74HC4094 logic chips

// Renders images stored in an array in program (FLASH)JPEG images are stored in header files
// (see jpeg1.h etc)

// The sketch does not need the SD or sdFat libraries since it does not access an SD Card.

// As well as the TFT_eSPI library:
// https://github.com/Bodmer/TFT_eSPI
// the sketch need the JPEG Decoder library. This can be loaded via the Library Manager.
// or can be downloaded here:
// https://github.com/Bodmer/JPEGDecoder

//----------------------------------------------------------------------------------------------------

#include <SPI.h>
#include <TFT_eSPI.h>

TFT_eSPI tft = TFT_eSPI();

// JPEG decoder library
#include <JPEGDecoder.h>

// Return the minimum of two values a and b
#define minimum(a,b)     (((a) < (b)) ? (a) : (b))

// Include the sketch header file that contains the image stored as an array of bytes
// More than one image array could be stored in each header file.
#include "jpeg1.h"
#include "jpeg2.h"
#include "jpeg3.h"
#include "jpeg4.h"
#include "car.h"

// Count how many times the image is drawn for test purposes
uint32_t icount = 0;
//----------------------------------------------------------------------------------------------------
#define TFT_BL 22

//####################################################################################################
// Setup
//####################################################################################################
void setup() {
  Serial.begin(115200);
  tft.begin();
#ifdef TFT_BL
    pinMode(TFT_BL, OUTPUT);
    digitalWrite(TFT_BL, HIGH);
#endif  
}

//####################################################################################################
// Main loop
//####################################################################################################
void loop() {

  tft.setRotation(2);  // portrait
  tft.fillScreen(random(0xFFFF));

[... 166 lines cut ...]
//####################################################################################################
// Show the execution time (optional)
//####################################################################################################
// WARNING: for UNO/AVR legacy reasons printing text to the screen with the Mega might not work for
// sketch sizes greater than ~70KBytes because 16 bit address pointers are used in some libraries.

// The Due will work fine with the HX8357_Due library.

void showTime(uint32_t msTime) {
  tft.setCursor(0, 0);
  tft.setTextFont(1);
  tft.setTextSize(2);
  tft.setTextColor(TFT_WHITE, TFT_BLACK);
  tft.print(F(" JPEG drawn in "));
  tft.print(msTime);
  tft.println(F(" ms "));
  Serial.print(F(" JPEG drawn in "));
  Serial.print(msTime);
  Serial.println(F(" ms "));
}
`

Please help where I am doing wrong.
BR
Rajbir

Bodmer, 2022-05-21:
The image is positioned off screen. x,y is position of top left corner.
Use:
drawArrayJpeg(car, sizeof(car), 0, 0);

rjrajbir, 2022-05-23:
Still not working, My esp32 is resetting with this code

Bodmer, 2022-05-24:
The image array in car.h is not a jpeg file, so that is why it does not render. You must save the image as a jpeg then create an array from that file.

rjrajbir, 2022-05-24:
I have shared the image above and its JPEG only

Bodmer, 2022-05-24:
The original may be a jpeg but it has been converted to a bitmap array by whatever tool you used.

Follow the steps here and recreate the array.
https://github.com/Bodmer/TFT_eSPI/discussions/1843

closed 2022-05-24 by Bodmer

rjrajbir, 2022-05-24:
I converted the file using above method but, sorry still same issue, its resetting the MCU

Bodmer, 2022-05-24:
Your array is not in the correct format, you have to fix that. Try converting another smaller image. Check the first bytes in the array match the jpeg magic number signature bytes 0xFF, 0xD8, 0xFF,

These bytes are part of the header for a jpeg file.

Bodmer, 2022-05-24:
I checked the jpeg and it is in an unsupported jpeg format. Loading into MS Paint and saving again corrected this.

This is the resultant output array and this works fine.
[car.zip](https://github.com/Bodmer/TFT_eSPI/files/8765661/car.zip)

rjrajbir, 2022-05-25:
I wasn't aware that this is not supported Format, thank you for your help.