Bodmer/TFT_eSPI #1840: How to use a custom made GLCD font in TFT_eSPI?

Amitabh-wq, 2022-05-21:
Hi, Bodmer.
I converted a font called 'digitalism.ttf' to a C Array using MikroElektronika GLCD Font Creator 1.2.0.0. I want to use this GLCD font in a project that uses the TFT_eSPI library. I would like to know if this is possible and if you could give me a tip on how to do it. Attached are the original font and the one generated in C Array. Any hints will be welcome. Thanks.

[Digitalism7_13 GLCD.txt](https://github.com/Bodmer/TFT_eSPI/files/8748447/Digitalism7_13.GLCD.txt)
[digitalism.zip](https://github.com/Bodmer/TFT_eSPI/files/8748446/digitalism.zip)

Bodmer, 2022-05-21:
TFT_eSPI has no built in support for that font format.

See #1752

closed 2022-05-21 by Bodmer

Amitabh-wq, 2022-05-22:
Thank you for your great support.