Bodmer/TFT_eSPI #1842: ILI9488 touch spi communication issue / Calibration can´t complete

gitbow, 2022-05-23:
Hi Bodmer, first of all thanks very much for your great work.
Ili9488 works well as graphic controller, my problem is with the SPI touch interface 
on ESP32.  I tried different sketches and I can not complete the calibration run, it just
pulls over all steps, without allowing me to press each corner with the stylus.
When I run the TEST touch controller I get 
X 0   Y0   Z4096
......
I followed the standart procedure to edit usert_setup, I tried with 
conecting / disconecting SDD/MISO to  Touch-Dout ( TFT display continues
working well when SDD/MISO is without connection. I first wired it in parallel
to TDOUt. Now only TDOUT is wired to PIN19 on ESP32. Both clock signals
are wired together to D18.  Touch-CS is wired to D21 on the ESP32.
SDI/MOSI is still wired to T-DIN and both to pin D23 on ESP32

But there is no change, the calibration shows  calibration run... and very fast
goes through all 4 steps, without allowing me to touch the screen on the corners...
Can you tell me what else I can try ?

Thank you in advance, here is my  
User_setup.h 
// ###### EDIT THE PIN NUMBERS IN THE LINES FOLLOWING TO SUIT YOUR ESP32 SETUP   ######

// For ESP32 Dev board (only tested with ILI9341 display)
// The hardware SPI can be mapped to any pins

 #define TFT_MISO 19
 #define TFT_MOSI 23
 #define TFT_SCLK 18
 #define TFT_CS   15  // Chip select control pin
 #define TFT_DC    2  // Data Command control pin
 #define TFT_RST   4  // Reset pin (could connect to RST pin)
 //#define TFT_RST  -1  // Set TFT_RST to -1 if display RESET is connected to ESP32 board RST
// For ESP32 Dev board (only tested with GC9A01 display)
// The hardware SPI can be mapped to any pins
//#define TFT_MOSI 15 // In some display driver board, it might be written as "SDA" and so on.
//#define TFT_SCLK 14
//#define TFT_CS   5  // Chip select control pin
//#define TFT_DC   27  // Data Command control pin
//#define TFT_RST  33  // Reset pin (could connect to Arduino RESET pin)
//#define TFT_BL   22  // LED back-light
#define TOUCH_CS 21     // Chip select pin (T_CS) of touch screen
//#define TFT_WR 22    // Write strobe for modified Raspberry Pi TFT only

Bodmer, 2022-05-24:
Run the Test_Touch_Controller example. Look at the serial output when touching the screen. This will show if the interface is working. Nothing will appear on the display.

Check your display has the touch controller fitted, check wiring.

gitbow, 2022-05-24:
> Run the Test_Touch_Controller example. Look at the serial output when touching the screen. This will show if the interface is working. Nothing will appear on the display.
> 
> Check your display has the touch controller fitted, check wiring.

Thanks Bodmer, Touch is working now.  I found one of your suggestions to set down  Touch SPI frequency to
 #define SPI_READ_FREQUENCY    16000000
#define SPI_TOUCH_FREQUENCY  2500000

and it works ok. 

thanks for all. If you wish this issue can be deleted.

closed 2022-05-24 by Bodmer

summerfind, 2023-06-14:
Hi Bodmer.
my serial output don't have any changes when touched on ILI9488, seems it doesn't work at all, but it shown well for other example such like:  Free_Font_Demo.