Bodmer/TFT_eSPI #1851: Sample "Animated_Eyes_1" writes stream of messages "E (12049) gpio: gpio_set_level(226): GPIO output gpio_num error" to Serial Monitor

dheydeck, 2022-05-26:
Thank you for providing such a fantastic library, and for providing so much support (you seem to be handling a constant stream of issues). It must consume lots of your time.
It seems churlish to raise an issue with one of the examples, but I use the Animated Eyes as an adjunct to an analogue meter to display the temperature from a BME280 sensor. I have recreated the problem that I encountered using the sketch provided with the library.
To minimise effort to resolve issues the following should be provided as a minimum:
1. A description of the problem and the conditions that cause it to occur
Sample "Animated_Eyes_1" writes stream of messages:
```
E (12049) gpio: gpio_set_level(226): GPIO output gpio_num error
E (12064) gpio: gpio_set_level(226): GPIO output gpio_num error
E (12065) gpio: gpio_set_level(226): GPIO output gpio_num error
E (12080) gpio: gpio_set_level(226): GPIO output gpio_num error
E (12080) gpio: gpio_set_level(226): GPIO output gpio_num error
E (12096) gpio: gpio_set_level(226): GPIO output gpio_num error
```
To serial monitor.
Hardcoding TFT1_CS and TFT2_CS in config.h resolves problem:
```
// DISPLAY HARDWARE SETTINGS (screen type & connections) -------------------
#define TFT_COUNT 1        // Number of screens (1 or 2)
//#define TFT1_CS -1         // TFT 1 chip select pin (set to -1 to use TFT_eSPI setup) DGH
//#define TFT2_CS -1         // TFT 2 chip select pin (set to -1 to use TFT_eSPI setup) DGH
#define TFT1_CS 15         // TFT 1 chip select pin (set to -1 to use TFT_eSPI setup)   DGH
#define TFT2_CS 15         // TFT 2 chip select pin (set to -1 to use TFT_eSPI setup)   DGH
#define TFT_1_ROT 1        // TFT 1 rotation
#define TFT_2_ROT 1        // TFT 2 rotation
#define EYE_1_XPOSITION  0         // x shift for eye 1 image on display
#define EYE_2_XPOSITION  320 - 128 // x shift for eye 2 image on display
```
2. IDE (e.g. Arduino or PlatformIO)
Arduino IDE 1.8.19 (Windows Store 1.8.57.0)
3. TFT_eSPI library version (try the latest, the problem may have been resolved!) from the Manage Libraries... menu
TFT_eSPI library version 2.4.61
4. Board package version (e.g. 2.0.3) available from the Boards Manager... menu
esp32 2.0.3
5. Procesor, e.g RP2040, ESP32 S3 etc
ESP32 WROOM-32D
6. TFT driver (e.g. ILI9341), a link to the vendors product web page is useful too.
ILI9341
7. Interface type (SPI or parallel)
SPI

Plus further information as appropriate to the problem:
1. TFT to processor connections used
As per setup file- the sketch is working, other than the error messages
2. A zip file containing your setup file (just drag and drop in message window - do not paste in long files!)
[Setup999_Alt_ILI9341.zip](https://github.com/Bodmer/TFT_eSPI/files/8778553/Setup999_Alt_ILI9341.zip)
3. A zip file  containing a simple and complete example sketch that demonstrates the problem but needs no special hardware sensors or libraries.
Sample "Animated_Eyes_1" as provided with current library
4. Screen shot pictures showing the problem  (just drag and drop in message window)
[image]
(Note that the "65" in the stream of messages relates to the frame update rate)

Thank you once more for your time!

Bodmer, 2022-05-26:
This is a problem associated with the latest 2.0.3 ESP32 board package.

Board package versions 2.0.2 and earlier do not have this problem.

It appears there is a new software team working on the ESP32 board package and recent changes are breaking many useful Arduino libraries that are unlikely to get updated for one rogue board package.

The issue has been raised here, the answer is to include a debug message switch for this case but we will have to see what happens.

In the meantime your solution is a valid workaround.

dheydeck, 2022-05-26:
Thank you once mot, Bodmer. 
closing as this is an issue with the ESP32 Board Package.

dheydeck, 2022-05-26:
Issue with the ESP32 Board Package, Version 2.0.3.

closed 2022-05-26 by dheydeck

Bodmer, 2022-05-26:
It has been raised with the ESP32 team here:
https://github.com/espressif/arduino-esp32/issues/6737

It seems to be causing havoc with other libraries.

I will watch and see if it gets fixed. Since by default the core debug level is set to "none" you would not expect the USB serial stream to get flooded with debug messages.

dheydeck, 2022-05-27:
Thank you once more.