Bodmer/TFT_eSPI #1853: ESP32, ST7789 and Animated_dial sketch

NDC-SC, 2022-05-26:
The Animated_dial sketch only displays the dial without the moving needle, when displayed on a 240x240 pixel ST7789 LCD screen powered by an ESP322 DEVKIT DOIT module.

The TFT_eSPI library V2.4.61 was installed in the Arduino IDE V 1.8.19.
The default ESP32 SPI connections in the Setup24_ST7789 file are used.

Can you please advise as to how to resolve the problem.
Thanks
Neil

Bodmer, 2022-05-26:
This is a problem associated with the latest 2.0.3 ESP32 board package.

Board package versions 2.0.2 and earlier do not have this problem.

It appears there is a new software team working on the ESP32 board package and recent changes are breaking many useful Arduino libraries that are unlikely to get updated for one rogue board package.

The issue has been raised here, the answer is to include a debug message switch for this case but we will have to see what happens.

In the meantime your solution is a valid workaround.

Bodmer, 2022-05-26:
Correct answer to wrong issue!

Bodmer, 2022-05-26:
This happens if the TFT CGRAM cannot be read.  Run the TFT_ReadWrite_Test to see if the dispays read works.

NDC-SC, 2022-05-26:
I've deleted the ESP32 folder and installed V2.0.2.
The needle is still missing in the Animated_dial sketch, even with #define TFT_CS 15 , instead of #define TFT_CS  -1.

With the TFT_ReadWrite_Test sketch, the ST7789 screen is purple and the output (for a range of pixel writen values, is 
 Pixel value written = 200
 Pixel value read    = 0
 ERROR                 ^^^^

NDC-SC, 2022-05-27:
Bodmer, do you know why the combination of ESP32 V1.0.6 and ST7899 supports sprites, but there are problems with sprites in V 2.0.N with the ST7899?
Is there a work around ?

Bodmer, 2022-05-27:
The issue is not with the sprites unless there is another problem I am not aware of.

The sketch works by reading ther area of the screen where the needle will be plotted, then when the needle moves it replaces the old area, reads the screen where for the new needle position and then draws the needle again. his repeats for every nedle movement. If the screen cannot be read then the sketch typically reads 0 for the pixels so the needle zone gets wiped with black.

The TFT_ReadWrite_Test tells me the pixel colour is always read as black. There can be a number of reasons for this so I need more information on the display. For example does it have SDI (MOSI) and SDO (MISO) pins connected?  The MOSI line writes to the display and the MISO line is used to read the pixels stored in the screen memory.

I am guessing there is no MISO pin, but the display might have a SDA pin which is bi-directional, in that case there is a way to make the library run in half duplex mode and this may work.  To use half duplex mode the following must be included in the setup file:

#define TFT_SDA_READ      // This option is for ESP32 ONLY

NDC-SC, 2022-05-27:
Thank you for your reply. The Animated-dial sketch was highlighted as an example of a sketch not working when a Sprite was included. From your explanation, the immediate problem is the lack of an MISO pin on the ST7899 LCD screen, rather than specifically due to use of a Sprite.
As you suggested, the #define TFT_SDA_READ instruction was included in the SetUp24_ST7789 file, but the TFT_ReadWrite_Test sketch still produces read screen errors.  Although pixel write 1 or 800 frequently produce a PASS. The screen display functions correctly as the screen displays Magenta (0xF81F). The SPI read frequency was also reduced to 6MHz, but there was no improvement on the read screen errors.

However, there are sketches that run perfectly well, although there is flicker due to the slower display speed, when images are displayed with "tft.display command", while the same sketch does not run when "img.display command" is used, with "img." referring to a Sprite. There is something inherently causing the sketch to fail when the image is to be displayed as a Sprite.

NDC-SC, 2022-05-27:
For example, the attached sketch displays a background image and then adds foreground images. With ESP32 V 1.0.6 the sketch runs with the Sprite, but the sketch does not run properly with V 2.0.0, as only the background image is displayed without the foreground images. If the "img.display command" are simply changed to "tft.display command" then the sketch runs with V.2.0.0, but with flicker. The sketch with "img.display command" commented out and the "tft.display command" are included in the attached folder
[example.zip](https://github.com/Bodmer/TFT_eSPI/files/8789191/example.zip)

Bodmer, 2022-05-27:
The 240 x 240 sprite will require a lot of RAM (115.2 kbytes) and it is likely a smaller sprite will work fine. Early board packages only allowed sprites that consumed up to ~90kbytes.

At some point the RAM partitioning was changed in the board package and this allows slightly larger sprites to be used. I ran the example sketch with sprites using board package 2.0.1 and it worked fine, but 2.0.0 failed because there was not enough RAM for the sprite. This seems to have been fixed in later versions.

CreateSprite returns a nullptr if the memory request failed so, for example, the following can be used:

```
  uint16_t* ptr = (uint16_t*)img.createSprite(width, height);      // full screen Sprite
  if (ptr == nullptr) Serial.println("Sprite not allocated!");
  else Serial.println("Sprite allocated!");
```

Of course this still does not explain why there is no needle drawn in the Animated dial example. I have just tested the Animated dial example with board package 2.0.0 and it worked as expected on a TFT that cannot be read, i.e. needle is drawn but dial area is swept to black.

NDC-SC, 2022-05-28:
Bodmer, Thanks for taking the time to explore the problem. So it's an issue with the amount of available RAM, that is ESP32 version dependent. The size of the data arrays determines whether or not there is sufficient RAM available to allocate the Sprite. In the example sketch, the threshold array size, corresponding to the Sprite being allocated RAM or not, is between 150 and 200, with ESP32 V 2.0.1. Your suggestion about parameterising the img.createSprite command was very helpful.
Thanks again.
Neil

closed 2022-05-29 by Bodmer

PeterMikeSmooth, 2022-08-17:
Hello Dodmer 

Tried Animated_dial sketch as well on ESP-C3-MINI dev kit and needle is missing as well. TFT_eSPI 2.4.72 on arduino 1.8.12.

Here is my conf : 
`#define TFT_MISO -1
#define TFT_MOSI  6
#define TFT_SCLK  4
#define TFT_CS    7  // Chip select control pin
#define TFT_DC   18  // Data Command control pin
#define TFT_RST  19  // Reset pin (could connect to RST pin)`

Half-duplex used : 
`#define TFT_SDA_READ`

Also I needed to define theses in the User_Setup.h : 
`#define VSPIQ_IN_IDX                  64
#define VSPIQ_OUT_IDX                 64
#define VSPID_IN_IDX                  65
#define VSPID_OUT_IDX                 65`

Memory request seems ok according to your suggestions : 
`ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x438
load:0x403ce000,len:0x918
load:0x403d0000,len:0x24e4
SHA-256 comparison failed:
Calculated: 080c5cb68a075ced55f248b97bca965e3e5bd5da80a64e34e6a1638f89d6f64e
Expected: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
Attempting to boot anyway...
entry 0x403ce000
Main  Sprite allocated!
Needle  Sprite allocated!
`

With the TFT_ReadWrite_Test sketch, the ST7789 screen is purple and the output gives : 
` Pixel value written = 200
 Pixel value read    = 40
 ERROR                 ^^^^
 Pixel value written = 400
 Pixel value read    = 800
 ERROR                 ^^^^
 Pixel value written = 800
 Pixel value read    = 1
 ERROR                 ^^^^
 Pixel value written = 1000
 Pixel value read    = 40
 ERROR                 ^^^^
 Pixel value written = 2000
 Pixel value read    = 800
 ERROR                 ^^^^
 Pixel value written = 4000
 Pixel value read    = 1
 ERROR                 ^^^^
 Pixel value written = 8000
 Pixel value read    = 40
 ERROR                 ^^^^
 Pixel value written = 1
 Pixel value read    = 800
 ERROR `

Can you please advise on this issue ? 
Thanks
Peter