Bodmer/TFT_eSPI #1862: ESP32C3 loop reboot with  last esp32 core + last rev

0x0fe, 2022-06-01:
With last ESP32 (C3) core and last rev (2.4.61)  the esp32c3 reboot sin loop, with no error in the log (set to verbose). I could reproduce on different esp32c3 platform, i also changed the pins defines to make sure it was not bootstrap related (it is not). AS soon as tft.begin(); is called it will reboot, if the function is not called, no problem. A printf inside the begin function does never prints, so somehow it crashez just by calling the function.
I tested the factory SPI driver with the TFT and it works fine (GC9A01), this error only occurs with TFT_eSPI.

Bodmer, 2022-06-02:
Try the latest Github master v2.4.70

0x0fe, 2022-06-05:
ok thank you i will give a try, meanwhile what i did was to extract the lower leve functions that i needed for the aplication, which worked (partly). A related question is DMA, is the dead code related to DMA working or is it from tensilica ESP32?

arjena, 2022-06-09:
Hi,
I have the same problem. You suggest to use v2.4.70 but where can I find it? The latest release version is v2.4.61 and when I use the raw code (not a very experienced GitHubber, what I mean here is using the 'download as .zip' from the code tab) I get newer files but these also crash the esp.

arjena, 2022-06-16:
In Platformio V2.4.70 is now the default. I'm using a ESP32-cam (AI-Thinker). On esp platform 3.50 there is no problem, but when I switch to 4.30 (or 4.40) I get the reboot loop. But I need the newer esp platform for ssl features that do not work in 3.50.
I am trying to compile the sprite example you wrote for a GitHub member "jpg_sprite_toSprite_865". This also uses TJpg_decoder for which I use V1.0.5 (as the default in Platformio is still 0.20). 
The reboot starts as soon as tft_begin is called:

Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x4008ba3c  PS      : 0x00060a33  A0      : 0x8008d754  A1      : 0x3ffb26a0  
A2      : 0xae000000  A3      : 0xb33fffff  A4      : 0x0000cdcd  A5      : 0x00060a23  
A6      : 0x00060a20  A7      : 0x0000abab  A8      : 0x0000abab  A9      : 0xffffffff  
A10     : 0x00000000  A11     : 0x00000001  A12     : 0x00000032  A13     : 0x00001389  
A14     : 0x6e800000  A15     : 0x003fffff  SAR     : 0x00000010  EXCCAUSE: 0x0000001d  
EXCVADDR: 0xae000000  LBEG    : 0x40087539  LEND    : 0x40087549  LCOUNT  : 0xffffffff  

Backtrace:0x4008ba39:0x3ffb26a00x4008d751:0x3ffb26e0 0x4008d9c1:0x3ffb2700 0x40083542:0x3ffb2720 0x40083555:0x3ffb2750 0x4008375c:0x3ffb2770 0x400d917f:0x3ffb2790 0x400d54ba:0x3ffb27b0 0x400d55cb:0x3ffb27d0 0x400d164b:0x3ffb27f0 0x400d98d2:0x3ffb2820 

ELF file SHA256: 0000000000000000

Rebooting...

'Funny' thing, in my 'own' test code, which is basically a stripped down version of yours, the reboot starts when I call createSprite. So tft_begin does not pose a problem there. The dump looks pretty much the same though...

Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x4008ba3c  PS      : 0x00060a33  A0      : 0x8008d754  A1      : 0x3ffb26a0  
A2      : 0xaf000000  A3      : 0xb33fffff  A4      : 0x0000cdcd  A5      : 0x00060a23  
A6      : 0x00060a20  A7      : 0x0000abab  A8      : 0x0000abab  A9      : 0xffffffff  
A10     : 0x00000000  A11     : 0x00000001  A12     : 0x000000f0  A13     : 0x00012c01  
A14     : 0x6f800000  A15     : 0x003fffff  SAR     : 0x00000010  EXCCAUSE: 0x0000001d  
EXCVADDR: 0xaf000000  LBEG    : 0x40087539  LEND    : 0x40087549  LCOUNT  : 0xffffffff  

Backtrace:0x4008ba39:0x3ffb26a00x4008d751:0x3ffb26e0 0x4008d9c1:0x3ffb2700 0x40083542:0x3ffb2720 0x40083555:0x3ffb2750 0x4008375c:0x3ffb2770 0x400d8cff:0x3ffb2790 0x400d53b2:0x3ffb27b0 0x400d54c3:0x3ffb27d0 0x400d16a5:0x3ffb27f0 0x400d9452:0x3ffb2820 

ELF file SHA256: 0000000000000000

Rebooting...

Any ideas? I found something about using psram (of which this esp32 has 4MB), but turning it off (`spr.setAttribute(PSRAM_ENABLE, false);`) only works if I create a small sprite. Anything bigger than 4x50 still causes a reboot.

Bodmer, 2022-06-16:
I have tested an ESP32 C3 with an ILI9341 display with TFT_eSPI v 2.4.70.

I am using the Arduino IDE and the ESP32 Arduino 2.0.3 board package which is based on ESP-IDF 4.4.1. Library examples without a problem.

I do not use PlatformIO so unfortunately do not know why it is failing for you..

Bodmer, 2022-06-16:
I managed to find the jpg_sprite_toSprite_865 sketch, it uses DMA which is not currently supported on the C3. Try other examples.

The jpg_sprite_toSprite_865 will work OK if you adapt to avoid DMA.

Comment out this line:
```
  //tft.initDMA();
```
Change this line:
```
tft.pushImageDMA(x, y, w, h, bitmap, dmaBuf);
```
to:
```
tft.pushImage(x, y, w, h, bitmap);
```

@0x0fe: I assume the issue is resolved now by ising the latest TFT_eSPI. Open another issue if not.

closed 2022-06-16 by Bodmer

arjena, 2022-06-16:
Ah, unfortunately no, DMA is not used in the jpg_sprite_toSprite_865 sketch, at least not in the copy I was using. 
I will try it on another esp32 if I can find the time. But for now I will have to use the older esp platform core to get my project going. It uses Telegram and that works much better in the latest esp core, but without the display the whole thing is useless so I'll settle for less reliable ssl with a working display. I'll get back to you if I find anything useful concerning my crashes.

arjena, 2022-06-17:
Succes!(?) 
First of all, I must apologise for my manners. Re-reading my previous post(s) it may come across as criticism, which is far from what I mean. I really do appreciate all the hard work people like you put in projects. So thank you for sharing your work with us.  
Now back to my issue.
I am working on a breadboard and a couple of wires from my esp to the display came loose. I accidentally switched them when I put them back which also caused a reboot loop. That got me thinking that it could be related to my  reboot loops under the latest core. So I started reading about the esp32 and the 'special functions' some of the pins have. The reset pin was connected to gpio_16, but according to the datasheet:   `Pins IO16, IO17, CMD, CLK, SD0 and SD1 are used to connect the embedded flash, and can not be used for other purposes.` Why it is exposed as a gpio pin if it cannot be used for anything is beyond me...

Anyway I changed the pin definitions to:
`#define TFT_CS   12  // Chip select control pin
#define TFT_RST  13  // Reset pin (could connect to RST pin)
#define TFT_DC   15  // Data Command control pin
#define TFT_MOSI 14
#define TFT_SCLK 2
`
Now the display functions perfectly with the latest esp core. Too early to say my issue is solved, but this is a good start.
So thank you again for your excellent work and your patience. On with my project.