Bodmer/TFT_eSPI #1863: DMA not supported on esp32-s3

code-is-art, 2022-06-01:
DMA is not supported on esp32-s3. I think the issue is that [s3 uses SPI2 and SPI3 general purpose SPI controllers plus dam_chan set to SPI_DMA_CH_AUTO](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-reference/peripherals/spi_master.html). SPI0 and SPI1 are used internally to access the ESP32-S3’s attached flash memory. Could it be as simple as setting the `DMA_CHANNEL` to `SPI_DMA_CH_AUTO` and the `spi_host_device_t spi_host` to `SPI2_HOST`?

Bodmer, 2022-06-02:
OK, have you tried it? I am busy with other activities so it will be a few weeks before I can try it.

copyCATandpasteDOG, 2022-08-13:
Since i updated the esp32 package board to the latest 2.04,i was impossible to
compile anything with esp32 and TFT e_SPI,even changing the package down to 1.06,
i notice there was SPIFFS notifications in the verbose,so i downloaded the last
esp32 library and updated the SPIFFS library,and then...BINGO!!
All is working now :)

Bodmer, 2022-08-13:
Not sure what is happening there. I am using 2.0.4 and have no issues. SPIFFS notification come from the board package so I can't stop them.

copyCATandpasteDOG, 2022-08-13:
Talking about the notifications was to write about the compilation verbose errors
before uploading from arduino AVR
All i can say is that i had an old SPIFFS version before i updated the package to 2.04
my SPIFFS was dated from 2020 !
But while my problems occurs,i made several clean installs of arduino AVR and
deleted  all,exept the libs and sketches of course,and the problems were all the
same.
So i took the last version of SPIFFS and by security,took only the 2.02 version of
the board package version,knowing there is people having problems with the 2.04.
So i'm currently updating to 2.04 to verify it,
I will tell you if that works :)

copyCATandpasteDOG, 2022-08-13:
Just finished compile and upload with the 2.04 version and everything's fine :)

Bodmer, 2022-08-14:
Great, glad it is working now. Note that DMA support has not been added for S3 yet.

mrfaptastic, 2022-12-12:
Hi @Bodmer - Is it only the S3 that doesn't have DMA support, or is it all ESP32's? 

Happy to contribute here given the learnings I've gone through with this, and what I've learned with the way https://github.com/lovyan03/LovyanGFX/ does it,

Bodmer, 2022-12-13:
DMA is supported for the original ESP32, RP2040 and some STM32 processors.
Modifications to support the S3 have been hinted at in #2233 but these have not been implemented yet.
A relatively simple DMA approach has been taken to avoid complexity, simplify multiple processor support and permit memory usage to be controlled. It you wish to implement the currently adopted approach for the S3 then that would be welcome. It is essential to avoid adding any additional functionality.

Bodmer, 2023-02-02:
Added for S3.

closed 2023-02-02 by Bodmer

maciekish, 2023-11-02:
Sorry for necroing this, does this mean DMA is supported on the S3 now? I believe readme says no so i never tried it.

Bodmer, 2023-11-02:
DMA is supported on S3 with 16bit SPI displays. See ReadMe for the interface compatibility table and the display driver list.