Bodmer/TFT_eSPI #1864: Platform compile error "undefined reference to `TFT_eSPI::initDMA(bool)'"

MiroHibler, 2022-06-01:
I'm trying to compile a simple example program for Wemos/Espduino D1 R32 (ESP32) board with ILI9341 as Arduino shield (with modification as per README.md file), but I'm getting this error:

```
/Users/miro/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/esp32dev/src/main.cpp.o:(.literal._Z5setupv+0x74): undefined reference to `TFT_eSPI::initDMA(bool)'
/Users/miro/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/esp32dev/src/main.cpp.o:(.literal._Z4loopv+0x2c): undefined reference to `TFT_eSPI::dmaWait()'
/Users/miro/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/esp32dev/src/main.cpp.o:(.literal._Z4loopv+0x30): undefined reference to `TFT_eSPI::pushImageDMA(int, int, int, int, unsigned short*, unsigned short*)'
/Users/miro/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/esp32dev/src/main.cpp.o: in function `setup()':
/Users/miro/Electronics/Espressif/ESP32/3D Graphics/ESP32 TGX Demo/src/main.cpp:82: undefined reference to `TFT_eSPI::initDMA(bool)'
/Users/miro/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/esp32dev/src/main.cpp.o: in function `loop()':
/Users/miro/Electronics/Espressif/ESP32/3D Graphics/ESP32 TGX Demo/src/main.cpp:206: undefined reference to `TFT_eSPI::dmaWait()'
/Users/miro/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /Users/miro/Electronics/Espressif/ESP32/3D Graphics/ESP32 TGX Demo/src/main.cpp:217: undefined reference to `TFT_eSPI::pushImageDMA(int, int, int, int, unsigned short*, unsigned short*)'
collect2: error: ld returned 1 exit status
*** [.pio/build/esp32dev/firmware.elf] Error 1
```

Here's my `User_Setup.h` file:

```
// See SetupX_Template.h for all options available
// #define USER_SETUP_ID 14
#define USER_SETUP_INFO "User_Setup"

#define ILI9341_DRIVER
#define TFT_PARALLEL_8_BIT

// The ESP32 and TFT the pins used for testing are:
#define TFT_CS   33  // Chip select control pin (library pulls permanently low
#define TFT_DC   15  // Data Command control pin - must use a pin in the range 0-31
#define TFT_RST  32  // Reset pin, toggles on startup

#define TFT_WR    4  // Write strobe control pin - must use a pin in the range 0-31
#define TFT_RD    2  // Read strobe control pin

#define TFT_D0   12  // Must use pins in the range 0-31 for the data bus
#define TFT_D1   13  // so a single register write sets/clears all bits.
#define TFT_D2   26  // Pins can be randomly assigned, this does not affect
#define TFT_D3   25  // TFT screen update performance.
#define TFT_D4   17
#define TFT_D5   16
#define TFT_D6   27
#define TFT_D7   14

#define LOAD_GLCD   // Font 1. Original Adafruit 8 pixel font needs ~1820 bytes in FLASH
#define LOAD_FONT2  // Font 2. Small 16 pixel high font, needs ~3534 bytes in FLASH, 96 characters
#define LOAD_FONT4  // Font 4. Medium 26 pixel high font, needs ~5848 bytes in FLASH, 96 characters
#define LOAD_FONT6  // Font 6. Large 48 pixel font, needs ~2666 bytes in FLASH, only characters 1234567890:-.apm
#define LOAD_FONT7  // Font 7. 7 segment 48 pixel font, needs ~2438 bytes in FLASH, only characters 1234567890:.
#define LOAD_FONT8  // Font 8. Large 75 pixel font needs ~3256 bytes in FLASH, only characters 1234567890:-.
#define LOAD_GFXFF  // FreeFonts. Include access to the 48 Adafruit_GFX free fonts FF1 to FF48 and custom fonts

#define SMOOTH_FONT
```

Here's my platformio.ini file:

```
[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino
lib_deps =
	bodmer/TFT_eSPI@^2.3.81
```

If I comment out the `#define TFT_PARALLEL_8_BIT` line in the `User_Setup.h` file, the compile finishes successfully, but the display is empty (as expected).

MiroHibler, 2022-06-02:
@Bodmer Thank you for the clarification :)

It would be nice to have that somewhere documented (I didn't see it yet...).

Anyway, I decided to switch to SPI display as parallel takes to much IO pins and I need a few more for other purposes :)

Bodmer, 2022-06-02:
The ReadMe does explain the DMA constraints but I realise that these details can easily be missed.