Bodmer/TFT_eSPI #1867: WT32-SC01 with LCD ST7796, TFT_espi > readRect not working

elC1a, 2022-06-04:
Hello
Thanks for TFT_espi!
WT32-SC01 with LCD ST7796, TFT_espi > readRect not working.
Arduino code below.
Is something wrong with my code or my setup?
I like the WT32-SC01 device and a solution for the problem would be very, very nice.

// ***************************************************************************************
// Content                    - WT32-SC01 with LCD ST7796, TFT_espi > readRect not working
// ---------------------------------------------------------------------------------------
// Device WT32-SC01           - http://myosuploads3.banggood.com/products/20220111/20220111020446WT32-SC01DataSheetV3.3.pdf
// LCD Device ST7796          - https://cdn.hackaday.io/files/1625926956336128/ST7796S.pdf
// Touch Device FT6336U       - https://focuslcds.com/content/FT6236.pdf
// FT6336U library            - https://github.com/aselectroworks/Arduino-FT6336U
// TFT_eSPI library           - https://github.com/Bodmer/TFT_eSPI
// ***************************************************************************************
#include <TFT_eSPI.h>                       // TFT_espi Library
TFT_eSPI tft = TFT_eSPI();

uint16_t* tft_buffer;

void setup()
{
  // Init
  tft.init();
  tft.setRotation(1);
  tft.fillScreen(0x001F);
  
  // Create content to read from screen
  tft.setTextColor(0xFFFF);
  tft.setTextSize(2);
  tft.fillRect(200, 80, 96, 48, 0x03E0);
  tft.setCursor(208, 96, 1);
  tft.print("Example");

  // Create buffer with specific size
  tft_buffer = (uint16_t*) malloc( ((96) + 8) * ((48) + 8) * 2 );

  // Read rectangle from screen to buffer
  tft.readRect(200, 80, 96, 48, tft_buffer);

  // Paste back rectangle from buffer to screen
  tft.pushRect(200, 180, 96, 48, tft_buffer);
}

void loop()
{

}

Setup201_WT32_SC01.h below.

//                            USER DEFINED SETTINGS
//   Set driver type, fonts to be loaded, pins used and SPI control method etc
//
//   See the User_Setup_Select.h file if you wish to be able to define multiple
//   setups and then easily select which setup file is used by the compiler.
//
//   If this file is edited correctly then all the library example sketches should
//   run without the need to make any more changes for a particular hardware setup!
//   Note that some sketches are designed for a particular TFT pixel width/height
#define USER_SETUP_ID 201

// LCD Device: ST7796
#define ST7796_DRIVER

//#define TFT_WIDTH  480
//#define TFT_HEIGHT 320

#define TFT_BACKLIGHT_ON HIGH

#define USE_HSPI_PORT

#define TFT_MISO 12
#define TFT_MOSI 13
#define TFT_SCLK 14
#define TFT_CS   15
#define TFT_DC   21
#define TFT_RST  22
#define TFT_BL   23

// Touch Device: FT6336U
#define PIN_SDA  18
#define PIN_SCL  19
//#define TFT_SDA_READ      // This option is for ESP32 ONLY, tested with ST7789 and GC9A01 display only
#define TOUCH_CS -1 // Chip select pin (T_CS) of touch screen

#define LOAD_GLCD   // Font 1. Original Adafruit 8 pixel font needs ~1820 bytes in FLASH
#define LOAD_FONT2  // Font 2. Small 16 pixel high font, needs ~3534 bytes in FLASH, 96 characters
#define LOAD_FONT4  // Font 4. Medium 26 pixel high font, needs ~5848 bytes in FLASH, 96 characters
#define LOAD_FONT6  // Font 6. Large 48 pixel font, needs ~2666 bytes in FLASH, only characters 1234567890:-.apm
#define LOAD_FONT7  // Font 7. 7 segment 48 pixel font, needs ~2438 bytes in FLASH, only characters 1234567890:-.
#define LOAD_FONT8  // Font 8. Large 75 pixel font needs ~3256 bytes in FLASH, only characters 1234567890:-.
#define LOAD_GFXFF  // FreeFonts. Include access to the 48 Adafruit_GFX free fonts FF1 to FF48 and custom fonts

#define SMOOTH_FONT

// SPI frequency for TFT writes
// #define SPI_FREQUENCY  10000000
// #define SPI_FREQUENCY  20000000
#define SPI_FREQUENCY  27000000
// #define SPI_FREQUENCY  40000000
// #define SPI_FREQUENCY  80000000

// Optional reduced SPI frequency for reading TFT
#define SPI_READ_FREQUENCY  6000000
//20000000

Bodmer, 2022-06-04:
A bi-directional SDA pin interface to the TFT is not supported when the HSPI port is used.

If the VSPI port is used then
#define TFT_SDA_READ
In the setup file invokes bi-directional interface reads.

closed 2022-06-04 by Bodmer

elC1a, 2022-06-04:
Thanks for the answer.
Uncomment in the above Setup201_WT32_SC01.h
#define TFT_SDA_READ
isn't working. The same black rectangle.
Is there something else i should do?

elC1a, 2022-06-05:
// ***************************************************************************************
// WT32-SC01, LCD ST7796      - Screen in Sprite to restore screen content if tft.readRect is not working
// ---------------------------------------------------------------------------------------
// Device WT32-SC01           - http://myosuploads3.banggood.com/products/20220111/20220111020446WT32-SC01DataSheetV3.3.pdf
// LCD Device ST7796          - https://cdn.hackaday.io/files/1625926956336128/ST7796S.pdf
// Touch Device FT6336U       - https://focuslcds.com/content/FT6236.pdf
// FT6336U library            - https://github.com/aselectroworks/Arduino-FT6336U
// TFT_eSPI library           - https://github.com/Bodmer/TFT_eSPI
// ***************************************************************************************
#include <TFT_eSPI.h>                             // TFT_eSPI Library
TFT_eSPI tft = TFT_eSPI();
TFT_eSprite spriteScreen = TFT_eSprite(&tft);     // Sprite screen object
TFT_eSprite spriteButton = TFT_eSprite(&tft);     // Sprite button object

int16_t   screenWidth   = 480;
int16_t   screenHeight  = 320;
int16_t   buttonWidth   = 96;
int16_t   buttonHeight  = 48;
int16_t   angle, min_x, min_y, max_x, max_y;

void setup()
{
  // Init
  tft.init();
  tft.setRotation(1);
  tft.setPivot(screenWidth / 2, screenHeight / 2);

  // Build screen with content
  tft.fillScreen(0x001F);                           // Fill color blue
  tft.setTextColor(0xFFFF);                         // Text color white
  tft.setTextSize(2);
  tft.fillRect(192, 80, 96, 48, 0x03E0);            // Fill color green
  tft.setCursor(198, 88, 2);
  tft.print("Screen");

  // Create sprite screen with content, exactly the same screen content  
  spriteScreen.setColorDepth(16);
  spriteScreen.createSprite(screenWidth, screenHeight);
  spriteScreen.fillSprite(0x001F);                    // Fill color blue
  spriteScreen.setTextColor(0xFFFF);                  // Text color white
  spriteScreen.setTextSize(2);
  spriteScreen.fillRect(192, 80, 96, 48, 0x03E0);     // Fill color green
  spriteScreen.setCursor(198, 88, 2);
  spriteScreen.print("Screen");

  // Create sprite button with content
  spriteButton.setColorDepth(16);
  spriteButton.createSprite(buttonWidth, buttonHeight);
  spriteButton.fillSprite(0xFFE0);                    // Fill color yellow
  spriteButton.setPivot(buttonWidth / 2, buttonHeight / 2); // Set pivot point in sprite
  spriteButton.setTextColor(0x0000);                  // Text color black
  spriteButton.setTextSize(2);
  spriteButton.setCursor(8, 8, 2);
  spriteButton.print("Button");
}

void loop()
{
  delay(3000);
  
  // Draw button and get position and size
  angle = random(360);                                // random angle
  spriteButton.pushRotated(angle);                    // no transparency
  spriteButton.getRotatedBounds(angle, &min_x, &min_y, &max_x, &max_y);
  delay(3000);
  
  // Restore screen
  spriteScreen.pushSprite((int32_t)min_x, (int32_t)min_y, (int32_t)min_x, (int32_t)min_y, (int32_t)max_x, (int32_t)max_y);
}

elC1a, 2022-06-14:
**At the end i think reading from LCD isn't supported by the WT32-SC01 device.**