Bodmer/TFT_eSPI #1870: ESP8266-12E + ST7789 Backlight Only

CC-4477, 2022-06-06:
I've set up my ESP-8266-12E NodeMCU board with a [DFR0649 LCD module](https://www.dfrobot.com/product-2072.html) (uses the ST7789 driver as per [this datasheet](https://wiki.dfrobot.com/1.54%20Inches%20240_240_IPS_TFT_LCD_Display_with_MicroSD_Card_Breakout_SKU_DFR0649#target_5)). However, it seems that I can't get any graphical output on the LCD, and all I see is the backlight (which turns off, as expected, when connected to GND). 

I'm using TFT+eSPI v2.4.61, with the Arduino IDE. My board package version is 3.0.2. The TFT driver I have is ST7789, and I'm interfacing it using SPI.

Here are some images of my setup: 
**LCD**  **ESP**
VCC         3.3V
GND       GND
CS           D8
MOSI      D7
MISO      D6
SCLK       D5
RES         D4
DC          D3

[image]
[image]
[image]

Here's my User_Setup.h:
[User_Setup.zip](https://github.com/Bodmer/TFT_eSPI/files/8842522/User_Setup.zip)

Here's the sketch I'm trying to run: https://github.com/Bodmer/TFT_eSPI/blob/master/examples/Generic/Gradient_Fill/Gradient_Fill.ino

Bodmer, 2022-06-14:
Looks OK apart from the SPI clock speed, drop that to 10MHz. The display board has a logic level shifter so it is unlikely to work at high clock speeds.

Also adapt and run the examples provided in the display Wiki and make sure all solder connections are good.

CC-4477, 2022-06-14:
> Looks OK apart from the SPI clock speed, drop that to 10MHz. The display board has a logic level shifter so it is unlikely to work at high clock speeds.
> 
> Also adapt and run the examples provided in the display Wiki and make sure all solder connections are good.

I'm travelling currently, I'll be able to test this around a week from now. I think it may be the soldering, so I'll disconnect all the pins, clean up the solder, and redo the process. where can I find the examples?

Bodmer, 2022-06-18:
The examples are in the product Wiki.

closed 2022-06-18 by Bodmer

CC-4477, 2022-06-20:
> Looks OK apart from the SPI clock speed, drop that to 10MHz. The display board has a logic level shifter so it is unlikely to work at high clock speeds.
> 
> Also adapt and run the examples provided in the display Wiki and make sure all solder connections are good.

changed this and still nothing but the backlight. I've added `#define TFT_SPI_MODE SPI_MODE3` after creating the TFT object, but nothing's changed.