Bodmer/TFT_eSPI #1871: RP2040 with ILI9486 and SD on shared SPI doesn't reset SPI Frequency

iotricity, 2022-06-06:
First of all, thanks for this awesome library. Really appreciate your time and effort. <3

Using a Raspberry Pi 2040 with a ILI9486 TFT with touch enabled and the SD-card on the display. These three share the SPI bus. The TFT_eSPI library resets the SPI frequency when used between writing the display and reading the touch screen. Works fine so far. But when adding the SD-card and using the ESP8266SdFat SD-card library, the SPI frequency isn't set to the right speed for writing the display. Writing the display is done at the SPI frequency set by the SD-card. Until the touch screen is read. It doesn't matter if the option  SUPPORT_TRANSACTIONS is defined in the (customized) SetupXXX_ILI9486.h.

The reason for using this libraries is because it supports many types of file systems and works very well. I Would like the keep using the libary.

Used libraries: 
TFT_eSPI v2.4.61
ESP8266SdFat (which comes with the RP2040 board files https://github.com/earlephilhower/ESP8266SdFat/tree/43a65b42d5a827092188cfe11fb27237da252692)

IDE:
Arduino IDE 1.8.13

Board:
Raspberry Pi Pico Arduino core 2.0.3 (by Earle F. Philhower, III)

Wiring:
TFT MISO --> GPIO 16
TFT MOSI --> N.C.
TFT CLK --> GPIO 18
TFT CS --> GPIO 17
TFT DS --> GPIO 20
TFT RST --> GPIO 21

Touch MISO --> GPIO 16
Touch MOSI --> GPIO 19
Touch CLK --> GPIO 18
Touch CS --> GPIO 22

SD MISO --> GPIO 16
SD MOSI --> GPIO 19
SD CLK --> GPIO 18
SD CS --> GPIO 13

Attached sketch and user setup file for demonstration (added .txt extension for upload). It displays the touch location when the pen is down and flashes a large rectangle blue/green to show the speed degradation. When pressing the BOOTSEL button on the RP2040, the SD-card is check and info is shown over the serial port. After reading the SD card, writing to the display becomes very slow (@ 1MHz). Touching the screen sets the SPI frequency back to normal.

[RPi2040_ILI9486_SD_Test.ino.txt](https://github.com/Bodmer/TFT_eSPI/files/8845438/RPi2040_ILI9486_SD_Test.ino.txt)
[Setup2040_RP2040_ILI9486.h.txt](https://github.com/Bodmer/TFT_eSPI/files/8845456/Setup2040_RP2040_ILI9486.h.txt)

Bodmer, 2022-06-17:
I think this is due to the way Phil Earlehower's Rp2040 board package manages the SPI configuration. The board package does not update the SPI configuration unless it notices the configuration changes. The TFT_eSPI works with an instance of the SPI port and manages the TFT and touch controller rates, if the ESP8266SdFat changes the configuration in the SPI instance it has then the board package does not notice it has been changed and thus when the TFT rate is set it is ignored. It is only when the touch SPI speed is changed that the TFT speed change then subsequently gets noticed.  SPI configuration can be quite slow so Phil added the feature to only reconfigure the SPI if it is changed by the correcsponding instance to improve transaction performance.

It would be worth reporting the issue to Phil as so that a reconfiguration can be forced and the SPI instance change check bypassed.

As far as I am aware there is not much I can do in the library to get arounf this other than to change the SPI configuration twice with different values so it gets updated.

Bodmer, 2022-06-18:
I had a look at the ESP8266SdFat library and see that it can be initialised to use a particular SPI instance.

The SPI instance used by TFT_eSPI can be read with the function:
```
static   SPIClass& getSPIinstance(void); // Get SPI class handle
```
So the ESP8266SdFat library can then be set to the same SPI instance and solve the problem.

closed 2022-06-18 by Bodmer