Bodmer/TFT_eSPI #1873: esp32 st7789 TFT_eSPI issue

argi19, 2022-06-11:
I am new with esp32 and I would like to connect it to a 240*240 st7789 screen and I would like to use your TFT_eSPI library how can I do it

pipi61, 2022-06-11:
check and modify User_Setup.h and User_Setup_Select.h
"editing the User_Setup.h file in the library folder, or by selecting your own configuration in the "User_Setup_Selet,h" file."
https://github.com/Bodmer/TFT_eSPI/wiki/Installing-on-PlatformIO  ->>> 4. Configure library

CC-4477, 2022-06-12:
> check and modify User_Setup.h and User_Setup_Select.h "editing the User_Setup.h file in the library folder, or by selecting your own configuration in the "User_Setup_Selet,h" file." https://github.com/Bodmer/TFT_eSPI/wiki/Installing-on-PlatformIO ->>> 4. Configure library

hey, do you happen to know the specific process for doing this with an ESP8266? I'm having some trouble doing so at #1870

Bodmer, 2022-06-17:
Can you try adding this line  in your setup:
```
#define TFT_SPI_MODE SPI_MODE0
```

Bodmer, 2022-06-18:
This online tutorial details the process:
https://www.google.com/url?sa=t&source=web&rct=j&url=https://www.youtube.com/watch%3Fv%3DWFVjsxFMbSM&ved=2ahUKEwjUuOzY4Lb4AhUHB8AKHdlvCRQQwqsBegQIDBAB&usg=AOvVaw1iAITY3O9buIYd6Q7w5oz0

closed 2022-06-18 by Bodmer