Bodmer/TFT_eSPI #1874: ProS3 and ST7789 SPI

H4nky84, 2022-06-12:
I have the library working perfectly on a normal ESP32 S3 dev kit with the SPI connected to 11, 12, 13 14 etc and its great.

I tried to move the same application across to the new ProS3 from Unexpected Maker but here you have to use pins 35, 36 and 37 for SPI. I set the User Setup as i normally would and i can confirm that with the Read Config example the correct pins are being assigned.

The screen would come up with backlight but no display whatsoever. I tried with trhe standard Adafruit GFX library for ST7789 and that works fine on the ProS3 pins so the hardware SPI can be made to function, i suspect something in the SPI driver in the TFT_ESPI doesnt like the alternate pins. I got out a logic analyser and scoped the pins to discover that no signals are being output on pins 34, 35, 36 and 37 at all. My pin setup is as follows:

...
#define TFT_CS   34 //     10 or 34
#define TFT_MOSI 35 //     11 or 35
#define TFT_SCLK 36 //     12 or 36
#define TFT_MISO 37 //     13 or 37

#define TFT_DC   7
#define TFT_RST  6
...

Read Config Output:
...
[code]
TFT_eSPI ver = 2.4.70
Processor    = ESP32
Frequency    = 240MHz
Transactions = Yes
Interface    = SPI
Display driver = 7789
Display width  = 172
Display height = 320

R0 x offset = 34
MOSI    = GPIO 35
MISO    = GPIO 37
SCK     = GPIO 36
TFT_CS   = GPIO 34
TFT_DC   = GPIO 7
TFT_RST  = GPIO 6

TFT_BL           = GPIO 9
TFT_BACKLIGHT_ON = HIGH

Font GLCD   loaded
Font 2      loaded
Font 4      loaded
Font 6      loaded
Font 7      loaded
Font 8      loaded
Smooth font enabled

Display SPI frequency = 27.00
[/code]
...

and i have also tried with the inbuilt pin defines for MOSI, MISO, SCK and SS (which also gives the same read config output but still doesnt work with the screen). Running Arduino 1.8.19 and ESP32 for Arduino 2.0.3

Has anyone else tried getting the library working with the ProS3?

CC-4477, 2022-06-12:
hey! I'm using the ESP-8266-12E (NodeMCU v1) and I have the same issue. I've tried everything I know and I still can't figure out how to get the display working. I just get a blank screen with only the backlight on. not sure if I have the wrong pins, or my display is broken, but I've opened another issue last week at #1870

Bodmer, 2022-06-16:
I have used the UM ProS3 with an ILI9341 and it worked with those pins.

The UM ProS3 board MUST be selected in the board menu, the "ESP32S3 Dev module" option will not work.

H4nky84, 2022-06-17:
Yes, it works brilliantly with the ILI9341 and i can get that working on the ProS3 no worries at all with the same pins. Its only when i try the ST7789 that it doesnt seem to work.

Bodmer, 2022-06-17:
Can you try adding this line  in your setup:
```
#define TFT_SPI_MODE SPI_MODE0
```

H4nky84, 2022-06-17:
Yes! That worked! That fixed it. Thank you so much!