Bodmer/TFT_eSPI #1879: ERROR IN COMMUNICATION BETWEEN 1-RECEIVER ESP LCD ILI 9486 and 3- SENDER ESP

AjitN111, 2022-06-18:
We are unable to communicate all three sender data simultaneously on the receiver esp's LCD ILI9486.
Data is being received successfully and can be seen in serial monitor but we are unable to print it simultaneously on our LCD.
Please help.

THE CODE:

#include <esp_now.h>
#include <WiFi.h>
#include "ESPAsyncWebServer.h"
#include <Arduino_JSON.h>
#include <HTTPClient.h>
#include <SPI.h>
#include "MCUFRIEND_kbv.h"
#include <Wire.h>
#include <Adafruit_GFX.h>
#include <TFT_eSPI.h> 
TFT_eSPI tft = TFT_eSPI(); 

#define ESP32_PARALLEL
#define ILI9486_DRIVER
String temperature;
String humidity;

#define Board ID 1
#define Board ID 2
#define Board ID 3
#define Board ID 4
#define Board ID 5

// Replace with your network credentials (STATION)
const char* ssid = "vivo 1919";
const char* password = "12345678";

//ACCESS POINT credentials
const char* ssidAP = "ESP_E63B4D";
const char* passwordAP = "";

// Structure example to receive data
// Must match the sender structure
typedef struct struct_message {
  int id;
float temperature; 
float humidity;
unsigned int readingId;
} struct_message;

struct_message incomingReadings;
struct_message incomingReadings1;
struct_message incomingReadings3;

#define TFT_CD D15 
#define TFT_RTS D32
#define TFTWIDTH   480
#define TFTHIGHT   320

#define TFT_BLACK       0x0000      /*   0,   0,   0 */
#define TFT_NAVY        0x000F      /*   0,   0, 128 */
#define TFT_DARKGREEN   0x03E0      /*   0, 128,   0 */
#define TFT_DARKCYAN    0x03EF      /*   0, 128, 128 */
#define TFT_MAROON      0x7800      /* 128,   0,   0 */
#define TFT_PURPLE      0x780F      /* 128,   0, 128 */
#define TFT_OLIVE       0x7BE0      /* 128, 128,   0 */
#define TFT_LIGHTGREY   0xD69A      /* 211, 211, 211 */
#define TFT_DARKGREY    0x7BEF      /* 128, 128, 128 */
#define TFT_BLUE        0x001F      /*   0,   0, 255 */
#define TFT_GREEN       0x07E0      /*   0, 255,   0 */
#define TFT_CYAN        0x07FF      /*   0, 255, 255 */
#define TFT_RED         0xF800      /* 255,   0,   0 */
#define TFT_MAGENTA     0xF81F      /* 255,   0, 255 */
#define TFT_YELLOW      0xFFE0      /* 255, 255,   0 */
[... 379 lines cut ...]
      tft.setTextSize(2);
      tft.print("C");
      
      // display humidity
      tft.setTextSize(2);
      tft.setCursor(190, 235);
      tft.print("H: ");
      tft.print(incomingReadings.humidity);
      tft.print(" %"); 

  delay(5000);

                   // Write Text on LCD
        // display temperature

    
      
      // save the last HTTP GET Request
      //previousMillis = currentMillis;
    
   // else {
     // Serial.println("WiFi Disconnected");
  }
 
} _

Bodmer, 2022-06-18:
I am not able to support projects. Run the libray examples and study how they work.

closed 2022-06-18 by Bodmer

AjitN111, 2022-06-18:
Any specific example please